ros-navigation/navigation2 #2921: Nav2 BT_Navigator referencing deprecated Libary when defining namespace

isiko, 2022-04-20:
## Bug report

**Required Info:**

- Operating System:
  - Docker on Windows, Container running Ubuntu 20.04
- ROS2 Version:
  - Foxy Binaries
- Version or commit hash:
  - 0.4.7-1focal.20220307.024942
- DDS implementation:
  - didn't change anything, so probably `eProsima’s Fast DDS`

#### Steps to reproduce issue
1. Install Nav2 and required Packages via `apt`

2. Run launch File:
```Python
#!/usr/bin/python3

import os
from ament_index_python.packages import get_package_share_directory

from launch import LaunchDescription
from launch.actions import IncludeLaunchDescription
from launch.launch_description_sources import PythonLaunchDescriptionSource
from launch.substitutions import LaunchConfiguration
from launch.actions import DeclareLaunchArgument

def generate_launch_description():
    pkg_nav = get_package_share_directory('nav2_bringup')

    nav2 = IncludeLaunchDescription(
        PythonLaunchDescriptionSource(os.path.join(pkg_nav, 'launch', 'navigation_launch.py')),
        launch_arguments={
            'use_sim_time': LaunchConfiguration('use_sim_time'),
        }.items(),
    )

    return LaunchDescription([
        # Arguments
        DeclareLaunchArgument('use_sim_time', default_value='true'),
        DeclareLaunchArgument('namespace', default_value='robot'),

        # Actual Scripts
        nav2,
    ])
```

**I also got the same behavior when using these methods of changing the namespace:**
- `GroupAction` in combination with `PushRosNamespace`
- Copying [the launchfile](https://github.com/ros-planning/navigation2/blob/3e4dd6fa24f4d155d962b1dc9ba92110ef121641/nav2_bringup/launch/navigation_launch.py) and hardcoding the namespace `LaunchArgument`

#### Expected behavior
Nav2 stack starts in the defined `/robot` namespace with the default parameters and no errors.

#### Actual behavior
If the namespace LaunchArgument is set to anything else than an empty string, launching fails while trying to reference the `libnav2_change_goal_node_bt_node.so` library, a library that is deprecated and thus shouldn't be referenced.

```
[bt_navigator-4] [INFO] [1650484249.617820054] [bt_navigator]: Configuring
[bt_navigator-4] [ERROR] [1650484249.644469581] []: Caught exception in callback for transition 10
[bt_navigator-4] [ERROR] [1650484249.644524616] []: Original error: Could not load library: libnav2_change_goal_node_bt_node.so: cannot open shared object file: No such file or directory
[bt_navigator-4] [WARN] [1650484249.644572338] []: Error occurred while doing error handling.
[bt_navigator-4] [FATAL] [1650484249.644586034] [bt_navigator]: Lifecycle node bt_navigator does not have error state implemented
[lifecycle_manager-6] [ERROR] [1650484249.645424275] [lifecycle_manager_navigation]: Failed to change state for node: bt_navigator
[lifecycle_manager-6] [ERROR] [1650484249.645490381] [lifecycle_manager_navigation]: Failed to bring up all requested nodes. Aborting bringup.
```

#### Additional information
I already submitted #2919, which is basically the same issue, and was told to remove the library from my parameters and ask on `answers.ros.org`. I'm resubmitting this issue because I believe to have sorted out every source of error from my side (if not, please tell me what I forgot). Also, my [Question on the Forum](https://answers.ros.org/question/399119/nav2-missing-libary-when-changing-namespace/) didn't get any useful results. 

> I really **do not** want to annoy the devs, I would not resubmit if I knew any possible source of error on my side and also without additional research I wouldn't do it. If this error was my fault I am 100% sorry for wasting your time.

SteveMacenski, 2022-04-21:
This is not a bug, you ddin't push your bt yaml params into the namespace as well or included extra BT nodes in your yaml. Please review ROS 2 parameter file declarations / launch file semantics. Also make sure you're using libraries of BT nodes available in your distribution, probably by basing it off of the nav2 yaml file in your distribution's branch NOT mainline's,

closed 2022-04-21 by SteveMacenski

dohlenaugur, 2022-08-24:
If this is not a bug, is there a tutorial, how to do as you requested?

I found the parameter namespace in the [launch file](https://github.com/ros-planning/navigation2/blob/ad3fb48e7ca83969ed8893a26ac5ecafc79df0f2/nav2_bringup/launch/navigation_launch.py#L33) and it seemed as if the RewrittenYaml() should take care of the namespace.

I have installed ros-foxy-navigation2, ros-foxy-nav2-bringup and '~ros-foxy-turtlebot3-.*' via apt and did not edit any files, so why should it not work out of the box and not be a bug?

dohlenaugur, 2022-08-24:
> This is not a bug, you ddin't push your bt yaml params into the namespace as well or included extra BT nodes in your yaml. Please review ROS 2 parameter file declarations / launch file semantics. Also make sure you're using libraries of BT nodes available in your distribution, probably by basing it off of the nav2 yaml file in your distribution's branch NOT mainline's,
> 
> Also in general, please don't cross post.

If this is not a bug, is there a tutorial, how to do as you requested?

I found the parameter namespace in the [launch file](https://github.com/ros-planning/navigation2/blob/ad3fb48e7ca83969ed8893a26ac5ecafc79df0f2/nav2_bringup/launch/navigation_launch.py#L33) and it seemed as if the RewrittenYaml() should take care of the namespace.

I have installed ros-foxy-navigation2, ros-foxy-nav2-bringup and '~ros-foxy-turtlebot3-.*' via apt and did not edit any files, so why should it not work out of the box and not be a bug?