ros-navigation/navigation2 #2925: map_server needs an initial map

NikolasE, 2022-04-25:
The current implementation needs an initial parameter set that points to a valid map during on_reconfigure. If the parameters are not set, on_configure fails and the load_map_service is not started:

https://github.com/ros-planning/navigation2/blob/3e4dd6fa24f4d155d962b1dc9ba92110ef121641/nav2_map_server/src/map_server/map_server.cpp#L97

This imho limits the usability because we can't start the map server until we know which map we want to use. 

I could write a short PR that starts the services before checking the parameters and remove the thrown exception if there was no valid parameter set.

SteveMacenski, 2022-04-26:
I don't think removing the throw is a good idea, that's still valuable for users that fundamentally give it an invalid map.

What I think is a decent idea is have a special case for empty string, but explicitly empty string, map file paths. If that situation is true, then the user is explicitly stating there is no map and then it could launch and not attempt to load anything for later service calls to supply. 

So on configure/activate, we'd simply pass on loading if empty string, but still pass the overall lifecycle transition. But to do this, there MUST be a valid parameter set, of empty string. I would not support a lack of map specification being considered valid. Does that seem reasonable?

A short 2 sentences in the documentation would only be needed to explain this to users.

SteveMacenski, 2022-04-28:
merging imminent

closed 2022-04-28 by SteveMacenski