ros-navigation/navigation2 #2926: Pausing/resuming a navigation task using Nav2 Python Simple Commander

ljaniec, 2022-04-26:
## Feature request

I am using the ROS2 Galactic and Nav2 stack with Python Simple Commander in a Gazebo simulation (Ubuntu 20.04). I would like to have an option to pause (and later resume) the robot's movement while navigating.

#### Feature description

E.g. A TB3 is moving along a planned path through an intersection. I want to immediately pause its movement using a service call to trigger this, wait a few seconds, and then resume the previous navigating plan.

I checked [Simple Commander API](https://navigation.ros.org/commander_api/index.html) but I didn't find a similar command that could be used in a similar way that I would like.

#### Implementation considerations

One can try to work around this problem by saving the current position and the goal pose, then canceling the task with `cancelTask()` + resuming the task with `goToPose()`/..., but there should be a better solution, I think? `cancelTask()` cancels `future = self.goal_handle.cancel_goal_async()` too, so everything has to be recalculated later again?

ljaniec, 2022-04-26:
I guessed similar arguments and I agree. 

If this is an established way of working with the pause implementation in Nav2 (as a set u cancel task / restart target), then it works for me. In that case, I close this request.

closed 2022-04-26 by ljaniec

kleanbotmk2, 2023-08-29:
hi, so how do you stop the navigation ? Do you do it with a python node ? Thanks in advance

SteveMacenski, 2023-08-29:
You can use the API to cancel navigation https://navigation.ros.org/commander_api/index.html