ros-navigation/navigation2 #2938: Is it possible to have a galactic sync?

padhupradheep, 2022-05-06:
It is almost 6 months, since we had our last galactic sync, a lot of new features has been added as well.. 

so with humble in the horizon, how does it go? do we migrate only the selected features to Galactic?

SteveMacenski, 2022-05-06:
Fair enough, I forget about releases, I need to set a calendar reminder every 6-8 weeks to do a release...

Will do sometime today. Thanks for the reminder. If you see I haven't done one in a long time, feel free to poke me, its almost certainly not intentional.

SteveMacenski, 2022-05-06:
Closing but its on my TODO list for today so it'll get done

closed 2022-05-06 by SteveMacenski

SteveMacenski, 2022-05-06:
https://github.com/ros/rosdistro/pull/33092 ask and ye shall receive

padhupradheep, 2022-05-09:
awwweeesssomee..!!!