ros-navigation/navigation2 #2939: Adding a method to access costmap in goal_checker

hardesh, 2022-05-06:
Can we also have a method to access local costmap in the goal checker? In the current implementation, it's possible to access the goal checker in the controller. we're not passing the costmap pointer to that class. In some applications, this can be very useful. 

Ideally, I think goal checking (whether it's occupied or not) should be done in the behavior tree. But if we check the goal directly in the goal checker/controller the logic in the BT can be bypassed

padhupradheep, 2022-05-06:
> Can we also have a method to access local costmap in the goal checker?

I believe you want to reject the goal if the space is occupied?

SteveMacenski, 2022-05-06:
That's not what the goal checker is for, its for checking if the goal has been achieved yet or not. It's a binary yes/no for 
- Yes: We have reached the goal pose and can stop navigating 
- No: We have not yet reached the goal pose and must been navigating

If there was something that the costmap would be useful to you for in that context, then I would be OK adding a costmap pointer to this plugin type. However, what you're proposing is checking the status of the costmap at the goal pose, either way, that responses that the Goal Checker plugin have available wouldn't be particularly useful, if I'm understanding you use-case properly.

Even if you knew it was occupied in the Goal Checker, there wouldn't be anything it could do. It's just returning "yes I'm done navigating in my goal tolerance" or "no keep going".

I suppose though you could check if its occupied and then preemptively say "yes I'm done" regardless of the set goal tolerances / pose constraints. Is that what you had in mind? 

---

Separately, if you guys are open to donating a BT node for checking if a goal / pose / footprint is occupied, I think that would be a useful BT primitive to have available. Always happy to have more generic BT nodes that would be useful for navigation system developers (and less you need to maintain long term).

hardesh, 2022-05-07:
> Even if you knew it was occupied in the Goal Checker, there wouldn't be anything it could do. It's just returning "yes I'm done navigating in my goal tolerance" or "no keep going".

Yes this is exactly what we want to do. 

In navigation (ROS1), move_base/move_base_flex both have `isGoalReached` function directly in the controller plugin. So we do not worry about this. But in nav2 since the controller and the goal checker are separated there's no way for the goal_checker to influence the controller or vice versa.

SteveMacenski, 2022-05-09:
While we could add it, within the controller you can always throw an exception that it can't follow due to a collision. But that would return a failed status rather than a succeeded status, but have the same otherwise termination condition. I suppose it matters more semantically what you want to have happen in that situation (e.g. stop as success or start enacting failure conditions).

If you want the positive situation, please submit a PR making the change to the goal checker plugin ID and update the corresponding goal checkers to use it (so it builds)

hardesh, 2022-05-20:
It's on my TODO list. I'll try to open up a PR of the BT plugin this weekend

hardesh, 2022-05-27:
https://github.com/ros-planning/navigation2/pull/2965 Adds costmap_ros pointer in the goal_checker plugin. I do plan to add a separate BT plugin as well. Will try to open a PR in the next 2-3 weeks since it does not introduce any breaking changes for Humble release

SteveMacenski, 2022-05-27:
Awesome! I look forward to it!

closed 2022-05-27 by SteveMacenski

mini-1235, 2025-09-24:
Found this issue when refactoring the controller server

> Separately, if you guys are open to donating a BT node for checking if a goal / pose / footprint is occupied, I think that would be a useful BT primitive to have available.

So we don't actually have something like `IsPoseOccupied` in the BTNode? I think it can be very useful for https://github.com/ros-navigation/navigation2/issues/2599, for example if start/goal is occupied then we enter different behaviors in the recovery subtree

SteveMacenski, 2025-09-24:
I believe we do now, this ticket is very old 😉 https://github.com/ros-navigation/navigation2/blob/main/nav2_msgs/srv/IsPathValid.srv https://github.com/ros-navigation/navigation2/blob/main/nav2_msgs/srv/GetCosts.srv