ros-navigation/navigation2 #2946: Cannot send new navigator goals after initial invalid goal

mradams-bastian, 2022-05-12:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- ROS2 Version:
  - Galactic binaries
- Version or commit hash:
  - 1.0.8-1focal.20220430.173751

#### Steps to reproduce issue
With Nav2 running, send a NavigateToPose goal with an invalid behavior_tree parameter. 
e.g. `goal_msg.behavior_tree = "invalid/goal"`

As expected, goal is rejected with the following errors:
```
[ERROR] : Couldn't open input XML file: invalid/path
[ERROR] : BT file not found: invalid/path. Navigation canceled.
```
Then, send another goal, this time with valid parameters.

#### Expected behavior

I would expect that after correcting the goal parameters, a successful goal could be sent and executed.

#### Actual behavior

The corrected goal is rejected with another error:
```
[ERROR] : Requested navigation from navigate_to_pose while another navigator is processing, rejecting request.
```

#### Additional information

It looks like the issue rests in navigator.h in the onGoalReceived() function:

```
    if (plugin_muxer_->isNavigating()) {
      RCLCPP_ERROR(
        logger_,
        "Requested navigation from %s while another navigator is processing,"
        " rejecting request.", getName().c_str());
      return false;
    }

    plugin_muxer_->startNavigating(getName());

    return goalReceived(goal);
```

The function calls plugin_muxer_->startNavigating() whether or not goalReceived() returns true. The only time plugin_muxer->stopNavigating() is called is in the onCompletion function. But if the goal is never actually received by the navigator (due to an invalid parameter, for example), the action never initiates, and thus will never complete. Thus, the plugin_muxer_ gets stuck in the isNavigating()=true state with no way out.

I believe plugin_muxer_->startNavigating() should only be called if goalReceived() returns true.

SteveMacenski, 2022-05-12:
Are you open to submitting a PR with that? I think that sounds definitely reasonable

SteveMacenski, 2022-05-13:
Merged

SteveMacenski, 2022-05-13:
Backported to galactic

closed 2022-05-13 by SteveMacenski