ros-navigation/navigation2 #2947: Backport nav2 bt navigator fix to galactic

stan-guer, 2022-05-12:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04, applicable to all
- ROS2 Version:
  - Galactic
- Version or commit hash:
  - 20800f0ae4b50e3470f6fb12d8bed054f7f1c7e6
- DDS implementation:
  - n/a

#### Steps to reproduce issue
 Try to build navigation2

#### Additional information
I suppose the fix in #2836  to feature #2829  that was added to galactic branch in PR #2940 should be backported to galactic branch, right?  otherwise even [this typoed line of code](https://github.com/ros-planning/navigation2/blob/galactic/nav2_bt_navigator/src/bt_navigator.cpp#L111) won't build, let alone the other bugs. 

Or am I missing something?

SteveMacenski, 2022-05-12:
I believe you are correct, I should have slowed down on the galactic release I just made. Apologies. I'm building now just to reproduce, apply, and make sure that's the only error introduced.

SteveMacenski, 2022-05-12:
Thanks for the report, handled and new release is being made as we speak https://github.com/ros-planning/navigation2/pull/2948

closed 2022-05-12 by SteveMacenski