ros-navigation/navigation2 #2955: `test_msgs` are using inconsistently in the `nav2_utils` package

destogl, 2022-05-17:
[Our CI is reporting missing `test_msgs` dependency](https://github.com/robosoft-ai/SMACC2/runs/6463921234?check_suite_focus=true#step:4:29317) for `nav2_utils package.

Looking more closely to the package it looks like the `test_msgs` are needed for the tests ([ref](https://github.com/ros-planning/navigation2/blob/7abb3d983db1345b6d8bde6971ca2d5c4d44551c/nav2_util/package.xml#L40)). But those are exported in the "normal pipeline"  in the [CmakeLists.txt file](https://github.com/ros-planning/navigation2/blob/7abb3d983db1345b6d8bde6971ca2d5c4d44551c/nav2_util/CMakeLists.txt#L31). Also there is no `find_package` command for the `test_msgs`. I know how to fix this, but I am not sure what is the right logic. The easiest would be to remove `test_msgs` from [this line](https://github.com/ros-planning/navigation2/blob/7abb3d983db1345b6d8bde6971ca2d5c4d44551c/nav2_util/CMakeLists.txt#L31) and add it only under `BUILD_TESTING` section with `find_package(test_msgs REQUIRED)`
**Required Info:**

- Operating System: 20.04
- ROS2 Version: galacitc
  - 
- Version or commit hash: galactic source (7abb3d983db1345b6d8bde6971ca2d5c4d44551c)

SteveMacenski, 2022-05-19:
This has been fixed in main, we have https://github.com/ros-planning/navigation2/blob/main/nav2_util/package.xml#L40 as a test dependency and https://github.com/ros-planning/navigation2/blob/main/nav2_util/test/CMakeLists.txt#L7 use in the tests (only). iRobot came around and fixed that a bit ago. 

A PR to fix Galactic would be appreciated

tfoote, 2022-05-25:
This has broken the galactic builds making navigation2 unable to build in Galactic since 1.0.9 was released. 

https://build.ros2.org/job/Gbin_uF64__nav2_amcl__ubuntu_focal_amd64__binary/

The last successful build was almost a month ago. April 30th. 

The failure of the debian builds looks like this:

```
18:28:47 CMake Error at /opt/ros/galactic/share/nav2_util/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package):
18:28:48   By not providing "Findtest_msgs.cmake" in CMAKE_MODULE_PATH this project
18:28:48   has asked CMake to find a package configuration file provided by
18:28:48   "test_msgs", but CMake did not find one.
18:28:48 
18:28:48   Could not find a package configuration file provided by "test_msgs" with
18:28:48   any of the following names:
18:28:48 
18:28:48     test_msgsConfig.cmake
18:28:48     test_msgs-config.cmake
18:28:48 
18:28:48   Add the installation prefix of "test_msgs" to CMAKE_PREFIX_PATH or set
18:28:48   "test_msgs_DIR" to a directory containing one of the above files.  If
18:28:48   "test_msgs" provides a separate development package or SDK, be sure it has
18:28:48   been installed.
18:28:48 Call Stack (most recent call first):
18:28:48   /opt/ros/galactic/share/nav2_util/cmake/nav2_utilConfig.cmake:41 (include)
18:28:48   CMakeLists.txt:16 (find_package)
18:28:48 
```

SteveMacenski, 2022-05-25:
https://github.com/ros-planning/navigation2/pull/2969

Release for 1.0.12 to follow shortly

closed 2022-05-25 by SteveMacenski

tfoote, 2022-05-26:
Thanks for the quick fix for galactic!

vinay06vinay, 2023-10-21:
Hello , Im still getting the below error in ros galactic

```
CMake Error at CMakeLists.txt:70 (find_package):
  By not providing "Findtest_msgs.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "test_msgs", but CMake did not find one.

  Could not find a package configuration file provided by "test_msgs" with
  any of the following names:

    test_msgsConfig.cmake
    test_msgs-config.cmake

  Add the installation prefix of "test_msgs" to CMAKE_PREFIX_PATH or set
  "test_msgs_DIR" to a directory containing one of the above files.  If
  "test_msgs" provides a separate development package or SDK, be sure it has
  been installed.
```

SteveMacenski, 2023-10-21:
Sudo apt install ros-<your distro>-test-msgs

vinay06vinay, 2023-10-21:
Thank you for sharing but the below command worked not the one above "sudo apt install ros-galactic-test-msgs"