ros-navigation/navigation2 #2956: Troubles with Odom TF in ROS2 Foxy using Gazebo Ignition Fortress

Marcelomm103, 2022-05-18:
- Operating System:
 Ubuntu 20.04
- ROS2 Version:
  Foxy binary
- Version or commit hash:
  ros-foxy-navigation2	0.4.7-1focal.20220307.024942

#### Steps to reproduce issue

Hi everyone, i'm having troubles utilizing the slam_toolbox with gazebo ignition fortress. At step 1 of the nav2 tutorial (https://navigation.ros.org/tutorials/docs/navigation2_with_slam.html), i'm having the following msg: 

[controller_server-1] [INFO] [1652916158.895786595] [local_costmap.local_costmap]: Timed out waiting for transform from prius_hybrid/base_link to prius_hybrid/odom to become available, tf error: Invalid frame ID "prius_hybrid/odom" passed to canTransform argument target_frame - frame does not exist

I'm obtaining my odom from the gazebo ignition trough ign_bridge as follow:

ros2 run ros_ign_bridge parameter_bridge /model/prius_hybrid/odometry@nav_msgs/msg/Odometry[ignition.msgs.Odometry --ros-args -r /model/prius_hybrid/odometry:=/odom

When i use ros2 topic echo i obtain the following:

header:
  stamp:
    sec: 1839
    nanosec: 600000000
  frame_id: prius_hybrid/odom
child_frame_id: prius_hybrid/base_link
pose:
  pose:
    position:
      x: -5.712495709791409e-16
      y: 2.003399361101539e-40
      z: 0.0
    orientation:
      x: 0.0
      y: 0.0
      z: -2.8910903798419944e-25
      w: 1.0
  covariance:
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
twist:
  twist:
    linear:
      x: -4.52670644978346e-19
      y: 0.0
      z: 0.0
    angular:
      x: 0.0
      y: 0.0
      z: -4.526706449773389e-28
  covariance:
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0

I configured the nav2_params.yaml accordingly but get that result. I used static_tf and it worked but no map was built with slam_toolbox probably because the odom needs to come from a localization node or something, so if is static it generates error in the mapping process. I can move the ackermann car with ign_bridge and teleop_keyboard, also i can visualize lidar data on rviz2. I would like to know if anyone is using the gazebo ignition or having some similar problem, all suggestions will help. I'm sharing my project file if anyone like to try it.

[prius_gazebo.zip](https://github.com/ros-planning/navigation2/files/8721602/prius_gazebo.zip)

SteveMacenski, 2022-05-19:
I (at least) don't have any experience with Ignition to give you any advise, this might be a better topic to ask them about.

Something I'd sanity check is if you can have `/` in TF frames, I thought we were supposed to stop doing that in ROS 2. 

```
ros2 run ros_ign_bridge parameter_bridge /model/prius_hybrid/odometry@nav_msgs/msg/Odometry[ignition.msgs.Odometry --ros-args -r /model/prius_hybrid/odometry:=/odom
```

That just pushes the odometry data topic around, not the TF contribution that's used for transforms. Does the ignition plugin for odometry update TF? Maybe that topic also needs to be bridged? You'd have to look at the ignition plugin.

Marcelomm103, 2022-05-19:
Thank you for the fast reply. I'm going to ask in the gazebo answers to see if they can help me. I really dont think any TF information is being sent by the ignition. I had to use ros_ign_bridge to communicate over ros with the odom topic. I already searched a lot for tutorials or if anyone had a problem like this but nothing. Thank you for your contributions.

SteveMacenski, 2022-05-19:
OK, I'd find some examples of Ignition robot systems and see how they're dealing with TF. I would have to think whatever ROS plugins are in ignition for dealing with diff/omni/ackermann drive are going to be publishing TF, that's part of their main role.

SteveMacenski, 2022-05-19:
https://docs.ros.org/en/galactic/Tutorials/Simulators/Ignition/Setting-up-a-Robot-Simulation-Ignition.html

This tutorial makes me think you need to bridge the TF topic too

Marcelomm103, 2022-05-20:
Thank you for your help!

I've manage to find a package that implements SLAM using the slam_toolbox in ignition and ros2 (https://github.com/TechnoYantra/ros2-igt), and from what i saw in their codes they use a tf2 bridge from ignition to ros, so it seems that ignition should be publishing the tf in some way. 

Yeah i have to bridge the TF topic too to use it on ROS, but ignition doesnt generate it. I'm trying to generate the tf's using robot_state_publisher and a .urdf originally from my .sdf.

I'm going to try to generate the tf's using the robot_state_publisher or robot_localization

closed 2022-05-27 by SteveMacenski