ros-navigation/navigation2 #2958: Access action feedback within nav2_behavior_tree::BtActionNode  during node status BT::NodeStatus::RUNNING

gstrommer, 2022-05-19:
## Feature request

#### Feature description
I am using the nav2_behavior_tree::BtActionNode for executing a self defined ros2 action. When the action is running, action feedback is published by the action server. However, the nav2_behavior_tree::BtActionNode has not defined a feedback callback. I also could not find a way to access the action feedback at all. There was a similar issue that has been closed recently (#2839) with the same feature request. 

#### Implementation considerations
One could define a feedback_callback within the `send_goal_options` in the `send_new_goal()` function or the feedback could be accessible within the `void on_wait_for_result()` function.

SteveMacenski, 2022-05-19:
Agreed, what to do somewhat depends on how you want to use that information. Can you give a bit more specificity? 

I think something reasonable could be a to have `send_goal_options` have a feedback callback that stores the most recent message and then returns it to the BT node as an argument to `on_wait_for_result()`. It would need to be reset after each termination though so we didn't return stale information. 

Alternatively, have the BTActionNode have an additional public member `getFeedback()` or something that would return the most recent feedback that the BT nodes could use anywhere they desire, but probably in or around `on_wait_for_result()`.

gstrommer, 2022-05-20:
_Agreed, what to do somewhat depends on how you want to use that information. Can you give a bit more specificity?_    
The BtActionNode we are using triggers a simple movement. The feedback would give us some status information like the remaining distance to the goal position. 

I have added a feedback callback using the `send_goal_options` you mentioned as a quick fix for that problem. I think a public member like `getFeedback()` would be a nice fit here.

SteveMacenski, 2022-05-20:
> The feedback would give us some status information like the remaining distance to the goal position.

What do you aim to do with that information in the BT node? I figure the Action itself should decide when it stops, so what context does the feedback provide you in the context of a BT node being ticked that's useful? I'm trying to understand the usecase better.

>  I think a public member like getFeedback() would be a nice fit here.

You'd prefer that over having it be given to you as an argument of `on_wait_for_result` (ex `on_wait_for_result(feedback)`)?

gstrommer, 2022-05-23:
We use the same action for several different behaviors. In some cases, we verify the triggered motion by some additional measurements. Therefore, we compare the action feedback with measures, and if they do not match, we abort the action. This monitoring process should run in a parallel BTree node. Consequently, we need the feedback as an output port of our node.

I understand your point here, that the action itself should decide when it should stop or be aborted. However, for us, it is more convenient to decide this on BTree level. That's the design decision we have made.

> You'd prefer that over having it be given to you as an argument of on_wait_for_result (ex on_wait_for_result(feedback))?  

You're right, the option with an argument of `on_wait_for_result` is the better option.

SteveMacenski, 2022-05-23:
Ok. Can you submit a PR?

SteveMacenski, 2022-05-27:
Merging imminently

closed 2022-05-27 by SteveMacenski

gstrommer, 2022-05-30:
Sorry I was on holiday last week. Thank you, @SteveMacenski, for your implementation.