ros-navigation/navigation2 #2959: Apt installing is installing an old version

omerts, 2022-05-19:
- Operating System:
  - Ubuntu 20.04 
- ROS2 Version:
  - Galactic - binaries
- Version or commit hash:
```
ros-galactic-navigation2:
  Installed: 1.0.8-1focal.20220430.173751
  Candidate: 1.0.8-1focal.20220430.173751
  Version table:
 *** 1.0.8-1focal.20220430.173751 500
        500 http://packages.ros.org/ros2/ubuntu focal/main amd64 Packages
        100 /var/lib/dpkg/status
```

- DDS implementation:
 Fast-RTPS

#### Steps to reproduce issue
```
sudo apt install ros-galactic-navigation2
```

#### Expected behavior
It should install latest 1.0.11 version

#### Actual behavior
It installs version 1.0.8

padhupradheep, 2022-05-19:
We need to wait for it until the next Galactic sync

omerts, 2022-05-19:
@padhupradheep Looking at the release notes 1.0.8 was released months ago:
[image]

SteveMacenski, 2022-05-19:
It takes time to run release syncs at OSRF.

closed 2022-05-19 by SteveMacenski