ros-navigation/navigation2 #2963: Ros2 AMCL problem   exit code -11

lunanueva, 2022-05-22:
## Bug report
 Ros Navigation2  AMCL  process died with exit code -11

**Required Info:**

- Operating System:
  Ubuntu 20.04 in Host machine and Raspberry pi4
- ROS2 Version:
  -  ROS2  version Foxy  (host and Raspberry)
- Version or commit hash:
     Debian packages (host and raspi) 
- DDS implementation:
   DDS  cyclonedds

#### Steps to reproduce issue
Robot mecanum omnidirectional  4 wheels

Navigation config
amcl:
  ros__parameters:
    set_initial_pose: True
    initial_pose.x: 0.0
    initial_pose.y: 0.0
    use_sim_time: False
    alpha1: 0.2
    alpha2: 0.2
    alpha3: 0.2
    alpha4: 0.2
    alpha5: 0.2
    base_frame_id: "base_footprint"
    beam_skip_distance: 0.5
    beam_skip_error_threshold: 0.9
    beam_skip_threshold: 0.3
    do_beamskip: false
    global_frame_id: "map"
    lambda_short: 0.1
    laser_likelihood_max_dist: 2.0
    laser_max_range: 100.0
    laser_min_range: -1.0
    laser_model_type: "likelihood_field"
    max_beams: 60
    max_particles: 2000
    min_particles: 500
    odom_frame_id: "odom"
    pf_err: 0.05
    pf_z: 0.99
    recovery_alpha_fast: 0.0
    recovery_alpha_slow: 0.0
    resample_interval: 1
    robot_model_type: “nav2_amcl::OmniMotionModel" # "differential"
    save_pose_rate: 0.5
    sigma_hit: 0.2
    tf_broadcast: true
    transform_tolerance: 1.0
    update_min_a: 0.2
    update_min_d: 0.25
    z_hit: 0.5
    z_max: 0.05
    z_rand: 0.5
    z_short: 0.05
    scan_topic: scan

amcl_map_client:
  ros__parameters:
    use_sim_time: False

amcl_rclcpp_node:
  ros__parameters:
    use_sim_time: False

[... 411 lines cut ...]
[planner_server-5] [INFO] [1653222485.872199224] [global_costmap.global_costmap]: Subscribed to Topics: scan base_scan
[planner_server-5] [INFO] [1653222486.078324232] [global_costmap.global_costmap]: Initialized plugin "obstacle_layer"
[planner_server-5] [INFO] [1653222486.086307876] [global_costmap.global_costmap]: Using plugin "voxel3d_layer"
[planner_server-5] [INFO] [1653222486.124765040] [global_costmap.global_costmap]: Subscribed to Topics: realsense zed
[planner_server-5] [INFO] [1653222486.394128802] [global_costmap.global_costmap]: Initialized plugin "voxel3d_layer"
[planner_server-5] [INFO] [1653222486.398672390] [global_costmap.global_costmap]: Using plugin "inflation_layer"
[planner_server-5] [INFO] [1653222486.436991253] [global_costmap.global_costmap]: Initialized plugin "inflation_layer"
[planner_server-5] [INFO] [1653222486.618999088] [planner_server]: Created global planner plugin GridBased of type nav2_navfn_planner/NavfnPlanner
[planner_server-5] [INFO] [1653222486.619360579] [planner_server]: Configuring plugin GridBased of type NavfnPlanner
[planner_server-5] [INFO] [1653222486.768267229] [planner_server]: Planner Server has GridBased  planners available.
[lifecycle_manager-9] [INFO] [1653222487.004653087] [lifecycle_manager_navigation]: Configuring recoveries_server
[recoveries_server-6] [INFO] [1653222487.006472951] [recoveries_server]: Configuring
[recoveries_server-6] [INFO] [1653222487.137011882] [recoveries_server]: Creating recovery plugin spin of type nav2_recoveries/Spin
[recoveries_server-6] [INFO] [1653222487.149194802] [recoveries_server]: Configuring spin
[recoveries_server-6] [INFO] [1653222487.287777597] [recoveries_server]: Creating recovery plugin back_up of type nav2_recoveries/BackUp
[recoveries_server-6] [INFO] [1653222487.307423495] [recoveries_server]: Configuring back_up
[recoveries_server-6] [INFO] [1653222487.384254372] [recoveries_server]: Creating recovery plugin wait of type nav2_recoveries/Wait
[recoveries_server-6] [INFO] [1653222487.394989772] [recoveries_server]: Configuring wait
[lifecycle_manager-9] [INFO] [1653222487.472520631] [lifecycle_manager_navigation]: Configuring bt_navigator
[bt_navigator-7] [INFO] [1653222487.474559908] [bt_navigator]: Configuring
[bt_navigator-7] [INFO] [1653222488.191143588] [bt_navigator_rclcpp_node]: Waiting for "compute_path_to_pose" action server
[bt_navigator-7] [INFO] [1653222488.192493216] [bt_navigator_rclcpp_node]: "ComputePathToPose" BtActionNode initialized
[bt_navigator-7] [INFO] [1653222488.213841204] [bt_navigator_rclcpp_node]: Waiting for "global_costmap/clear_entirely_global_costmap" service
[bt_navigator-7] [INFO] [1653222488.214752191] [bt_navigator_rclcpp_node]: "ClearGlobalCostmap-Context" BtServiceNode initialize
[ERROR] [amcl-2]: process has died [pid 2385, exit code -11, cmd '/opt/ros/f

padhupradheep, 2022-05-23:
issue can be closed now..

lunanueva, 2022-05-23:
Hi

After testing my robot,  I have noted that the problem is solved only if the goal is very close and in a straight line

It is posible that it could have erroneous parameters for  omnidirectional robot but they are according with the  ROS documentation.

Config AMCL
  robot_model_type: “omnidirectional”

Config Followpath
      min_vel_x: 0.0
      min_vel_y: 0.0
      max_vel_x: 0.35
      max_vel_y: 0.35
      max_vel_theta: 1.0
      min_speed_xy: 0.0
      max_speed_xy: 0.35
      min_speed_theta: 0.0
      critics: ["RotateToGoal", "Oscillation", "BaseObstacle", "GoalAlign", "PathAlign", "PathDist",   "GoalDist", "Twirling"]
      Twirling.scale: 0.0

Error in the log file

[ERROR] [amcl-2]: process has died [pid 2441, exit code -11, cmd '/opt/ros/foxy/lib/nav2_amcl/amcl --ros-args -r __node:=amcl --params-file /tmp/tmprh8h1bi6 -r /tf:=tf -r /tf_static:=tf_static'].

Many thanks for your help.

Florencio Garcia

closed 2022-05-23 by SteveMacenski