ros-navigation/navigation2 #2966: [Nav2 Core] Change all shared pointer references to shared pointers

SteveMacenski, 2022-05-24:
Currently, that won't actually increment the reference counter

naiveHobo, 2022-05-27:
Hello! I'm trying to jump back into contributing to nav2 and just trying to get through some of the minor looking prehumble tickets for starters.

I don't see any shared pointer references in nav2_core, only `const rclcpp_lifecycle::LifecycleNode::WeakPtr & parent`. When the weak pointers are locked, it should increment the shared pointer counter correctly right?

Can you point me to the shared pointer references you're talking about?

SteveMacenski, 2022-05-27:
Hi! Welcome back! I'd love some assistance

The 2 places are https://github.com/ros-planning/navigation2/blob/main/nav2_core/include/nav2_core/smoother.hpp#L50-L52 and https://github.com/ros-planning/navigation2/blob/main/nav2_core/include/nav2_core/controller.hpp#L77-L78

SteveMacenski, 2022-05-27:
Found it https://github.com/ros-planning/navigation2/pull/2934#issuecomment-1139895429 :wink: but I know you could have found it too as the author of that hah.

But happy to have you around! Maybe ping me on slack and we can chat about what you might be interested in working on? But certainly helping on things in the issue tracker is always appreciated.

closed 2022-05-27 by SteveMacenski