ros-navigation/navigation2 #2967: [Release] Release for Humble

SteveMacenski, 2022-05-24:
Blockages today:
- CI is not up and running on 22.04
- Gazebo to ignition blockers since Gazebo is not longer supported

vinnnyr, 2022-05-25:
Could Turtlebot4 (which supports Ignition) be used instead of TB3? https://github.com/turtlebot/turtlebot4_simulator 
Sorry if this issue is not the correct place to discuss the GZ -> Ign Gz issue

SteveMacenski, 2022-05-25:
As fine of a place as any! Even in a nice demo room as well. I certainly wouldn't have any issues switching to the TB4, I'd even prefer it

SteveMacenski, 2022-05-25:
It would be nice to break out the gazebo launching into its own launch file + add an ignition launch file and have it be a launch parameter in tb3_simulation_launch.py which to apply (or even have a new tb4_simulation_launch.py which is just using ignition)

vinnnyr, 2022-05-26:
I might have gotten too excited to see the TB4 repo reference Ignition Gazeebo. 

It doesn't (or has dependencies that don't) build in Jammy / Humble.
https://github.com/turtlebot/turtlebot4_simulator/issues/16

SteveMacenski, 2022-06-01:
It seems that gazebo might have been released recently and our issues were docker related, running some experiments now but we may be able to release to humble and deal with gazebo issues in a hot minute

closed 2022-06-02 by SteveMacenski