ros-navigation/navigation2 #2970: Branch confusion

muttistefano, 2022-05-26:
hi everyone,
i cannot understand which one is the development branch.
i can see galactic and main are the most developed one, is the main branch the one corresponding to ros2 rolling ?
Thanks

automech-rb, 2022-05-26:
Yeah, the main branch corresponds to rolling. It is documented [here](https://navigation.ros.org/build_instructions/index.html)

_Build or install ROS 2 rolling using the [build instructions](https://docs.ros.org/en/rolling/Installation/Ubuntu-Development-Setup.html) provided in the ROS 2 documentation. All development is done using the rolling distribution on Nav2’s main  branch.._

automech-rb, 2022-05-26:
Actually live Groot monitoring support was dropped as explained in the migration page [here](https://navigation.ros.org/migration/Galactic.html)

muttistefano, 2022-05-26:
Thanks for the clarification

closed 2022-05-26 by SteveMacenski