ros-navigation/navigation2 #2996: Goal complete issue: main branch vs galactic branch

afonsofonbraga, 2022-06-02:
## Bug report

**Required Info:**

- Operating System: 
  - Ubuntu 
- ROS2 Version: 
  - Galactic 
- Version or commit hash:
  - 
- DDS implementation:
  - 

#### Steps to reproduce issue

```

```
I'm testing with an Ackerman robot model and I found something interesting. Using the galactic branch and RPP the robot is able to achieve the goal and finish the mission, with the base_footprint frame at the goal.  The base_footprint frame is located between the two back wheels. However, something happens in the main branch, and the mission finishes before achieving the correct position. I think it is considering the middle of the footprint and comparing it to the goal position.

#### Expected behavior
Using the current galactic branch the behavior is the following:
[image]

#### Actual behavior
However, on the Master branch, the mission is finished when the middle of the footprint is located at the goal position:
[image]

#### Additional information
This behavior happened also using the TEB. The global planner is generating the entire trajectory based on the base_footprint frame but I couldn't the reason why the goal_checker is finishing in another spot. 

----
## Feature request

#### Feature description

#### Implementation considerations

SteveMacenski, 2022-06-02:
I'm looking over the migration guide and not seeing any reason for it https://navigation.ros.org/migration/Galactic.html that at the time of a change I felt was significant enough to warrant documentation if there is indeed a change (or at least not something obvious). 

I went through the commit history in RPP and I don't see anything that stands out to me. 

What if you change the tolerances in the goal checker? I'm curious if its actually stopping at the center of the footprint or that's an accidental attribute of the goal tolerance. I'm also wondering if maybe we added a `base_frame` parameter or something that defaults to `base_link` --> is `base_link` at that point where it seems to be stopping at? Is there any frame there at all?

afonsofonbraga, 2022-06-02:
Currently, the tolerances in the goal checker are:
```
    goal_checker:
      plugin: "nav2_controller::SimpleGoalChecker"
      xy_goal_tolerance: 0.2
      yaw_goal_tolerance: 0.5
      stateful: True 
```
I also thought the base_frame was changed so I moved the base_link to the middle of the back wheels as well. It didn't change the results.

SteveMacenski, 2022-06-02:
You're going to have to do a little debugging on your side to try to narrow it down some more since this isn't something I can easily replicate on my side. Happy to help throw around ideas or once you have a general area to look in I can take a look with an expert eye and propose potential causes.

afonsofonbraga, 2022-06-03:
Hey Steve,
Thanks for your help. I thought both yamls were pointing to the same BT so I didn't bother checking them. 
Because I had to change a param I discover that the one I was using on the main branch had the "truncate_path" enabled. :clown_face:
Everything is working now. Sorry for the mess and thanks again for being available and whiling to help.

closed 2022-06-03 by afonsofonbraga

SteveMacenski, 2022-06-03:
No worries!