ros-navigation/navigation2 #2997: Migrate to Ignition / TB4 simulation

SteveMacenski, 2022-06-02:
We want to use the time in Humble to migrate to Ignition for Nav2 moving forward. 

- [ ] Provide TB4 simulation launch file (`tb4_simulation_launch.py`) for Ignition analog to TB3 for Gazebo using whatever standard launch file provided by TB4 (I believe its a warehouse)
- [ ] Migrate AWS warehouse world to Ignition
- [ ] Update simple commander examples to using Ignition with the AWS warehouse world migration
- [ ] Migrate system tests to using ignition with AWS warehouse world after https://github.com/ros-planning/navigation2/pull/2797 is merged
- [ ] Consider updating TB4 simulation launch examples to using the AWS warehouse world that the system tests and simple commander API uses. Or migrate all a different warehouse. Doesn't have to all be the same, but consider it.

padhupradheep, 2022-08-31:
Additionally there are some more suggestions:

- would be also nice to see an Omnidirectional robot part of the stack (will definitely be useful for people who wants to get started with MPPIC ). 
- On my initial experience with the Ignition, I see that, Ignition messages already have namespaces in them, so it would be nice to remap the same to the ROS messages.

vinnnyr, 2022-12-18:
I found a robot model that has a public (Ignition) Gazebo sim: https://github.com/iRobotEducation/create3_sim 
Isn't Create3 what TB4 is based off of?

SteveMacenski, 2022-12-19:
A bit too is having the binaries released in rolling and such so that we can use it in CI without an underlay space (and so users can have a nice out-of-the-box experience). I think there were things before, but not quite ready and binaries not released yet.

borongyuan, 2023-04-14:
Hi, I have migrated AWS Warehouse World to Ignition Gazebo.
https://github.com/borongyuan/aws-robomaker-small-warehouse-world
Currently it's just an empty world and lacks testing. Currently Ignition Gazebo already supports ros2_control, so there should be no problem integrating TB4 simulation. But there is still a lot of work to be done.

SteveMacenski, 2023-04-14:
Thanks! That's a great step forward though! We were just talking about this at the working group meeting yesterday.

ahcorde, 2023-06-09:
Hi, 

I'm planning to contribute with the new Gazebo simulation. But it's not clear for me where I should start and I have some questions:

 - Do you want to get rid of Gazebo-classic ? 
 - Are you thinking about living with both simulator at the same time in the repository ?
 - is there any roadmap or issue for this ?
 - which ROS 2 version should I target? `humble`, `iron` or `rolling`?
 - what's about [aws-robomaker-small-warehouse-world](https://github.com/borongyuan/aws-robomaker-small-warehouse-world) ? I checked that is was migrated in a personal Github repository but I couldn't see any open PR or branch in `https://github.com/aws-robotics/aws-robomaker-small-warehouse-world`.

If you can clarify these questions and maybe some others related to these topics that may have raised in the nav2 meeting , I'm happy to open PRs and contribute.

SteveMacenski, 2023-06-09:
Awesome! We've had a few false starts here. 

> Do you want to get rid of Gazebo-classic ?

Long term, yes, because its not supported anymore and I want Nav2 to be an example of best practices for ROS 2 and dogfood all the same stuff others are using. 

> Are you thinking about living with both simulator at the same time in the repository ?

For a distribution or two, yes, so that we can have the option to fallback if there are issues and to show side-by-side launch options for each. Just having 2 launch files for bringing up one versus the other. 

> is there any roadmap or issue for this ?

That is this. 
- [ ] Create tested and working launch files to use new-Gazebo. Probably just with the default TB3 sandbox to get started
- [ ] In no particular order: Update the robot model to be something more practical and/or TB4 if its finally ready
- [ ] In no particular order: Update the world environment to something more practical. The Sandbox world doesn't exactly inspire confidence from an out-of-the-box user experience
- [ ] Update the system tests to use new gazebo
- [ ] Optional: Update the system tests to use a more practically complex environment -- though, there's some good work here already done in a PR

> which ROS 2 version should I target? humble, iron or rolling?

Rolling, I'll plan to backport launch files for the new simulator to Humble/Iron if the matching Gazebo versions allow for it

I'm not sure what you mean about the migration to a personal github, it still appears to be released under the parent: https://github.com/ros/rosdistro/blob/master/rolling/distribution.yaml#LL500C19-L500C19

ahcorde, 2023-06-16:
> * [ ]  Create tested and working launch files to use new-Gazebo. Probably just with the default TB3 sandbox to get started
> *

I started with this bullet point. Here it's the PR https://github.com/ros-planning/navigation2/pull/3634

I added some new dependencies, let me know how does it sounds

xkaraman, 2023-06-28:
Hey, 
I can see that a check is on `tb4_simulation` launch file. I followed through #3634 but I can't seem to find a commit about it. Is this work done or maybe I can check the `turtlebot4` package and its launch files on how to integrate this?

ahcorde, 2023-06-28:
You can run the turtlebot4 simulation using nav2 from the [official repository](https://github.com/turtlebot/turtlebot4_simulator). You can follow [these steps](https://turtlebot.github.io/turtlebot4-user-manual/software/turtlebot4_simulator.html).

closed 2024-06-05 by SteveMacenski