ros-navigation/navigation2 #3010: Nav2 How to change controller publisher topic name?

rtnow, 2022-06-09:
**Required Info:**

- Operating System:
  -Ubuntu 20.04
- ROS2 Version:
  - Foxy

my robot subscribe topic such as `robot1/cmd_vel` as it's move command, but nav2 always use `cmd_vel` as move command(default in controller). And there is no information about changing controller topic publisher name from default 'cmd_vel' to other in [doc](https://navigation.ros.org/configuration/packages/configuring-controller-server.html#). Besides, remapping like [this](https://answers.ros.org/question/388925/remapping-diff-drive-controller-to-cmd_vel/) has no example.
How can I make nav2 change it's topic publish name?

SteveMacenski, 2022-06-09:
Remapping the topic in the launch file or throwing the entire navigation system into the `robot1` namespace would both equally work, depending on your aims. Both can be done from the launch file

SteveMacenski, 2022-06-09:
Example of remapping: https://github.com/ros-planning/navigation2/pull/2964/files#diff-8773725b95ea1d1d37eb7c3113851d22dd832977896cbefcc217f7a1cfaaadb2R113

rtnow, 2022-06-09:
Thank you, I solve my question

SteveMacenski, 2022-06-09:
No worries, happy to help

closed 2022-06-09 by SteveMacenski