ros-navigation/navigation2 #3012: When I use smac_planner and nav2_regulation_pure_pursuit_controlle in Ackerman, I still can't reverse the car

miku54, 2022-06-10:
- Operating System:
Ubuntu20.04
- ROS2 Version:
 Galactic

A few days ago, I posted my problem in github and kept updating and debugging. Until now, I still can't realize the function of Ackerman robot reversing in navigation2. Besides setting the param file, is there any setting for the bt file?
#### Parameters

```
progress_checker:
      plugin: "nav2_controller::SimpleProgressChecker"
      required_movement_radius: 0.5
      movement_time_allowance: 10.0
    goal_checker:
      plugin: "nav2_controller::SimpleGoalChecker"
      xy_goal_tolerance: 0.25
      yaw_goal_tolerance: 0.25
      stateful: True
    FollowPath:
      plugin: "nav2_regulated_pure_pursuit_controller::RegulatedPurePursuitController"
      desired_linear_vel: 0.5
      lookahead_dist: 0.6
      min_lookahead_dist: 0.3
      max_lookahead_dist: 0.9
      lookahead_time: 1.5
      rotate_to_heading_angular_vel: 1.8
      transform_tolerance: 0.1
      use_velocity_scaled_lookahead_dist: false
      min_approach_linear_velocity: 0.05
      use_approach_linear_velocity_scaling: true
      max_allowed_time_to_collision_up_to_carrot: 1.0
      use_regulated_linear_velocity_scaling: true
      use_cost_regulated_linear_velocity_scaling: false
      regulated_linear_scaling_min_radius: 0.9
      regulated_linear_scaling_min_speed: 0.25
      use_rotate_to_heading: true
      rotate_to_heading_min_angle: 0.785
      max_angular_accel: 3.2
      max_robot_pose_search_dist: 10.0
      use_interpolation: false
      cost_scaling_dist: 0.3
      cost_scaling_gain: 1.0
      inflation_cost_scaling_factor: 3.0
      allow_reversing: True

controller_server_rclcpp_node:
  ros__parameters:
    use_sim_time: False

planner_server:
  ros__parameters:
    planner_plugins: ["GridBased"]
    use_sim_time: True

    GridBased:
      plugin: "nav2_smac_planner/SmacPlannerHybrid"
      downsample_costmap: false           # whether or not to downsample the map
      downsampling_factor: 1              # multiplier for the resolution of the costmap layer (e.g. 2 on a 5cm costmap would be 10cm)
      allow_unknown: true                 # allow traveling in unknown space
      max_iterations: 1000000             # maximum total iterations to search for before failing (in case unreachable), set to -1 to disable
      max_planning_time: 5.0              # max time in s for planner to plan, smooth
      motion_model_for_search: "Redds-Shepp"    # Hybrid-A* Dubin, Redds-Shepp
      angle_quantization_bins: 72         # Number of angle bins for search
      analytic_expansion_ratio: 3.5       # The ratio to attempt analytic expansions during search for final approach.
      analytic_expansion_max_length: 3.0  # For Hybrid/Lattice nodes: The maximum length of the analytic expansion to be considered valid to prevent unsafe shortcutting
      minimum_turning_radius: 0.40        # minimum turning radius in m of path / vehicle
      reverse_penalty: 2.0                # Penalty to apply if motion is reversing, must be => 1
      change_penalty: 0.0                 # Penalty to apply if motion is changing directions (L to R), must be >= 0
      non_straight_penalty: 1.2           # Penalty to apply if motion is non-straight, must be => 1
      cost_penalty: 2.0                   # Penalty to apply to higher cost areas when adding into the obstacle map dynamic programming distance expansion heuristic. This drives the robot more towards the center of passages. A value between 1.3 - 3.5 is reasonable.
      retrospective_penalty: 0.015
      lookup_table_size: 20.0             # Size of the dubin/reeds-sheep distance window to cache, in meters.
      cache_obstacle_heuristic: false     # Cache the obstacle map dynamic programming distance expansion heuristic between subsiquent replannings of the same goal location. Dramatically speeds up replanning performance (40x) if costmap is largely static.
      smooth_path: True                   # If true, does a simple and quick smoothing post-processing to the path

      smoother:
        max_iterations: 1000
        w_smooth: 0.3
        w_data: 0.2
        tolerance: 1e-10
        do_refinement: true

planner_server_rclcpp_node:
  ros__parameters:
    use_sim_time: False

recoveries_server:
  ros__parameters:
    costmap_topic: local_costmap/costmap_raw
    footprint_topic: local_costmap/published_footprint
    cycle_frequency: 10.0
    recovery_plugins: ["spin", "backup", "wait"]
    spin:
      plugin: "nav2_recoveries/Spin"
    backup:
      plugin: "nav2_recoveries/BackUp"
    wait:
      plugin: "nav2_recoveries/Wait"
    global_frame: odom_combined
    robot_base_frame: base_footprint
    transform_timeout: 0.1
    use_sim_time: False
    simulate_ahead_time: 2.0
    max_rotational_vel: 0.3
    min_rotational_vel: 0.05
    rotational_acc_lim: 0.3
```

#### Expected behavior

#### Actual behavior
 ERROR when running: controller_server: ros__parameters: use_sim_time: True controller_frequency: 20.0 min_x_velocity_threshold: 0.001 min_y_velocity_threshold: 0.5 min_theta_velocity_threshold: 0.001 progress_checker_plugin: "progress_checker" goal_checker_plugins: "goal_checker" controller_plugins: ["FollowPath"]

padhupradheep, 2022-06-10:
> motion_model_for_search: "Redds-Shepp"

Should be given as `REEDS_SHEPP`

miku54, 2022-06-10:
Thanks a lot for your reply!
After changing the parameter file in the way you suggested, (use_rotate_to_heading is false, allow_reversing is true), the process dies, why is this?
(Because I did not receive a reply to the post I posted in https://answers.ros.org/, I was in a hurry, so I came here to post again, please forgive me for this impulsive behavior)
[image]

miku54, 2022-06-10:
> This does not help a lot, could show the logs above?

launch log :
```
`1654851047.0848248 [INFO] [launch]: All log files can be found below /home/wheeltec/.ros/log/2022-06-10-16-50-47-071829-wheeltec-150542
1654851047.0855494 [INFO] [launch]: Default logging verbosity is set to INFO
1654851066.5649493 [INFO] [wheeltec_robot_node-1]: process started with pid [150670]
1654851066.5657527 [INFO] [static_transform_publisher-2]: process started with pid [150672]
1654851066.6091733 [INFO] [static_transform_publisher-3]: process started with pid [150674]
1654851066.6155338 [INFO] [joint_state_publisher-4]: process started with pid [150677]
1654851066.6237698 [INFO] [robot_state_publisher-5]: process started with pid [150681]
1654851066.6243474 [INFO] [static_transform_publisher-6]: process started with pid [150685]
1654851066.6248353 [INFO] [static_transform_publisher-7]: process started with pid [150689]
1654851066.6253004 [INFO] [ekf_node-8]: process started with pid [150699]
1654851066.6257820 [INFO] [rplidarNode-9]: process started with pid [150710]
1654851066.6263156 [INFO] [map_server-10]: process started with pid [150712]
1654851066.6268058 [INFO] [amcl-11]: process started with pid [150723]
1654851066.6703887 [INFO] [lifecycle_manager-12]: process started with pid [150734]
1654851066.6837327 [INFO] [controller_server-13]: process started with pid [150736]
1654851066.6844501 [INFO] [planner_server-14]: process started with pid [150738]
1654851066.7110486 [INFO] [recoveries_server-15]: process started with pid [150749]
1654851066.7117119 [INFO] [bt_navigator-16]: process started with pid [150751]
1654851066.7122245 [INFO] [waypoint_follower-17]: process started with pid [150754]
1654851066.7127182 [INFO] [lifecycle_manager-18]: process started with pid [150757]
1654851067.0110850 [static_transform_publisher-2] [INFO] [1654851066.444516327] [base_to_link]: Spinning until killed publishing transform from 'base_footprint' to 'base_link'
1654851067.0430501 [static_transform_publisher-3] [INFO] [1654851066.292928477] [base_to_gyro]: Spinning until killed publishing transform from 'base_footprint' to 'gyro_link'
1654851067.0449023 [static_transform_publisher-7] [INFO] [1654851066.401563919] [base_to_camera]: Spinning until killed publishing transform from 'base_footprint' to 'camera_link'
1654851067.0792429 [robot_state_publisher-5] [WARN] [1654851066.757226453] [robot_state_publisher]: No robot_description parameter, but command-line argument available.  Assuming argument is name of URDF file.  This backwards compatibility fallback will be removed in the future.
1654851067.0812073 [rplidarNode-9] [INFO] [1654851066.774471365] [rplidar_node]: RPLIDAR running on ROS 2 package rplidar_ros. SDK Version:1.12.0
1654851067.0837514 [static_transform_publisher-6] [INFO] [1654851066.815853123] [base_to_laser]: Spinning until killed publishing transform from 'base_footprint' to 'laser'
1654851067.1466355 [robot_state_publisher-5] Error:   Material [white] has malformed color rgba values: Unable to parse component [255] to a double (while parsing a color value)
1654851067.1647010 [robot_state_publisher-5]          at line 90 in /tmp/binarydeb/ros-galactic-urdfdom-2.3.5/urdf_parser/src/link.cpp
1654851067.1774449 [robot_state_publisher-5] Link left_front_link had 0 children
1654851067.1779311 [robot_state_publisher-5] Link left_wheel_link had 0 children
1654851067.1797945 [robot_state_publisher-5] Link right_front_link had 0 children
1654851067.1862774 [robot_state_publisher-5] Link right_wheel_link had 0 children
1654851067.2034421 [robot_state_publisher-5] [INFO] [1654851066.873365244] [robot_state_publisher]: got segment base_link
1654851067.2358479 [robot_state_publisher-5] [INFO] [1654851066.873466652] [robot_state_publisher]: got segment left_front_link
1654851067.2455041 [robot_state_publisher-5] [INFO] [1654851066.873747751] [robot_state_publisher]: got segment left_wheel_link
1654851067.2460511 [robot_state_publisher-5] [INFO] [1654851066.873798012] [robot_state_publisher]: got segment right_front_link
1654851067.2467799 [robot_state_publisher-5] [INFO] [1654851066.873832439] [robot_state_publisher]: got segment right_wheel_link
1654851069.2740121 [ERROR] [controller_server-13]: process has died [pid 150736, exit code -6, cmd '/home/wheeltec/wheeltec_ros2/install/nav2_controller/lib/nav2_controller/controller_server --ros-args --params-file /tmp/tmpq1kodvl_ -r /tf:=tf -r /tf_static:=tf_static'].
1654851069.3516805 [rplidarNode-9] [INFO] [1654851069.350409366] [rplidar_node]: Firmware Ver: 1.29
1654851069.3522465 [rplidarNode-9] [INFO] [1654851069.350554941] [rplidar_node]: Hardware Rev: 7
1654851069.3538666 [rplidarNode-9] [INFO] [1654851069.352009183] [rplidar_node]: RPlidar health status: 0
1654851069.9002542 [rplidarNode-9] [INFO] [1654851069.894551134] [rplidar_node]: Current scan mode: Sensitivity, max_distance: 12.0 m, Point number: 7.9K , angle_compensate: 2
1654851071.4044685 [joint_state_publisher-4] [INFO] [1654851071.403358212] [joint_state_publisher]: Waiting for robot_description to be published on the robot_description topic...
1654851108.4623396 [WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
1654851109.3349247 [INFO] [lifecycle_manager-18]: process has finished cleanly [pid 150757]
1654851109.3356736 [INFO] [lifecycle_manager-12]: process has finished cleanly [pid 150734]
1654851109.3513725 [rplidarNode-9] [INFO] [1654851108.438655249] [rclcpp]: signal_handler(signal_value=2)
[... cut ...]
1654851109.8529813 [INFO] [map_server-10]: process has finished cleanly [pid 150712]
1654851109.9777398 [INFO] [recoveries_server-15]: process has finished cleanly [pid 150749]
1654851110.0225563 [INFO] [ekf_node-8]: process has finished cleanly [pid 150699]
1654851110.0548642 [INFO] [joint_state_publisher-4]: process has finished cleanly [pid 150677]
1654851110.0685520 [INFO] [amcl-11]: process has finished cleanly [pid 150723]
1654851110.3251216 [INFO] [planner_server-14]: process has finished cleanly [pid 150738]
```
`

padhupradheep, 2022-06-10:
Could you please run the debugger using one of the methods shown in the link https://navigation.ros.org/tutorials/docs/get_backtrace.html ? 

It is really strange! For me it works fine.

SteveMacenski, 2022-06-10:
> both use_rotate_to_heading and allow_reversing cannot be set to true at the same time as it would result in ambiguous situations. for your case, you have to set use_rotate_to_heading false

Please read the logs in the terminal, you should have also received a notification about this. You also would have seen logs about the wrong parameter setting for reeds-shepp models in Smac. 

Have you in some way modified these files? It never even got to the lifecycle stage before crashing. It effectively does nothing but start the costmap so maybe you have something in there that you've changed? v

miku54, 2022-06-12:
The following is the result of running the GDB debugger, and the log file
[image]

```
`1655021805.0546465 [INFO] [launch]: All log files can be found below /home/wheeltec/.ros/log/2022-06-12-16-16-45-040457-wheeltec-15142
1655021805.0553973 [INFO] [launch]: Default logging verbosity is set to INFO
1655021823.2633674 [INFO] [wheeltec_robot_node-1]: process started with pid [15301]
1655021823.2675657 [INFO] [static_transform_publisher-2]: process started with pid [15303]
1655021823.2712448 [INFO] [static_transform_publisher-3]: process started with pid [15305]
1655021823.2776709 [INFO] [joint_state_publisher-4]: process started with pid [15307]
1655021823.2908502 [INFO] [robot_state_publisher-5]: process started with pid [15309]
1655021823.2915308 [INFO] [static_transform_publisher-6]: process started with pid [15311]
1655021823.2921767 [INFO] [static_transform_publisher-7]: process started with pid [15313]
1655021823.2928348 [INFO] [ekf_node-8]: process started with pid [15315]
1655021823.2936845 [INFO] [rplidarNode-9]: process started with pid [15317]
1655021823.2943556 [INFO] [map_server-10]: process started with pid [15322]
1655021823.2967291 [INFO] [amcl-11]: process started with pid [15332]
1655021823.2976625 [INFO] [lifecycle_manager-12]: process started with pid [15346]
1655021823.3030391 [INFO] [controller_server-13]: process started with pid [15352]
1655021823.3039668 [INFO] [planner_server-14]: process started with pid [15370]
1655021823.3047104 [INFO] [recoveries_server-15]: process started with pid [15374]
1655021823.3061652 [INFO] [bt_navigator-16]: process started with pid [15383]
1655021823.3092802 [INFO] [waypoint_follower-17]: process started with pid [15385]
1655021823.3143599 [INFO] [lifecycle_manager-18]: process started with pid [15387]
1655021823.4199505 [static_transform_publisher-3] [INFO] [1655021823.217118214] [base_to_gyro]: Spinning until killed publishing transform from 'base_footprint' to 'gyro_link'
1655021823.4431744 [robot_state_publisher-5] [WARN] [1655021823.149361047] [robot_state_publisher]: No robot_description parameter, but command-line argument available.  Assuming argument is name of URDF file.  This backwards compatibility fallback will be removed in the future.
1655021823.4459488 [static_transform_publisher-6] [INFO] [1655021823.095618006] [base_to_laser]: Spinning until killed publishing transform from 'base_footprint' to 'laser'
1655021823.4650693 [static_transform_publisher-7] [INFO] [1655021823.212544045] [base_to_camera]: Spinning until killed publishing transform from 'base_footprint' to 'camera_link'
1655021823.4957590 [robot_state_publisher-5] Error:   Material [white] has malformed color rgba values: Unable to parse component [255] to a double (while parsing a color value)
1655021823.4965971 [robot_state_publisher-5]          at line 90 in /tmp/binarydeb/ros-galactic-urdfdom-2.3.5/urdf_parser/src/link.cpp
1655021823.4975641 [robot_state_publisher-5] Link left_front_link had 0 children
1655021823.5046749 [robot_state_publisher-5] Link left_wheel_link had 0 children
1655021823.5059071 [robot_state_publisher-5] Link right_front_link had 0 children
1655021823.5068636 [robot_state_publisher-5] Link right_wheel_link had 0 children
1655021823.5073004 [robot_state_publisher-5] [INFO] [1655021823.418624486] [robot_state_publisher]: got segment base_link
1655021823.5159826 [robot_state_publisher-5] [INFO] [1655021823.418734539] [robot_state_publisher]: got segment left_front_link
1655021823.5187457 [robot_state_publisher-5] [INFO] [1655021823.418783290] [robot_state_publisher]: got segment left_wheel_link
1655021823.5242133 [robot_state_publisher-5] [INFO] [1655021823.418820790] [robot_state_publisher]: got segment right_front_link
1655021823.5405738 [robot_state_publisher-5] [INFO] [1655021823.419984346] [robot_state_publisher]: got segment right_wheel_link
1655021823.5486057 [static_transform_publisher-2] [INFO] [1655021823.454517000] [base_to_link]: Spinning until killed publishing transform from 'base_footprint' to 'base_link'
1655021823.7231655 [rplidarNode-9] [INFO] [1655021823.722116408] [rplidar_node]: RPLIDAR running on ROS 2 package rplidar_ros. SDK Version:1.12.0
1655021826.2653406 [rplidarNode-9] [INFO] [1655021826.262596605] [rplidar_node]: Firmware Ver: 1.29
1655021826.2670078 [rplidarNode-9] [INFO] [1655021826.264561160] [rplidar_node]: Hardware Rev: 7
1655021826.2704690 [rplidarNode-9] [INFO] [1655021826.267916096] [rplidar_node]: RPlidar health status: 0
1655021826.8134871 [rplidarNode-9] [INFO] [1655021826.811888967] [rplidar_node]: Current scan mode: Sensitivity, max_distance: 12.0 m, Point number: 7.9K , angle_compensate: 2
1655021827.4778273 [joint_state_publisher-4] [INFO] [1655021827.476780472] [joint_state_publisher]: Waiting for robot_description to be published on the robot_description topic...
`
```

miku54, 2022-06-12:
Hi @SteveMacenski ,Regarding the nav2_params.yaml file used in the navigation file, I mainly modified the parameters of controller_server and planner_server. The modified content is in accordance with [configuring-smac-hybrid](https://navigation.ros.org/configuration/packages/smac/configuring-smac-hybrid.html) and [controllers](https://navigation.ros.org/plugins/index.html#controllers) The routines provided by  are directly copied and used

SteveMacenski, 2022-06-12:
The error listed in your terminal is pretty clear. You have the wrong parameter type.

closed 2022-06-12 by SteveMacenski

miku54, 2022-06-13:
The error report of the terminal is only relative to the error report of the URDF model, and is not directly related to navigation2, right? I have modified this error, and the error that appears in the terminal is still the same.

The error report of the terminal is only relative to the error report of the URDF model, and is not directly related to navigation2, right? I have corrected this error, but the error in the terminal is still the same. Does "wrong parameter type" refer to the content prompted by GDB outputs?

[image]

```
`1655083252.0685458 [INFO] [launch]: All log files can be found below /home/wheeltec/.ros/log/2022-06-13-09-20-52-056358-wheeltec-13236
1655083252.0693400 [INFO] [launch]: Default logging verbosity is set to INFO
1655083269.5631840 [INFO] [wheeltec_robot_node-1]: process started with pid [13377]
1655083269.5639658 [INFO] [static_transform_publisher-2]: process started with pid [13379]
1655083269.5644691 [INFO] [static_transform_publisher-3]: process started with pid [13381]
1655083269.5745368 [INFO] [joint_state_publisher-4]: process started with pid [13383]
1655083269.5751429 [INFO] [robot_state_publisher-5]: process started with pid [13394]
1655083269.5756564 [INFO] [static_transform_publisher-6]: process started with pid [13413]
1655083269.5761547 [INFO] [static_transform_publisher-7]: process started with pid [13416]
1655083269.5811920 [INFO] [ekf_node-8]: process started with pid [13418]
1655083269.5838425 [INFO] [rplidarNode-9]: process started with pid [13420]
1655083269.5882790 [INFO] [map_server-10]: process started with pid [13435]
1655083269.5954709 [INFO] [amcl-11]: process started with pid [13442]
1655083269.5986059 [INFO] [lifecycle_manager-12]: process started with pid [13444]
1655083269.6008618 [INFO] [controller_server-13]: process started with pid [13446]
1655083269.6015217 [INFO] [planner_server-14]: process started with pid [13449]
1655083269.6020949 [INFO] [recoveries_server-15]: process started with pid [13459]
1655083269.6026146 [INFO] [bt_navigator-16]: process started with pid [13461]
1655083269.6031048 [INFO] [waypoint_follower-17]: process started with pid [13464]
1655083269.6360781 [INFO] [lifecycle_manager-18]: process started with pid [13473]
1655083269.7061319 [static_transform_publisher-2] [INFO] [1655083269.298557656] [base_to_link]: Spinning until killed publishing transform from 'base_footprint' to 'base_link'
1655083269.7168045 [static_transform_publisher-3] [INFO] [1655083269.307155282] [base_to_gyro]: Spinning until killed publishing transform from 'base_footprint' to 'gyro_link'
1655083269.7438383 [robot_state_publisher-5] [WARN] [1655083269.403731755] [robot_state_publisher]: No robot_description parameter, but command-line argument available.  Assuming argument is name of URDF file.  This backwards compatibility fallback will be removed in the future.
1655083269.7449148 [robot_state_publisher-5] Link left_front_link had 0 children
1655083269.7519698 [robot_state_publisher-5] Link left_wheel_link had 0 children
1655083269.7523952 [robot_state_publisher-5] Link right_front_link had 0 children
1655083269.7533970 [robot_state_publisher-5] Link right_wheel_link had 0 children
1655083269.7673151 [robot_state_publisher-5] [INFO] [1655083269.458481687] [robot_state_publisher]: got segment base_link
1655083269.7678480 [robot_state_publisher-5] [INFO] [1655083269.458578252] [robot_state_publisher]: got segment left_front_link
1655083269.7683539 [robot_state_publisher-5] [INFO] [1655083269.458619607] [robot_state_publisher]: got segment left_wheel_link
1655083269.7702277 [robot_state_publisher-5] [INFO] [1655083269.458654503] [robot_state_publisher]: got segment right_front_link
1655083269.7783468 [robot_state_publisher-5] [INFO] [1655083269.458688827] [robot_state_publisher]: got segment right_wheel_link
1655083269.8474953 [static_transform_publisher-6] [INFO] [1655083269.545883179] [base_to_laser]: Spinning until killed publishing transform from 'base_footprint' to 'laser'
1655083269.8621900 [static_transform_publisher-7] [INFO] [1655083269.453676327] [base_to_camera]: Spinning until killed publishing transform from 'base_footprint' to 'camera_link'
1655083269.8751888 [rplidarNode-9] [INFO] [1655083269.661119256] [rplidar_node]: RPLIDAR running on ROS 2 package rplidar_ros. SDK Version:1.12.0
1655083270.2508936 [ekf_node-8] Failed to meet update rate! Took 0.05608433499999999905seconds. Try decreasing the rate, limiting sensor output frequency, or limiting the number of sensors.
1655083271.7473233 [ERROR] [controller_server-13]: process has died [pid 13446, exit code -6, cmd '/home/wheeltec/wheeltec_ros2/install/nav2_controller/lib/nav2_controller/controller_server --ros-args --params-file /tmp/tmptvjcnqbh -r /tf:=tf -r /tf_static:=tf_static'].
1655083272.2012320 [rplidarNode-9] [INFO] [1655083272.200038966] [rplidar_node]: Firmware Ver: 1.29
1655083272.2017176 [rplidarNode-9] [INFO] [1655083272.200174125] [rplidar_node]: Hardware Rev: 7
1655083272.2037218 [rplidarNode-9] [INFO] [1655083272.202900796] [rplidar_node]: RPlidar health status: 0
1655083272.7466948 [rplidarNode-9] [INFO] [1655083272.745583332] [rplidar_node]: Current scan mode: Sensitivity, max_distance: 12.0 m, Point number: 7.9K , angle_compensate: 2
1655083273.4358058 [joint_state_publisher-4] [INFO] [1655083273.434752111] [joint_state_publisher]: Waiting for robot_description to be published on the robot_description topic...
`
```

miku54, 2022-06-13:
@SteveMacenski @padhupradheep Thank you two! After I modified the parameters, I successfully solved the reversing problem!