ros-navigation/navigation2 #3013: Getting Started example fails in Humble.

kylerlaird, 2022-06-14:
## Bug report
Ubuntu 22.04
ROS Humble
ros-humble-navigation2	1.1.0-2jammy.20220607.182532

#### Steps to reproduce issue

```
source /opt/ros/humble/setup.bash
export TURTLEBOT3_MODEL=waffle
export GAZEBO_MODEL_PATH=$GAZEBO_MODEL_PATH:/opt/ros/humble/share/turtlebot3_gazebo/models
ros2 launch nav2_bringup tb3_simulation_launch.py
```

#### Expected behavior
RViz and Gazebo as shown in the example.
https://navigation.ros.org/getting_started/index.html

#### Actual behavior
The map displays in RViz but there are transform errors flagged.  Frame [map] does not exist.

Gazebo does not appear.

At command line, these errors repeat:

[component_container_isolated-5] [INFO] [1655166859.793104318] [global_costmap.global_costmap]: Timed out waiting for transform from base_link to map to become available, tf error: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
[rviz2-4] [INFO] [1655166859.795595702] [rviz]: Message Filter dropping message: frame 'base_scan' at time 78.099 for reason 'discarding message because the queue is full'
[rviz2-4] [INFO] [1655166859.795768906] [rviz]: Message Filter dropping message: frame 'odom' at time 78.088 for reason 'discarding message because the queue is full'
[rviz2-4] [INFO] [1655166859.987788149] [rviz]: Message Filter dropping message: frame 'odom' at time 78.292 for reason 'discarding message because the queue is full'
[rviz2-4] [INFO] [1655166860.020048666] [rviz]: Message Filter dropping message: frame 'base_scan' at time 78.299 for reason 'discarding message because the queue is full'
[rviz2-4] [INFO] [1655166860.179511778] [rviz]: Message Filter dropping message: frame 'odom' at time 74.525 for reason 'discarding message because the queue is full'
[rviz2-4] [INFO] [1655166860.179674535] [rviz]: Message Filter dropping message: frame 'odom' at time 78.496 for reason 'discarding message because the queue is full'
[rviz2-4] [INFO] [1655166860.212172247] [rviz]: Message Filter dropping message: frame 'base_scan' at time 78.499 for reason 'discarding message because the queue is full'

#### Additional information
I'm delighted to see Nav2 in Humble.  Thanks!

kylerlaird, 2022-06-14:
I did not set an initial pose.  I just followed the directions.  I did attempt to set a Nav2 Goal (pose) from RViz.  That had no effect.

Step 3 is "If everything has started correctly, you will see the RViz and Gazebo GUIs like this:"  I did not get Gazebo.  Also, RViz did not have a "Startup" button available (for Step 4).

I dug through the output (because it wasn't in the logs - weird) and found this:
`[ERROR] [gzserver-1]: process has died [pid 699040, exit code -2, cmd 'gzserver -s libgazebo_ros_init.so -s libgazebo_ros_factory.so /opt/ros/humble/share/nav2_bringup/worlds/world_only.model'].
[ERROR] [spawn_entity.py-2]: process has died [pid 699042, exit code -2, cmd '/opt/ros/humble/lib/gazebo_ros/spawn_entity.py -entity turtlebot3_waffle -file /opt/ros/humble/share/nav2_bringup/worlds/waffle.model -robot_namespace -x -2.00 -y -0.50 -z 0.01 -R 0.00 -P 0.00 -Y 0.00 --ros-args'].
`

SteveMacenski, 2022-06-14:
https://navigation.ros.org/getting_started/index.html#navigating Please see the guide, it pretty clearly directs you to set the initial pose to complete the TF tree to finish bringup of the system. 

It may have autostarted automatically since we have that as a default for most situations, so you just need to set the initial pose.

For gazebo crashing, that's an entirely different issue - make sure you have everything installed & have set the gazebo model paths.

empirephoenix, 2022-06-15:
Hi,
so i just got stuck on this same issue. 
After quite some more time than I'm willing to admit it finally clicked what you mean. The example tells you to click start and wait before doing the initial estimation, but in reality I must do it in opposite order.

After starting the example I need to hurry for the initial estimation, as within a few seconds (10-15) the spawn process is dieing (running into an timeout?).
Apart from the gazebo window missing, everything seems to work after that.

closed 2022-06-15 by SteveMacenski

dohren, 2022-10-08:
I checked the /opt/ros/humble/share/turtlebot3_gazebo/models directory. It didn't exist anymore after my switch from foxy to humble. It seems you have to install "sudo apt-get install ros-humble-turtlebot3-gazebo" now.

Steven7099, 2022-11-22:
It's not working for me. I have everything installed:

```
sudo apt install ros-humble-navigation2
sudo apt install ros-humble-nav2-bringup
sudo apt install ros-humble-turtlebot3*
```

Then I set the key environment variables:

```
source /opt/ros/humble/setup.bash
export TURTLEBOT3_MODEL=waffle
export GAZEBO_MODEL_PATH=$GAZEBO_MODEL_PATH:/opt/ros/humble/share/turtlebot3_gazebo/models
```

Then after running 

```
ros2 launch nav2_bringup tb3_simulation_launch.py headless:=False
```
I rush to rviz, set the 2D pose estimation as quick as I can and only end up with this in the terminal.

```
[component_container_isolated-6] [INFO] [1669122316.072762902] [amcl]: Creating bond (amcl) to lifecycle manager.
[rviz2-5] [INFO] [1669122320.120000768] [rviz2]: Setting estimate pose: Frame:map, Position(-2.97266, -0.58257, 0), Orientation(0, 0, 0.0273898, 0.999625) = Angle: 0.0547864
[spawn_entity.py-3] [ERROR] [1669122345.950231796] [spawn_entity]: Service %s/spawn_entity unavailable. Was Gazebo started with GazeboRosFactory?
[spawn_entity.py-3] [ERROR] [1669122345.950698437] [spawn_entity]: Spawn service failed. Exiting.
[ERROR] [spawn_entity.py-3]: process has died [pid 20529, exit code 1, cmd '/opt/ros/humble/lib/gazebo_ros/spawn_entity.py -entity turtlebot3_waffle -file /opt/ros/humble/share/nav2_bringup/worlds/waffle.model -robot_namespace -x -2.00 -y -0.50 -z 0.01 -R 0.00 -P 0.00 -Y 0.00 --ros-args'].

```

kikass13, 2023-01-06:
same problem here

phil123456, 2023-08-25:
pi@pi-ros2:~/WORK$ sudo apt-get install ros-humble-turtlebot3-gazebo
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package ros-humble-turtlebot3-gazebo

SteveMacenski, 2023-08-25:
https://github.com/ros-planning/navigation2/issues/3766

See the pinned ticet