ros-navigation/navigation2 #3015: Unable to complete build tutorial with foxy

1larity, 2022-06-14:
I'm self teaching myself Robotics, so please excuse any noobisness. I'm Building a robot as a retirement hobby (you'd think after 40 years of assembler and C/C+,+ I would have got tired of it now :), I've managed to Get an ESP32 based bot communicating with my PC and a raspberry Pi, now I'd like to something fun/useful with all that sensor data. 
What better than Learning some SLAM and getting it to wander around my house (or more likely crash into walls). It's taken several months to find the perfect combination of ROS/micro-ros/platformio/ESPIDE that will actually talk to each other, but I've hit a problem getting Nav2 to compile. I can see what appear to be instructions in the log, but they don't really tell me what file I should alter, or what alteration should be made (they assume a familiarity with the build process, someone following a tutorial would not have). My experience is mainly on Windows and AIX, so linux is not not my first "language" but I'm definitely improving.
With a quick web search I can see 3 people have had the same problem, but no satisfactory fix offered, either here or on ROS answers, so I'm appealing to the devs, for help/bugfix?  I do hope that's not out of order.

- Operating System:
  - Ubuntu focal fossa 20.04.4 LTS

- ROS2 Version:
  - Foxy built from source
- Version or commit hash:
  from source: output of `git -C navigation2 rev-parse HEAD
  "fatal: cannot change to 'navigation2': No such file or directory", changing "navigation2 " to ~/nav2_ws, or ~/nav2_ws/src  results in "fatal: not a git repository"
         apt binaries: output of: dpkg-query --show "ros-$ROS_DISTRO-navigation2"
 results in prompt ">"
                              or: dpkg-query --show "ros-$ROS_DISTRO-nav2-*" -->
 results in prompt ">"
- DDS implementation:
Foxy default

#### Steps to reproduce issue
Follow build tutorial

mkdir -p ~/nav2_ws/src
cd ~/nav2_ws/src
git clone https://github.com/ros-planning/navigation2.git --branch foxy-devel
cd ~/nav2_ws
rosdep install -y -r -q --from-paths src --ignore-src --rosdistro foxy
colcon build --symlink-install

#### Expected behavior
Build ROS2 Nav2
#### Actual behavior
Build fails with log
~/nav2_ws$ colcon build --symlink-install
Starting >>> nav2_common
Starting >>> nav_2d_msgs
Finished <<< nav2_common [5.89s]                                   
Starting >>> nav2_msgs
[Processing: nav2_msgs, nav_2d_msgs]                                   
Finished <<< nav_2d_msgs [1min 5s]                                           
Starting >>> nav2_voxel_grid
Finished <<< nav2_voxel_grid [29.1s]                                           
Starting >>> dwb_msgs
[Processing: dwb_msgs, nav2_msgs]                                              
[Processing: dwb_msgs, nav2_msgs]                                              
[Processing: dwb_msgs, nav2_msgs]                                              
Finished <<< dwb_msgs [1min 33s]                                              
Starting >>> nav2_gazebo_spawner
Finished <<< nav2_gazebo_spawner [4.67s]                                      
Finished <<< nav2_msgs [3min 36s]                               
Starting >>> nav2_util
[Processing: nav2_util]                                   
[Processing: nav2_util]                                       
[Processing: nav2_util]                                         
Finished <<< nav2_util [1min 31s]                            
Starting >>> nav_2d_utils
Starting >>> nav2_behavior_tree                   
--- stderr: nav2_behavior_tree                                          
CMake Error at CMakeLists.txt:14 (find_package):
  By not providing "Findbehaviortree_cpp_v3.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "behaviortree_cpp_v3", but CMake did not find one.

  Could not find a package configuration file provided by
  "behaviortree_cpp_v3" with any of the following names:

    behaviortree_cpp_v3Config.cmake
    behaviortree_cpp_v3-config.cmake

  Add the installation prefix of "behaviortree_cpp_v3" to CMAKE_PREFIX_PATH
  or set "behaviortree_cpp_v3_DIR" to a directory containing one of the above
  files.  If "behaviortree_cpp_v3" provides a separate development package or
  SDK, be sure it has been installed.

---
Failed   <<< nav2_behavior_tree [7.02s, exited with code 1]
Aborted  <<< nav_2d_utils [9.98s]                                  
                                       
Summary: 7 packages finished [5min 23s]
  1 package failed: nav2_behavior_tree
  1 package aborted: nav_2d_utils
  1 package had stderr output: nav2_behavior_tree
  22 packages not processed

#### Additional information

SteveMacenski, 2022-06-14:
You did not install all of the dependencies required, it is missing the library above. Try updating rosdep or installing manually `ros-foxy-<xyz>`

1larity, 2022-06-16:
Maybe I do not have the correct version?
ros-foxy-nav2-behavior-tree is already the newest version (0.4.7-1focal.20220307.023048).

SteveMacenski, 2022-06-16:
That is not the same thing. Please see error above

1larity, 2022-06-16:
I made a new VM and went for a binaries install, which worked okay*. I want to work on getting all my odom and suchlike working from the the ESP32. I kind of wanted to cross compile for a FPGA/6502-like homebrew, but I can live without that for a bit.

Thanks for your help!

*There is a tiny error in the turtlebot section of the tutorial which says you should hit the "startup" button in Rviz, but the button it greyed out, just skipping that and setting up the starting pose and goal works.

closed 2022-06-16 by SteveMacenski