ros-navigation/navigation2 #3016: Feature request: Run Waypoint task executor at specific waypoint

sgk-000, 2022-06-15:
## Feature request
    
#### Feature description
The feature runs a waypoint task executor at the specified waypoint. This feature allows user to specify waypoint number for each waypoint task executor, and these executor are performed at only specified waypoints. In the current implementation, the waypoint task executor is performed at all waypoints if we enable it, but there are some cases we want the executor to be performed at a specified waypoint only. i.g., if we enable the InputAtWaypoint plugin, a robot waits at all waypoint.   

#### Implementation considerations
Add a parameter(flag) to the waypoint follower if a user wants a waypoint task executor to be performed at all waypoints or specified waypoints.  
Load CSV file of waypoint numbers or write waypoint numbers directly for each waypoint task executor in the yaml file, then perform each waypoint task executor at specified waypoints only.
If this feature is acceptable for nav2, I will work on it.

In addition, I have a question(Maybe, I should ask a general question on discourse, but this question is related to the feature I requested)
Is there a similar feature with the waypoint task executor in navigateThroughPoses??
I want to do custom behavior at specified waypoints on navigateThroughPoses.

padhupradheep, 2022-06-15:
closely related to https://github.com/ros-planning/navigation2/issues/2826

SteveMacenski, 2022-06-15:
>  This feature allows user to specify waypoint number for each waypoint task executor, and these executor are performed at only specified waypoints. 

You could bake this into the task executor plugin that you create and use pretty easily. I don't think adding on a per-point basis given there is no restriction on the total number of potential waypoints you are visiting. But having a task executor plugin for your application that can be sent a flag about what to do at each index or particular indices and passing when not those would be pretty simple.

sgk-000, 2022-06-16:
> But having a task executor plugin for your application that can be sent a flag about what to do at each index or particular indices and passing when not those would be pretty simple. That way you can do this however is most natural for your application.

It is the simple and natural way, I will implement a task executor plugin for our application.
 
>if you felt that this should belong in the interface for the waypoint follower

I think the feature I requested should be in each task executor. No need to be in waypoint follower. 

 >its just driving through those points as if they are hard pose constraints on the planner, so the output is a single long path through those points.

I appreciate your simple explanations. It was clear and easy to understand.

closed 2022-06-16 by sgk-000