ros-navigation/navigation2 #3020: Writing the behavior_tree file reports an error: Missing key [bt_loop_duration]

miku54, 2022-06-16:
When I run this file with Ubuntu20.04-foxy, there is no exception. I have migrated the program to Galactic version. In this version, has the format of the behavior tree file changed?This is the error I have encountered in the content of my file
[image]
[image]

SteveMacenski, 2022-06-16:
Did you check and update all appropriate settings from the migration guides? https://navigation.ros.org/migration/index.html

miku54, 2022-06-18:
Allow me to apologize for my lateness.
I checked my code and found that there is no variable bt_loop_duration_. I used std::chrono::milliseconds(1000) instead of bt_loop_duration_, which is the problem. After adding bt_loop_duration_, there is no error, but I don't Have run it in simulation, not sure if it will work. Anyway thank you!

closed 2022-06-18 by miku54