ros-navigation/navigation2 #3023: RViz and laserscan points are behaving weirdly

ljaniec, 2022-06-17:
## Bug report

- Operating System:
Ubuntu 22.04 
- ROS2 Version:
ROS2 Humble (debian packages)
- DDS implementation:
default

#### Steps to reproduce issue

Everything was done by this page: https://navigation.ros.org/getting_started/index.html

```
sudo apt install ros-humble-navigation2
sudo apt install ros-humble-nav2-bringup
sudo apt install ros-humble-turtlebot3*
source /opt/ros/humble/setup.bash
export TURTLEBOT3_MODEL=waffle
export GAZEBO_MODEL_PATH=$GAZEBO_MODEL_PATH:/opt/ros/humble/share/turtlebot3_gazebo/models
```

In the same terminal, run
`ros2 launch nav2_bringup tb3_simulation_launch.py`

#### Expected behavior

Laserscan is displayed as small red dots around detected obstacles.

#### Actual behavior

Laserscan dot sizes vary, it seems that the key here is the resolution of the window.
[image]
[image]
[image]

#### Additional information
Probably this lines (157-162) from `nav2_default_view.rviz` stopped working correctly with RViz:

>       Name: LaserScan
>       Position Transformer: XYZ
>       Selectable: true
>       Size (Pixels): 3
>       Size (m): 0.009999999776482582
>       Style: Points

SteveMacenski, 2022-06-17:
That's odd, I don't see that on my side on Ubuntu 22.04 / Humble source builds or binaries. I'm resizing and seeing it stay consistent. Do you see any errors in the terminal? 

If you remove and readd a new laser scan visualization display and set the size to whatever you think looks reasonable, does it stay? That's really what these files are storing. If you can find a configuration that works, I'd like to see what the change is but feel free to submit an analog PR.

mikeferguson, 2022-06-18:
I've noticed this same issue with the laser scan interface when using "points" on Humble (Debian build). I'm on a slightly older Intel integrated graphics - perhaps it is related to the underlying hardware?

SteveMacenski, 2022-06-18:
That could be the case. @ljaniec / @mikeferguson  please file with Rviz2, since this isn't a Nav2 issue and those are the folks best suited to work through this issue with you.

closed 2022-06-18 by SteveMacenski