ros-navigation/navigation2 #3033: lifecycle manager wait for service should have a finite, or at least configurable, timeout.

Aposhian, 2022-06-21:
## Bug report

- Operating System:
  - Ubuntu 20.04
- ROS2 Version:
  - rolling (focal)
- Version or commit hash:
  - main
- DDS implementation:
  - FastDDS

#### Steps to reproduce issue
Give lifecycle manager a node to manage that is erroring out, and for some reason doesn't create its lifecycle services properly.

#### Expected behavior
Failure to connect to lifecycle state services should eventually timeout, or indicate that the lifecycle node it is trying to manage is not working.

#### Actual behavior
lifecycle manager blocks indefinitely.

#### Additional information
I have a hard time getting this to reproduce reliably. I think the lifecycle services failing to come up has something to do with FastDDS shared mem. This is happening with just nav2 controller_server for me.

SteveMacenski, 2022-06-21:
>  I think the lifecycle services failing to come up has something to do with FastDDS shared mem

Do you see this with Cyclone? If not, then you should file this with Fast-DDS as a problem to resolve.

> This is happening with just nav2 controller_server for me.

Try reordering the bringup sequence for lifecycle, my guess is its not actually an issue with the controller server, its just first in line.

How is it a duplicate of #2917? That was on nested timer issues. 

Do you have a specific suggested change? From what I read in this, it seems like the service clients aren't being made properly, and without more specific information / lines / issue descriptions, I can only interpolate what I think you mean by that. I think you mean that they're just straight up not working, which is not a Nav2 issue, its a DDS/RMW issue

Aposhian, 2022-06-21:
No, this never happens with CycloneDDS.

SteveMacenski, 2022-06-21:
File a ticket please with Fast-DDS and link back here so there's a thread connection. It sounds like that's really the problem -- unless you found something we can do in Nav2 to make it work? But it sounds like this is a communication issue that's out of this project's control.

SteveMacenski, 2022-06-21:
CC @MiguelCompany @EduPonz Another DDS related issue has cropped up

Aposhian, 2022-06-21:
My suggestion is to not use `wait_for_service` with an essentially infinite timeout. Even if it is an RMW failure, using an infinite timeout masks it and just results in a hanging process. I would propose using a large, but finite timeout. It could default to be something reasonable (5 minutes?) but maybe it would be a good parameter.

SteveMacenski, 2022-06-21:
Isn't that what https://github.com/ros-planning/navigation2/pull/3029 does from https://github.com/ros-planning/navigation2/issues/3027? If so, we can close this ticket since we're on the same page about that and we're just working through the details to be merged. 

The fact that this is an issue in Fast-DDS is an issue that should be reported to them, though.

SteveMacenski, 2022-06-21:
> My suggestion is to not use wait_for_service with an essentially infinite timeout. Even if it is an RMW failure, using an infinite timeout masks it and just results in a hanging process. I would [...]

Feel free to propose a PR. 5 min seems a little nuts, 10 sec seems more in line with reasonable even if respawned from a remote server

Aposhian, 2022-06-22:
It is important to note that this error condition does not happen _every_ time. It is inconsistent.

MiguelCompany, 2022-06-23:
@Aposhian One thing I don't understand. On the issue description you say this happens on rolling, but on the gist you just shared it uses `ros:galactic` as the base docker image.

So my question is: Does the issue only reproduce in galactic? Does it also reproduce on humble/rolling?

Aposhian, 2022-06-24:
Yes, I am using old rolling on focal, so closer to galactic than to humble.

Aposhian, 2022-06-24:
Yes, I cannot reproduce this issue on Humble images.

SteveMacenski, 2022-06-24:
Keep in mind that Galactic is EOL in 5 months.

Since this is a galactic-specific problem, what do you want to do about it? Would it be possible to backport the patch @MiguelCompany? If not, we could try to patch something in just the galactic branch for you @Aposhian for the interim? If its _just_ for Galactic to deal with _just_ an issue with Fast-DDS, I'm OK being a bit more sloppy about it than I normally would since it's lifecycle is very limited and would not be pushed into future distributions -- the maintenance cost and technical reach is limited.

Aposhian, 2022-06-24:
@MiguelCompany suggested I try building FastDDS from source on the 2.3.x branch, which has unreleased fixes for galactic. I don't think any Nav2 action is required. I will update if that is a viable workaround for galactic.

SteveMacenski, 2022-06-24:
OK - if that fixes things feel free to close when ready!

Aposhian, 2022-06-24:
Ok, I tried building from 2.3.x FastDDS, but it doesn't resolve the issue. https://gist.github.com/Aposhian/043359e09a203900e8db55407a8b5e38
```
rmw: rmw_cyclonedds_cpp | daemon: ✅
  Try #1: ✅
  Try #2: ✅
  Try #3: ✅
rmw: rmw_cyclonedds_cpp | daemon: ❌
  Try #1: ✅
  Try #2: ✅
  Try #3: ✅
rmw: rmw_fastrtps_cpp | daemon: ✅
  Try #1: ❌
  Try #2: ✅
  Try #3: ✅
rmw: rmw_fastrtps_cpp | daemon: ❌
  Try #1: ✅
  Try #2: ✅
  Try #3: ✅
```

Aposhian, 2022-07-01:
@MiguelCompany while not specifically for Nav2, lifecycle services failing is still happening on humble intermittently. Can we migrate/copy this issue to Fast-DDS?

SteveMacenski, 2022-07-01:
Agreed, some private conversations I've had in recent weeks have also expressed issues with services in Humble

SteveMacenski, 2022-07-07:
In the same theme of #3032, I want to get this off the queue so trying to find out the actionable things we can do to resolve this. I know the main issue is Fast-DDS not handling lots of service calls well, but I suppose we can improve the way these failures occur. The lifecycle manager and BT Action Node in BT Navigator are the only 2 places where we have waits without timeouts intentionally since these are pretty critical operations.

For this (lifecycle manger), what should happen if a service for transitioning doesn't come up? Fail everything? I suppose we could also remove the checks for services being up in the constructors https://github.com/ros-planning/navigation2/blob/main/nav2_util/src/lifecycle_service_client.cpp#L39 knowing that we check if they're up before we submit requests in the change/get methods that were recently added due to supporting respawning. I'll say though that the current status of lifecycle services and the manager are in a stable but thoughtfully setup balance due to supporting Bond, Respawn, and Lifecycle all at the same time. Removing that could cause some real headaches other places, but I won't be sure until I try if that seems to be the best answer.

Aposhian, 2022-07-07:
Yeah I'm torn as to whether this is something that Nav2 should consider to fallible as a part of normal operation, and gracefully handle it. But once you start to question the underlying RMW, lots of other assumptions go out the window.
Even if we don't add in a configurable timeout, we could probably make the log messages more explicit as to what it is waiting on, so there is no ambiguity like "Waiting for lifecycle service..." or something like that.

jsantiago-eProsima, 2022-07-08:
Hi @Aposhian @SteveMacenski 

Since you also said this is happening on Humble, we have moved our investigations there with no luck so far. Using your reproducer on Humble yields no results, being able to run hundreds of times without error. 

Do you have any other clue on how to have this issue reproduced on Humble?

We also have [this PR](https://github.com/ros2/rmw_fastrtps/pull/619) open that aims to bring Fast DDS's Waitsets to the RMW. It's still under review but you could try using it and see if the situation improves.

Aposhian, 2022-07-08:
I think that sounds reasonable. However, we won't be much use testing this at the moment, since we have decided to use CycloneDDS with Humble for our main operation at the moment.

SteveMacenski, 2022-07-09:
OK, I can add that to my task list to change early next week to add the log / waits. I could actually do this too in the BT Action Node too so that in either case when waiting if it isn't connecting we're printing that for awareness. Do you think that's a good solution to #3032 as well?

Aposhian, 2022-07-11:
Yes I think so.

SteveMacenski, 2022-07-11:
Sweet, its on my queue for this afternoon

SteveMacenski, 2022-07-11:
https://github.com/ros-planning/navigation2/pull/3071 implements

SteveMacenski, 2022-07-13:
Sorry, this should stay open until the service reliability is handled in Fast-DDS

SteveMacenski, 2022-08-24:
@Aposhian that PR was merged that supposedly fixes this -- have you tested by chance?

Aposhian, 2022-08-24:
No, but this is probably something we can get to testing in the next month.

vinnnyr, 2022-08-25:
@SteveMacenski  can you please point us to the upstream PR that we should make sure we test with?

SteveMacenski, 2022-08-25:
yes!

SteveMacenski, 2022-10-13:
@Aposhian I don't suppose you tested as part of your migration back to Fast-DDS to use the features you liked?

Aposhian, 2022-10-13:
No not yet

EduPonz, 2022-11-24:
Hi everyone,

The new [Humble sync is out](https://discourse.ros.org/t/new-packages-and-patch-release-for-humble-hawksbill-2022-11-23/28495) which contains the changes made to solve this issue.

SteveMacenski, 2022-11-28:
@Aposhian / @vinnnyr if you could try that out since I know you want to be using Fast-DDS ideally, that would be fantastic.

vinnnyr, 2022-11-28:
will report back if we end up having time to try it. unfortunately both of us seem to be a bit short on time for the rest of the year, so I cannot promise anything.

EduPonz, 2023-02-02:
Hi everyone,

On January the 27th there was a Humble patch release which included additional fixes (mainly eProsima/Fast-DDS#3195). It'd great if someone could give this a try a see whether this ticket can be closed.

SteveMacenski, 2023-02-24:
Agreed!

AlexeyMerzlyakov, 2023-03-07:
> On January the 27th there was a Humble patch release which included additional fixes (mainly https://github.com/eProsima/Fast-DDS/pull/3195). It'd great if someone could give this a try a see whether this ticket can be closed.

Unfortunately, it is confirmed that the problem is reproduced on Fast-DDS with PR3195 patch: very rare, but still appear typically 1 time per ~50-100 Nav2 stack runs. Checked on Ubuntu 20.04 with ROS2 Rolling built from sources (20230110 version, PR3195 is there).

Just for the reference: for verification I've used the Nav2 system tests that are deploying full Nav2 stack in each turn. The stress-test script is attached to current message: [keepout_stress.sh](https://github.com/ros-planning/navigation2/files/10907931/keepout_stress.sh.txt)

ygoumaz, 2023-03-13:
I can confirm that we still encounter this issue with the `webots_ros2` package. Our tests that use the Nav2 package randomly fail. Failures occur more often on CI machines than locally. Changing to Cyclone DDS seems to resolve the issue.

SteveMacenski, 2024-04-05:
This hasn't received any updated in about a year with anyone noting an issue and I also haven't seen it in my day to day development in some time.

I'm closing this ticket as fixed by Fast-DDS since it does not appear to continue to be a problem of reportable frequency anymore. Though, happy to reopen if anyone can show that this is still and issue plaguing their application.

closed 2024-04-05 by SteveMacenski