ros-navigation/navigation2 #3034: No costmaps being published

tcdtech, 2022-06-21:
## Bug? report

**Required Info:**

- Operating System:
  Ubuntu 22.04 Jammy
- ROS2 Version:
  Humble
- Version or commit hash:
  ros-humble-navigation2	1.1.0-2jammy.20220607.182532
- DDS implementation:
 Fast-RTPS (I believe, as it's the default)

#### Steps to reproduce issue
I'm running the nav2 stack pretty much as specified in the nav2_bringup package but without AMCL or obstruction layers as I currently have no LIDAR.

#### Expected behavior
I should be able to see (and debug) my costmaps.

#### Actual behavior
I can't see any costmaps in rviz2 and don't see anything when doing "ros2 topic echo" on the costmap topics.

Rudimentary navigation seems to work (robot will drive from A to B and will plan paths around objects in my static map) so the nav stack seems to have _a_ costmap available to it. I'm not sure if they're being inflated properly though.

#### Additional information

Not sure if this is a bug or if I'm doing something wrong.

- They do not show up in RViz (No map received warning)
- If I do 'ros2 topic echo /global_costmap/costmap' I get no output no matter what order I run things in.
- The same is true for /global_costmap/costmap_raw, /global_costmap/costmap_updates, /local_costmap/costmap, /local_costmap/costmap_raw and /local_costmap/costmap_updates
- The footprint is being published for both global and local nodes though so they're definitely running.

The launch file is almost identical to the nav2_bringup one (though I'm happy to provide my version if it would help).

These are the costmap parameters I'm using though I've tried changing almost all of them already. 

`global_costmap:
  global_costmap:
    ros__parameters:
      update_frequency: 0.5
      publish_frequency: 1.0
      global_frame: map
      robot_base_frame: base_link
      use_sim_time: True
      footprint: "[ [0.3,-0.15], [0.3,0.15], [0.225,0.15], [0.225,0.5], [-0.225, 0.5], [-0.225, 0.15], [-0.3,0.15], [-0.3, -0.15] ]"
      resolution: 0.05
      track_unknown_space: False
      plugins: ["static_layer", "inflation_layer"]
      static_layer:
        plugin: "nav2_costmap_2d::StaticLayer"
        map_subscribe_transient_local: False
        subscribe_to_updates: True
        map_topic: /map
        enabled: True
      inflation_layer:
        plugin: "nav2_costmap_2d::InflationLayer"
        cost_scaling_factor: 3.0
        inflation_radius: 0.55
        enabled: True
      always_send_full_costmap: True
  global_costmap_client:
    ros__parameters:
      use_sim_time: True
  global_costmap_rclcpp_node:
    ros__parameters:
      use_sim_time: True

local_costmap:
  local_costmap:
    ros__parameters:
      update_frequency: 5.0
      publish_frequency: 2.0
      global_frame: odom
      robot_base_frame: base_link
      use_sim_time: True
      rolling_window: true
      width: 3
      height: 3
      resolution: 0.05
      footprint: "[ [-0.15, 0.3], [0.15,0.3], [0.15,0.225], [0.5,0.225], [0.5,-0.225], [0.15,-0.225], [0.15,-0.3], [-0.15,-0.3] ]"
      plugins: ["static_layer", "inflation_layer"]
      static_layer:
        plugin: "nav2_costmap_2d::StaticLayer"
        map_subscribe_transient_local: False
        map_topic: /map
        enabled: True
      inflation_layer:
        plugin: "nav2_costmap_2d::InflationLayer"
        cost_scaling_factor: 3.0
        inflation_radius: 0.55
        enabled: True
      always_send_full_costmap: True`

SteveMacenski, 2022-06-21:
Note https://github.com/ros-planning/navigation2/issues/3014, I think that's really the crux of this ticket. 

> will plan paths around objects in my static map

Your comment about echoing is probably due to the QoS and since you can see the static layer in rviz means its being published.

tcdtech, 2022-06-21:
I read that bug report before posting mine. Wasn't sure if it was relevant to me or not.

Key differences:
- I'm not using the obstacle layer at all.
- I have a static layer applied to both local and global costmaps (which someone mentioned fixed that issue).

To clarify, I can view the `/map` topic in RViz but none of the costmaps ever show up, no matter what QoS settings I set or what order I start things in (RViz before or after the navstack makes no difference).

Does it still sound like the same issue? I'm not at all familiar with the source code..

[image]

SteveMacenski, 2022-06-21:
Oh neat, can you publish your logs?

tcdtech, 2022-06-21:
Tyvm for your time/help btw.

I was just trying Cyclone-DDS. It doesn't _seem_ to make a difference. Will restart everything and try again just incase parts of the networks are still using Fast-DDS.

### Launch terminal output

```
[INFO] [launch]: All log files can be found below /home/tom/.ros/log/2022-06-21-22-02-39-389046-tom-VirtualBox-4176
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [map_server-1]: process started with pid [4178]
[INFO] [controller_server-2]: process started with pid [4180]
[INFO] [smoother_server-3]: process started with pid [4182]
[INFO] [planner_server-4]: process started with pid [4184]
[INFO] [behavior_server-5]: process started with pid [4211]
[INFO] [bt_navigator-6]: process started with pid [4213]
[INFO] [waypoint_follower-7]: process started with pid [4228]
[INFO] [lifecycle_manager-8]: process started with pid [4231]
[smoother_server-3] [INFO] [1655845360.968481024] [smoother_server]: 
[smoother_server-3] 	smoother_server lifecycle node launched. 
[smoother_server-3] 	Waiting on external lifecycle transitions to activate
[smoother_server-3] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[smoother_server-3] [INFO] [1655845360.971962158] [smoother_server]: Creating smoother server
[bt_navigator-6] [INFO] [1655845360.973841346] [bt_navigator]: 
[bt_navigator-6] 	bt_navigator lifecycle node launched. 
[bt_navigator-6] 	Waiting on external lifecycle transitions to activate
[bt_navigator-6] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[bt_navigator-6] [INFO] [1655845360.973923499] [bt_navigator]: Creating
[planner_server-4] [INFO] [1655845360.997047382] [planner_server]: 
[planner_server-4] 	planner_server lifecycle node launched. 
[planner_server-4] 	Waiting on external lifecycle transitions to activate
[planner_server-4] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[planner_server-4] [INFO] [1655845361.003563966] [planner_server]: Creating
[lifecycle_manager-8] [INFO] [1655845361.149762354] [lifecycle_manager_navigation]: Creating
[lifecycle_manager-8] [INFO] [1655845361.504895657] [lifecycle_manager_navigation]: Creating and initializing lifecycle service clients
[behavior_server-5] [INFO] [1655845361.775748970] [behavior_server]: 
[behavior_server-5] 	behavior_server lifecycle node launched. 
[behavior_server-5] 	Waiting on external lifecycle transitions to activate
[behavior_server-5] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[waypoint_follower-7] [INFO] [1655845362.470568889] [waypoint_follower]: 
[waypoint_follower-7] 	waypoint_follower lifecycle node launched. 
[waypoint_follower-7] 	Waiting on external lifecycle transitions to activate
[waypoint_follower-7] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[waypoint_follower-7] [INFO] [1655845362.471101123] [waypoint_follower]: Creating
[map_server-1] [INFO] [1655845362.649547247] [map_server]: 
[map_server-1] 	map_server lifecycle node launched. 
[map_server-1] 	Waiting on external lifecycle transitions to activate
[map_server-1] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[map_server-1] [INFO] [1655845362.650125279] [map_server]: Creating
[controller_server-2] [INFO] [1655845363.303613155] [controller_server]: 
[controller_server-2] 	controller_server lifecycle node launched. 
[controller_server-2] 	Waiting on external lifecycle transitions to activate
[controller_server-2] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[controller_server-2] [INFO] [1655845363.312442590] [controller_server]: Creating controller server
[planner_server-4] [INFO] [1655845364.672643226] [global_costmap.global_costmap]: 
[planner_server-4] 	global_costmap lifecycle node launched. 
[planner_server-4] 	Waiting on external lifecycle transitions to activate
[planner_server-4] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[planner_server-4] [INFO] [1655845364.688895319] [global_costmap.global_costmap]: Creating Costmap
[lifecycle_manager-8] [INFO] [1655845365.350184071] [lifecycle_manager_navigation]: Starting managed nodes bringup...
[lifecycle_manager-8] [INFO] [1655845365.350232968] [lifecycle_manager_navigation]: Configuring map_server
[map_server-1] [INFO] [1655845365.383835030] [map_server]: Configuring
[controller_server-2] [INFO] [1655845365.419280602] [local_costmap.local_costmap]: 
[controller_server-2] 	local_costmap lifecycle node launched. 
[controller_server-2] 	Waiting on external lifecycle transitions to activate
[controller_server-2] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[controller_server-2] [INFO] [1655845365.419791302] [local_costmap.local_costmap]: Creating Costmap
[lifecycle_manager-8] [INFO] [1655845366.038818345] [lifecycle_manager_navigation]: Configuring controller_server
[controller_server-2] [INFO] [1655845366.039740749] [controller_server]: Configuring controller interface
[controller_server-2] [INFO] [1655845366.040636001] [controller_server]: getting goal checker plugins..
[controller_server-2] [INFO] [1655845366.054031800] [controller_server]: Controller frequency set to 20.0000Hz
[... 416 lines cut ...]
    lifecycle_manager = launch_ros.actions.Node(
                package='nav2_lifecycle_manager',
                executable='lifecycle_manager',
[... cut ...]
                output='screen',
                parameters=[
                            #{'use_sim_time': use_sim_time},
                            {'autostart': True},
                            {'node_names': lifecycle_nodes}
                            ])

    return launch.LaunchDescription([

        map_server,
        controller,
        smoother,
        planner,
        behaviours,
        navigator,
        waypoint_follower,
        #velocity_smoother,
        
        lifecycle_manager
    ])
```

SteveMacenski, 2022-06-21:
```
      static_layer:
        plugin: "nav2_costmap_2d::StaticLayer"
        map_subscribe_transient_local: True
        map_topic: /map
```

The only thing that comes to mind looking over that is maybe your static layer isn't getting the initial map? Topic is wrong or the QoS settings are wrong (e.g. `map_subscribe_transient_local`). 

Your launch file does something weird and has a second configuration file its setting for map server

```
    configured_params = map_server_config = os.path.join(pkg_share, 'config/map_server_params.yaml')
``` 

I'd look into that.  What if you start from just the example `tb3_simulation_launch.py` file? Does that work? If so, I think you can start working backward from there to find the issue.

SteveMacenski, 2022-06-21:
> I believe these three lines in the launch log imply that the static layer is getting the map

Agreed. 

I'm not actually referring to double assignment as I am that you have a separate `config/map_server_params.yaml` being used (unless that's what you provided me above and its just named `map_server_params` but contains everything?) but the main file you linked to above isn't set anywhere in your launch file.

tcdtech, 2022-06-23:
> unless that's what you provided me above and its just named map_server_params but contains everything?
This was correct. The parameter file was just named poorly.. 

So I've figured it out and it was, of course, the **_dumbest_** thing. Got the tb3_simulation_launch going (which produced costmaps fine) then copied the key files over to my package and started breaking it down.

`use_sim_time: True` in my parameters file was the only culprit, I'd removed the reference to `RewrittenYaml()` in the launch file so it wasn't being re-written to False. I'm not running a sim, so several of the nodes were waiting for the sim clock to tick which it never did. I didn't even know that's how `use_sim_time` worked, hadn't used gazebo before last night.

Thank you so so much for your help, time and everything else you do for ROS. As tivial as it ended up being I wouldn't have worked it out for a long time without your advice. 

Bonus robot & cat tax:
https://photos.app.goo.gl/Zz4VmYNZS5LgBGSJ8

That's just my testbed, running on windscreen wiper motors and and old nissan leaf battery module. I'm welding up the chassis for my second bot as we speak!

closed 2022-06-23 by tcdtech

RobinBaruffa, 2024-06-25:
Had the exact same problem, thanks for the answer