ros-navigation/navigation2 #3037: Add a GoalUpdatedController to BT Nodes

SteveMacenski, 2022-06-23:
Essentially, if the goal is updated, allow the children to be ticked, analog to RateController or DistanceController but the triggering event is that the `goal` or `goals` BT blackboard variables are updated due to preemption or a new request. 

That way, for instance, we could replan only when a new goal is requested. Add a BT XML showcasing this.

Also worth considering adding some goal updated logic in the default BT XMLs so that we can replan immediately on a goal change and not wait for the next replanning cycle due to time, distance, etc.

SteveMacenski, 2022-06-23:
Output of discussion from https://github.com/ros-planning/navigation2/issues/3022

padhupradheep, 2022-06-24:
I was just thinking the exact same yesterday, while I was carrying out the tests for MPPIC. 

whatta coincidence !