ros-navigation/navigation2 #3039: Fix codecov metrics

SteveMacenski, 2022-06-23:
Coverage metrics for system tests are not being counted, potentially among other things

CC @ruffsl

SteveMacenski, 2022-06-27:
I think there's usually a little noise between runs due to the nature of system level smoke tests, but it shouldn't be massively different in the final number. 

That's odd, maybe the default compiling flags have changed in `colcon`/`ament` or other 22.04 related changes?  Obviously, this happened when we moved to 22.04 / Rolling, so something there changed. Coudl it be some change w.r.t. `lld`?

I was wondering also if a system test fails, if there could have been a change that resulted in none of the coverage metrics by the package being reported - though that may be an ill-informed thought, I haven't looked into detail about how exactly code coverage metrics are captured

ruffsl, 2022-06-28:
@tylerjw , would you have any ideas here? I'm thinking if it isn't lcov, then perhaps the bump in gcc version has changed the effectiveness of the default gcc args specified using the either coverage-gcc colcon mixin. Was moveit's code coverage effected by the migration to 22.04?

nuclearsandwich, 2022-06-30:
> @nuclearsandwich do you have any experience with this?

No it's not something that has crossed my feeds as far as I recall. The major jammy-related fallout that I've encountered has come down to 1. needing to update the docker seccomp profile so that new system calls are not masked inside jammy containers running on older hosts (primarily affects you if you're running docker from the Debian or Ubuntu repositories rather than from the official Docker repositories) or 2. [Phased rollout of stable release updates](https://wiki.ubuntu.com/PhasedUpdates) which affects apt-get / apt cli rather than just the GUI tools as of 22.04.

But both failures are usually more catastrophic for builds than what you're seeing.

@ruffsl nav2 CI currently uses the nightly packaging job archives right?
How readily possible is it to use a custom tarball in place of the most recent successful nightly? Since Humble maintains Ubuntu Focal as a tier 3 platform currently, I could try building an archive on Focal from a manual job so we could tell whether the change is platform/environmental or if something changed in CMake logic throughout the ROS system.

ruffsl, 2022-07-01:
> nav2 CI currently uses the nightly packaging job archives right?

No, we have switched to rolling for the time being, as all of our dependencies have been upstreamed, so it make development a bit more stable to work with.

> How readily possible is it to use a custom tarball in place of the most recent successful nightly?

Not too hard, I pushed some commits to my fork that revert the Docker image to build from nightly. You could modify from base image and URL or COPY the tarball in the nightly dockerfile, build and tag it, then change the FROM image in the Dockerfile on my fork to try it out. If you give me a URL for focal, I could also do it out myself.

https://github.com/osrf/docker_images/blob/c5d0709fd289e516a1c501c2120fcf1c08da6fd8/ros2/nightly/nightly/Dockerfile#L88-L92

https://github.com/ruffsl/navigation2/blob/707c65f9aaa106bafd046f328f92f67558065b52/Dockerfile#L9

One can also run all the tests by changing the default build args, E.g:

```
docker build --pull \
  -t ghcr.io/ruffsl/navigation2:nightly \
  --build-arg RUN_TESTS=True \
  --build-arg FAIL_ON_TEST_FAILURE="" \
  .
```

Doing this using the current `osrf/ros2:nightly` image from Dockerhub, I see about the same number of skipped tests:

```
...
[Processing: nav2_system_tests]
Finished <<< nav2_system_tests [26min 19s]	[ with test failures ]

Summary: 36 packages finished [28min 51s]
  6 packages had stderr output: nav2_behavior_tree nav2_behaviors nav2_controller nav2_map_server nav2_simple_commander nav2_system_tests
  5 packages had test failures: nav2_behavior_tree nav2_behaviors nav2_controller nav2_map_server nav2_system_tests
build/nav2_behavior_tree/Testing/20220701-1203/Test.xml: 52 tests, 0 errors, 1 failure, 0 skipped
build/nav2_behaviors/Testing/20220701-1205/Test.xml: 7 tests, 0 errors, 1 failure, 0 skipped
build/nav2_behaviors/test_results/nav2_behaviors/cpplint.xunit.xml: 17 tests, 0 errors, 8 failures, 0 skipped
build/nav2_controller/Testing/20220701-1204/Test.xml: 8 tests, 0 errors, 1 failure, 0 skipped
build/nav2_controller/test_results/nav2_controller/pctest.gtest.xml: 1 test, 1 error, 0 failures, 0 skipped
build/nav2_map_server/Testing/20220701-1203/Test.xml: 12 tests, 0 errors, 2 failures, 0 skipped
build/nav2_map_server/test_results/nav2_map_server/cpplint.xunit.xml: 22 tests, 0 errors, 1 failure, 0 skipped
build/nav2_map_server/test_results/nav2_map_server/uncrustify.xunit.xml: 22 tests, 0 errors, 1 failure, 0 skipped
build/nav2_system_tests/Testing/20220701-1205/Test.xml: 30 tests, 0 errors, 5 failures, 0 skipped
build/nav2_system_tests/test_results/nav2_system_tests/test_backup_recovery.xml: 1 test, 1 error, 0 failures, 0 skipped
build/nav2_system_tests/test_results/nav2_system_tests/test_behavior_tree_node.gtest.xml: 7 tests, 0 errors, 1 failure, 0 skipped
build/nav2_system_tests/test_results/nav2_system_tests/test_keepout_filter.xml: 1 test, 1 error, 0 failures, 0 skipped
build/nav2_system_tests/test_results/nav2_system_tests/test_speed_filter_global.xml: 1 test, 1 error, 0 failures, 0 skipped
build/nav2_system_tests/test_results/nav2_system_tests/test_waypoint_follower.xml: 1 test, 1 error, 0 failures, 0 skipped

Summary: 2902 tests, 5 errors, 21 failures, 572 skipped
Removing intermediate container a8a53a2ad0c4
 ---> 60f26533f0b5
Successfully built 60f26533f0b5
Successfully tagged ghcr.io/ruffsl/navigation2:nightly

Use 'docker scan' to run Snyk tests against images to find vulnerabilities and learn how to fix them
```

nuclearsandwich, 2022-07-01:
> If you give me a URL for focal, I could also do it out myself.

If https://ci.ros2.org/job/ci_packaging_linux/514 completes successfully, it should give you a devel archive built on focal.

ruffsl, 2022-07-04:
@nuclearsandwich , I modified the ros2 nightly dockerfile to build from focal with your provided tarball like so:

https://github.com/osrf/docker_images/compare/master...ruffsl:docker_images:ros2-nightly-focal?expand=1 

But it looks like there is the package `libignition-math6-dev` has unmet dependencies:

<details>

```
~/git/docker_images/ros2/nightly/nightly$ docker build -t osrf/ros2:nightly-focal .
Sending build context to Docker daemon  8.704kB
Step 1/34 : FROM ubuntu:focal as setuper
 ---> 20fffa419e3a
Step 2/34 : ARG DEBIAN_FRONTEND=noninteractive
 ---> Using cache
 ---> 149e41018ec5
Step 3/34 : RUN echo 'Etc/UTC' > /etc/timezone &&     ln -s /usr/share/zoneinfo/Etc/UTC /etc/localtime &&     apt-get update &&     apt-get install -q -y --no-install-recommends tzdata &&     rm -rf /var/lib/apt/lists/*
 ---> Using cache
 ---> f69bf7dcb411
Step 4/34 : RUN apt-get update && apt-get install -q -y --no-install-recommends     bash-completion     cmake     dirmngr     git     gnupg2     libssl-dev     lsb-release     python3-pip     wget     && rm -rf /var/lib/apt/lists/*
 ---> Using cache
 ---> 01733b49847d
Step 5/34 : RUN echo "deb http://packages.ros.org/ros2-testing/ubuntu jammy main" > /etc/apt/sources.list.d/ros2-testing.list
 ---> Using cache
 ---> 1187ba86d263
Step 6/34 : RUN apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-keys C1CF6E31E6BADE8868B172B4F42ED6FBAB17C654
 ---> Using cache
 ---> de15f6fbedf3
Step 7/34 : RUN apt-get update && apt-get install --no-install-recommends -y     build-essential     git     python3-colcon-common-extensions     python3-colcon-mixin     python3-rosdep     python3-vcstool     && rm -rf /var/lib/apt/lists/*
 ---> Using cache
 ---> 1e7664a98e7e
Step 8/34 : ENV ROS_DISTRO rolling
 ---> Using cache
 ---> dcdad54c3bd9
Step 9/34 : ENV LANG C.UTF-8
 ---> Using cache
 ---> 964f2486fda5
Step 10/34 : ENV LC_ALL C.UTF-8
 ---> Using cache
 ---> 2540ecbe1897
Step 11/34 : ENV ROSDISTRO_INDEX_URL https://raw.githubusercontent.com/osrf/docker_images/master/ros2/nightly/nightly/index-v4.yaml
 ---> Using cache
 ---> 4a283c197d88
Step 12/34 : RUN pip3 install -U     argcomplete     flake8     flake8-blind-except     flake8-builtins     flake8-class-newline     flake8-comprehensions     flake8-deprecated     flake8-docstrings     flake8-import-order     flake8-quotes     pytest-repeat     pytest-rerunfailures
 ---> Using cache
 ---> f778a835fb8f
Step 13/34 : RUN pip3 freeze | grep pytest     && python3 -m pytest --version
 ---> Using cache
 ---> cee3d5ed26e3
Step 14/34 : RUN colcon mixin add default       https://raw.githubusercontent.com/colcon/colcon-mixin-repository/master/index.yaml &&     colcon mixin update &&     colcon metadata add default       https://raw.githubusercontent.com/colcon/colcon-metadata-repository/master/index.yaml &&     colcon metadata update
 ---> Using cache
 ---> 9428fbc16225
Step 15/34 : RUN rosdep init
 ---> Using cache
 ---> a4eb0dfe2fde
Step 16/34 : COPY prereqs.yaml /etc/ros/rosdep/
 ---> Using cache
 ---> c75d6587ceea
Step 17/34 : RUN echo "yaml file:///etc/ros/rosdep/prereqs.yaml" |     cat - /etc/ros/rosdep/sources.list.d/20-default.list > temp &&     mv temp /etc/ros/rosdep/sources.list.d/20-default.list
 ---> Using cache
 ---> 2b31ca754b25
Step 18/34 : RUN rosdep update
 ---> Using cache
 ---> e82982b08f2b
Step 19/34 : FROM setuper AS cacher
 ---> e82982b08f2b
Step 20/34 : ARG TIMESTAMP
 ---> Using cache
 ---> 4de5d271d3cb
Step 21/34 : RUN mkdir -p /opt/ros/$ROS_DISTRO
[... 2883 lines cut ...]
After this operation, 8204 kB of additional disk space will be used.
Get:1 http://archive.ubuntu.com/ubuntu focal/universe amd64 libqt5qml5 amd64 5.12.8-0ubuntu1 [1438 kB]
Get:2 http://archive.ubuntu.com/ubuntu focal/universe amd64 libshiboken2-py3-5.14 amd64 5.14.0-1~exp1ubuntu5 [151 kB]
Get:3 http://archive.ubuntu.com/ubuntu focal/universe amd64 libpyside2-py3-5.14 amd64 5.14.0-1~exp1ubuntu5 [74.0 kB]
Get:4 http://archive.ubuntu.com/ubuntu focal/universe amd64 libpyside2-dev amd64 5.14.0-1~exp1ubuntu5 [154 kB]
debconf: delaying package configuration, since apt-utils is not installed
Fetched 1817 kB in 1s (3210 kB/s)
Selecting previously unselected package libqt5qml5:amd64.
(Reading database ... 50412 files and directories currently installed.)
Preparing to unpack .../libqt5qml5_5.12.8-0ubuntu1_amd64.deb ...
Unpacking libqt5qml5:amd64 (5.12.8-0ubuntu1) ...
Selecting previously unselected package libshiboken2-py3-5.14.
Preparing to unpack .../libshiboken2-py3-5.14_5.14.0-1~exp1ubuntu5_amd64.deb ...
Unpacking libshiboken2-py3-5.14 (5.14.0-1~exp1ubuntu5) ...
Selecting previously unselected package libpyside2-py3-5.14.
Preparing to unpack .../libpyside2-py3-5.14_5.14.0-1~exp1ubuntu5_amd64.deb ...
Unpacking libpyside2-py3-5.14 (5.14.0-1~exp1ubuntu5) ...
Selecting previously unselected package libpyside2-dev.
Preparing to unpack .../libpyside2-dev_5.14.0-1~exp1ubuntu5_amd64.deb ...
Unpacking libpyside2-dev (5.14.0-1~exp1ubuntu5) ...
Setting up libshiboken2-py3-5.14 (5.14.0-1~exp1ubuntu5) ...
Setting up libqt5qml5:amd64 (5.12.8-0ubuntu1) ...
Setting up libpyside2-py3-5.14 (5.14.0-1~exp1ubuntu5) ...
Setting up libpyside2-dev (5.14.0-1~exp1ubuntu5) ...
Processi

SteveMacenski, 2022-11-03:
I did an audit, we're missing coverage in, most notably,
- Navigate Through Poses / Compute Path Through Poses
- Controller execution
- Waypoint Following
- Assisted Teleop

All of which have system tests associated with them

SteveMacenski, 2022-11-03:
Well that's neat-o

@ruffsl I found that if I removed composition (e.g. `use_composition: False`) then our coverage comes back! May it have something to do with composition? I also notice if we don't use composition, the tests do exit fully cleanly, so it could be from that too. 

https://app.codecov.io/gh/ros-planning/navigation2/pull/3266 shows 85.94% test coverage just swapping the WPF test over to that. I just updated https://github.com/ros-planning/navigation2/pull/3266 to migrate all tests over, lets see how it goes

ruffsl, 2022-11-03:
Whoa, that is trippy. Changes to the inner working of the node composting during the rolling migration could explain the coverage regression. I just wonder what exactly the change impacted. Could the way that the processes are exited be interfering with writing coverage files to disk, say for some nodes in the compose node but not for all others, when in enabled using the GCC coverage flags?

SteveMacenski, 2022-11-03:
I honestly don't know, I just finished confirming this is true by migrating all the tests over to it from my single-test trial. I'm not sure which it is (or some mixture of both). 

https://app.codecov.io/gh/ros-planning/navigation2/pull/3266 back up to the old regime at 88.6%. I think we should technically be higher than this, but that's still a major improvement

SteveMacenski, 2022-11-03:
I'm not sure what I should do about this ticket now. Broadly I'd say its complete, but I feel a little uneasy about not investigating root causes, though I have to move on with my week. I'll leave it for now, but may close at a later date if either of us get a chance to look at it more.

I will at some point need to look at why 2 of the behavior server plugins are low coverage, but those are also both new plugins since we had these issues so it wouldn't surprise me if there was actually just some issue in those tests that need to be changed and not representative of the same issues

nuclearsandwich, 2022-11-04:
> @nuclearsandwich I don't expect you to do anything about that - but have you heard about coverage metrics being lost with composition before, perhaps? Or if sigkill has to be invoked, does that mess with it (that might not be an question for you though)

Nope. I haven't heard but I'm going to pass that along in case anyone else sees some dots to connect.

SteveMacenski, 2022-11-07:
I'm going to open another ticket regarding the Assisted Teleop / Drive on Heading lack of testing.  I think this ticket is otherwise done (for now), though it would be nice to be able get coverage metrics with composition - its unclear if its us (not clean exit) or composition in general causing it.

closed 2022-11-07 by SteveMacenski