ros-navigation/navigation2 #3043: Feature Request: Feedback for Navigation Failures

karl-schulz, 2022-06-25:
## Summary

When reacting to navigation failures, it is essential to know *why* it failed to be able to react to it in a reasonable way. 

## Explanation

* From the outside of navigation, there is also no way of knowing the overall reason for a navigation failure (e.g. Planner or Controller failure, no progress, ...). Calling the NavigateToPose action, the resulting response is just a `std_msgs/Empty`. 
* Even inside the stack, planner's `ComputePathToPose` and controller's `FollowPath` provide no additional information than success/failure and a `std_msgs/Empty`.

Example where feedback could be useful: In the behavior trees, we often see recovery actions for planner/controller failures. This makes sense if the computation failed because the robot is stuck somehow and turning/waiting can resolve the issue. But if the planner failed because the goal is occupied, it makes no sense that the robot backs up or turns around as soon as it sees it. Different reactions depending on the cause for failure are needed.

## What can we do about it?

First of all: Do you agree that this is problematic?

I am not sure what the best way of dealing with this issue is and wanted to openly ask what you think about it. 

Some ideas: 

* Planner and controllers could loosely publish failure reasons on separate topics for anyone to react to it
* Alternatively, they could use the blackboard for this
* The action interfaces could have `error` fields

Still, I don't see a clean way of dealing with failure reasons in the behavior trees, the architecture doesn't seem to allow for it.

vinnnyr, 2022-06-27:
In terms of implementation, I wonder if some of this is a good application of diagnostics

mikeferguson, 2022-06-27:
Just a note, ROS diagnostics are usually meant to be "human readable" - the format is a bunch of strings, so it's not usually the best system if you want to autonomously make decisions based on the data.

mikeferguson, 2022-06-27:
I will also note - this feature (different recovery behaviors depending on the actual cause of failure) was a frequent request in ROS1 as well - we never actually got there.

SteveMacenski, 2022-06-27:
>  But if the planner failed because the goal is occupied, it makes no sense that the robot backs up or turns around as soon as it sees it. Different reactions depending on the cause for failure are needed.

That is all logic to be specified and customized in your behavior tree. Everyone's application is different so there's only so much I can do in the default BTs so that they work for most everyone's use-cases using generalized BT nodes. You could, and should, in that situation create some custom BT nodes for checking if the goal is occupied to take different corrective action if you don't believe a given default recovery is rational. Ex. the flow would be something like 'if can't create path, check if goal is occupied, if so, do XYZ` and you can have a variety of condition nodes for specifying actions. 

With that said, there is on-going work to improve the default BTs and I'm definitely open to suggestions! 

But thinking about this a bit more generally, we could give users some notion of what failed (e.g. which component crapped out: planner or controller) as a result type to return. Then that information of the action result type could be available in the BT blackboard for other condition nodes to access so that they don't need to derive that context themselves, when readily available from the planner itself. That's already 'something' but not a great deal of context -- although in practical terms planning only fails if (1) there's no valid path (2) the goal or starting points are occupied. Controllers fail when (1) its stuck (2) its in collision already.  

If we want more information than just what component failed, you're then reliant on planner plugins to providing you with that proper context about failure - which requires updating the planners to throw standard exceptions / failure codes that can be caught by the planner server and returned via the `result` interface. This is certainly possible, but would involve a bit of leg work. We would also want to update documentation to make it clear to planner plugin authors that they are expected to return specific codes / throw exceptions in these particular situations (but that's just updating a tutorial + some doxygen in `nav2_core`).

Diagnostics are not the right answer. We could always improve the Action result fields to contain more information, that's not much of a problem. What's more of a problem is deciding how to what / to populate it with and how so. We need to come up with an exhaustive (but later extendable) list of return codes we'd want to send on failures so that they're API-documentation-able and usable in the BT within their contexts. 

> Still, I don't see a clean way of dealing with failure reasons in the behavior trees, the architecture doesn't seem to allow for it.

I disagree with that - we've highlighted now 2 perfectly reasonable ways of doing this with BTs :wink:

vinnnyr, 2022-06-27:
ok I think I totally misunderstood this ticket to be for user knowledge, rather than programmatic recovery :)

SteveMacenski, 2022-07-06:
Thoughts?

karl-schulz, 2022-07-07:
Sorry, we are have a bit too much on our plates at the moment :) 

One detail question: If I understand you correctly, you would put the error codes/exceptions on the blackboard for planner and controller, probably clearing them when a planner/controller iteration succeeds or a new goal is given. This would mean also putting both of these exceptions in the action result, right? 

I generally agree with your points. I also think that it would not be a huge amount of work. However, I think we currently don't have the capacity to implement this (and the required tests etc.) and will resort to the existing BT condition nodes for the moment.

SteveMacenski, 2022-07-07:
Yes, I think the action results should have some return code field that remaps to specific failure modes that can be used on the blackboard by other nodes to know why a failure was triggered. Each time we get a new result from an algorithm that return code would be updated.

jwallace42, 2022-08-22:
I will take a crack at this tomorrow. It should be pretty straight forward.

madgrizzle, 2022-09-16:
This solution will only work with rolling, correct? i.e., they can't be implemented in galactic and humble because it changes message definitions?

SteveMacenski, 2022-09-16:
Correct

SteveMacenski, 2022-09-23:
Which servers here would we want to call this complete? Planner/Controller? Also Smoother, Waypoint, Behavior, BT Navigator?

The BT Navigator / Waypoint would be for returning total statuses beyond just "failure" to what specifically failed to an autonomy stack. So it could be that these are just pass-throughs from the task servers. 

The smoother / behavior server are the 2 other real task server candidates. I'm not sure we can specifically enumerate the failure modes of a behavior server with such variants as spin to assisted teleop. Behavior Server is more an aggregating server to share resources more than they are really related to any particular task.

Smoothers are pretty specific. Timeout, general failure, collision, could be some options, though none are particularly insightful.

SteveMacenski, 2022-09-28:
Planner codes merged!

SteveMacenski, 2022-10-14:
I think that's a valuable exercise. 

Spin: Invalid inputs or collision detected or TF

Backup / Drive Forward: Invalid inputs or collision detected or TF

AssistedTeleop: TF, perhaps add in progress checkers and such to see if anything is happening successfully?

Smooth: Path cannot be smoothed, smoothed path resulted in collision or off of map, smoother time exceeded, TF, invalid inputs(path too short, costmap not updated, etc), invalid algorithm (doesn't exist from request)

Wait: None

Also, it just came to mind that we don't have an error code yet in the planners / controllers for invalid algorithm request (e.g. `FollowPath2` doesn't exist). That might be another good one to add to the core servers in a follow up PR

jwallace42, 2022-10-20:
It looks like writing error codes for the could be the next step. 

> I think the WPF server would just be a pass through of the error codes from the planners / controller / smoother servers and maybe a few (or 1) WP specific ones like invalid request. I think BT Nav would be similar.

I agree. We just need to pipe the error codes up. 

> Also, it just came to mind that we don't have an error code yet in the planners / controllers for invalid algorithm request (e.g. FollowPath2 doesn't exist). That might be another good one to add to the core servers in a follow up PR
Yep!

SteveMacenski, 2022-10-31:
So remaining:
- [ ] smoother server
- [ ] Add invalid algorithm selection exception / error code
- [ ] WPF / NavToPose / NavThroughPose ways to indicate error code and offending server
- [ ] Do AssistedTeleop / spin / backup / forward? Questionable.

SteveMacenski, 2022-11-07:
@jwallace42 want to do the smoother after the Pipe error one so that we can button up all the major task servers? That would be enough for us to announce the new feature while buttoning up the other small things

jwallace42, 2022-11-07:
Sure. We have a rough list of the error codes already so it should be straight forward to get that in.

closed 2023-01-03 by SteveMacenski

annamannucci, 2023-09-08:
Hi @jwallace42, @SteveMacenski  and all,

I'd like to rise a discussion over the error codes of current planning plugins. 
In the current implementation of the navfn_planner, StartOccupied and GoalOccupied are raised only when the start or the goal pixel are in a lethal obstacle. Though, it would make sense in my opinion to raise this error also when the start or the goal are actually in an INSCRIBED_INFLATED_OBSTACLE.

I already implemented a small change towards this direction here: https://github.com/boschresearch/navigation2/commit/f626a7595207a99962b7ff03b0ba555e2ad51f8d

Though, I think it's valid to have this discussion in larger audience also because the proposed change is actually changing the semantics of the two error. Thus, this choice should affect all the others nav2 planners.

What do you think about it?

Best regards,
Anna Mannucci

SteveMacenski, 2023-09-08:
I think this is a good point. Can you open a PR and we can discuss there? The biggest items I'd ask is making sure you fix all instances of checking the start/goal in all the task servers that have that as an error code option. And probably break out that `else` statement region search into another function for readability.

PR #3251 (2022-11-18): Pipe error codes

---

## Basic Info

| Info | Please fill out this column |
| ------ | ----- |
| Ticket(s) this addresses   | #3043  |
| Primary OS tested on | Ubuntu |
| Robotic platform tested on | Gazebo |

---

## Description of contribution in a few bullet points

This pr pipes error codes through the behavior tree to the `compute path to pose` and `compute path through poses` to help close #3043 

## Description of documentation updates required from your changes

---

## Future work that may be required in bullet points

#### For Maintainers: 
- [ ] Check that any new parameters added  [... cut]
PR #3324 (2023-01-10): Behavior Tree uses Error Codes

---

## Basic Info

| Info | Please fill out this column |
| ------ | ----------- |
| Ticket(s) this addresses   | #3043 |
| Primary OS tested on | Ubuntu |
| Robotic platform tested on | Gazebo|

---

## Description of contribution in a few bullet points
Adds a new behavior tree condition node to use to error codes. This pr closes #3043 .

## Description of documentation updates required from your changes

---

## Future work that may be required in bullet points

#### For Maintainers: 
- [ ] Check that any new parameters added are updated in navigation.ros.org [... cut]