ros-navigation/navigation2 #3052: SMAC: collision checking for non-circle footprint type

andriimaistruk, 2022-06-29:
Hi. In the project I contribute, we faced an issue of SMAC planning too close to obstacles, and in `nav2_smac_planner/include/nav2_smac_planner/collision_checker.hpp` function `inCollision`, I wonder why the final check for circle vs polygon footprint type is different, meaning for non-circle type we check `footprint_cost_ >= OCCUPIED`, but for circle`footprint_cost_ >= INSCRIBED`? 
In ros-planning/navigation2:main in `nav2_smac_planner/include/nav2_smac_planner/collision_checker.hpp` function `inCollision` at line 116 for non-circle footprint, there is the check whether cost at robot's footprint (0,0) position equals to `INSCRIBED`, but for footprint cost along polygon lines the collision threshold is `OCCUPIED`. And the question is why it's done like this?

SteveMacenski, 2022-06-29:
Please ask on ROS Answers, happy to follow up there.

andriimaistruk, 2022-06-30:
https://answers.ros.org/question/403196/smac-collision-checking-for-non-circle-footprint-type/