ros-navigation/navigation2 #3055: Costmap obstacle layer do not work

RayOwwen, 2022-07-01:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04 LTS
- ROS2 Version:
  - humble
- Version or commit hash:
  - Commit: f87609d7eed2ccbd798a9e0c7dff5eeab042cce4
- DDS implementation:
  - eProsima’s Fast DDS

#### Steps to reproduce issue

1. Start simulation
```
ros2 launch nav2_bringup tb3_simulation_launch.py headless:=False
```

#### Expected behavior

#### Actual behavior
global_costmap and local_costmap obstacle layer do not work.
#### Additional information
Modify the obstacle_layer.cpp, comment the line 226, the costmap seems work well.
```cpp
    if (data_type == "LaserScan") {
      auto sub = std::make_shared<message_filters::Subscriber<sensor_msgs::msg::LaserScan,
          rclcpp_lifecycle::LifecycleNode>>(node, topic, custom_qos_profile, sub_opt);
          // sub->unsubscribe();

      auto filter = std::make_shared<tf2_ros::MessageFilter<sensor_msgs::msg::LaserScan>>(
        *sub, *tf_, global_frame_, 50,
```

SteveMacenski, 2022-07-01:
Indeed More info here https://discourse.ros.org/t/nav2-issues-with-humble-binaries-due-to-fast-dds-rmw-regression/26128/2 but the other ticket is where we're tracking it.

closed 2022-07-01 by SteveMacenski

marpeja, 2022-08-31:
Sorry for my ignorance, but how can you change from Fast DDS to Cyclone RMW?

padhupradheep, 2022-08-31:
check here: https://docs.ros.org/en/humble/Installation/DDS-Implementations/Working-with-Eclipse-CycloneDDS.html

> I am strugling with the same problem in ROS2 Galactic, and want to see if this solve the problem there too

Are you sure? We didn't experience any issues off late. Could you explain it a bit in detail?

marpeja, 2022-08-31:
I am using RTABMAP along with Nav2 and ROS2 Galactic. The robot uses a RGBD camera and I am using the depthimage_to_laserscan package to transform the images into a laserscan.

First of all it is worth to mention that the local costmap only appears if it is defined with an static_layer, like the global costmap. Otherwise it doesn't appear.

Moreover, you can see here an image of the problem. Once the map is mapped using RTABMAP, I introduce a rock in the middle of the cave, that, as you can see, it is detected by the laserscan and so by the camera. However, the local costmap doesn`t reflect this obstacle in the map.

[image]

I attach here the definition of the local costmap plugins.

```
plugins: ["static_layer", "obstacle_layer", "inflation_layer"]
      inflation_layer:
        plugin: "nav2_costmap_2d::InflationLayer"
        cost_scaling_factor: 3.0
        inflation_radius: 0.6
      obstacle_layer:
        plugin: "nav2_costmap_2d::ObstacleLayer"
        enabled: True
        observation_sources: scan
        scan:
          topic: /scan
          max_obstacle_height: 5.0
          min_obstacle_height: 0.0
          obstacle_max_range: 6.0
          obstacle_min_range: 0.0
          raytrace_max_range: 10.0
          raytrace_min_range: 0.0
          clearing: True
          marking: True
          data_type: "LaserScan"
      static_layer:
        map_subscribe_transient_local: True
      always_send_full_costmap: True
```

padhupradheep, 2022-08-31:
I understand, but this issue is not anyways related to the original issue posted in this forum. The original issue is somehow related to the FastRTPS middleware and yours I assume it somehow related to the configuration. 

I hope someone can help you at answers.ros.org

marpeja, 2022-08-31:
So you believe this isn't a problem related to FastRPS right?