ros-navigation/navigation2 #3059: Obstacle layer don't subscribe laser scans

PSotiriadis, 2022-07-04:
Hi
I want to report an issue with nav2_costmap_2d package. Using navigation2 with ROS2 Humble, local costmap is not indicated in RViz, like in screenshot below:

[image]

However, global costmap is working properly:

[image]

So I thought that problem can be in obstacle layer and I took a look in navigation2->nav2_costmap_2d->plugins->obstacle_layer.cpp. I realized, that in line 226, we unsubscribe the laser scan subscriber, which is subscribed above in lines 224,225. I comment line 226 and local costmap is working now fine for me and is indicated in RViz normally:

[image]

This is logical for me, but I don't know if it is desired from the authors of navigation2 (maybe it is desired to unsubscribe in this position, but forgot to subscribe later). I want only to highlight the problem!!!

PS: Same happens also later  in line 255, where PointCloud2 subscriber is unsubscribed, but I didn't check if works or not. Please see also this position, because it can also produce a problem.

padhupradheep, 2022-07-04:
Already reported https://github.com/ros-planning/navigation2/issues/3014 ! More info https://github.com/ros-planning/navigation2/pull/3018

SteveMacenski, 2022-07-04:
Indeed, working through it with Fast-DDS, but there’s the temp PR patch that can be applied to work with Fast-DDS if absolutely required, but I’d recommend just using Cyclone

closed 2022-07-04 by SteveMacenski

PSotiriadis, 2022-07-05:
First, @SteveMacenski & @padhupradheep thanks a lot for the very fast response and sorry for the repetition of this issue. I switch to cyclone dds and everything works!!!