ros-navigation/navigation2 #3061: Error while developing robot footprint using ros2 nav2

rafism1997, 2022-07-07:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- ROS2 Version:
  - Foxy
- 
-  

#### Steps to reproduce issue

```

```

#### Expected behavior
The robot footprint should be available after deploying the parameters file 

###ERROR 
[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2022-07-06-06-22-54-564770-63384ab5841b-3672
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [controller_server-1]: process started with pid [3676]
[INFO] [planner_server-2]: process started with pid [3678]
[INFO] [recoveries_server-3]: process started with pid [3680]
[INFO] [bt_navigator-4]: process started with pid [3682]
[INFO] [waypoint_follower-5]: process started with pid [3684]
[INFO] [lifecycle_manager-6]: process started with pid [3686]
[waypoint_follower-5] [INFO] [1657088574.921531558] [waypoint_follower]: 
[waypoint_follower-5] 	waypoint_follower lifecycle node launched. 
[waypoint_follower-5] 	Waiting on external lifecycle transitions to activate
[waypoint_follower-5] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[waypoint_follower-5] [INFO] [1657088574.921617962] [waypoint_follower]: Creating
[bt_navigator-4] [INFO] [1657088574.931056279] [bt_navigator]: 
[lifecycle_manager-6] [INFO] [1657088574.931359423] [lifecycle_manager_navigation]: Creating
[bt_navigator-4] 	bt_navigator lifecycle node launched. 
[bt_navigator-4] 	Waiting on external lifecycle transitions to activate
[controller_server-1] [INFO] [1657088574.932477012] [controller_server]: 
[controller_server-1] 	controller_server lifecycle node launched. 
[controller_server-1] 	Waiting on external lifecycle transitions to activate
[controller_server-1] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[bt_navigator-4] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[bt_navigator-4] [INFO] [1657088574.932942651] [bt_navigator]: Creating
[controller_server-1] [INFO] [1657088574.934428753] [controller_server]: Creating controller server
[recoveries_server-3] [INFO] [1657088574.935132150] [recoveries_server]: 
[recoveries_server-3] 	recoveries_server lifecycle node launched. 
[recoveries_server-3] 	Waiting on external lifecycle transitions to activate
[recoveries_server-3] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[planner_server-2] [INFO] [1657088574.940171519] [planner_server]: 
[planner_server-2] 	planner_server lifecycle node launched. 
[planner_server-2] 	Waiting on external lifecycle transitions to activate
[planner_server-2] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[planner_server-2] [INFO] [1657088574.941743335] [planner_server]: Creating
[lifecycle_manager-6] [INFO] [1657088574.944718922] [lifecycle_manager_navigation]: [34m[1mCreating and initializing lifecycle service clients[0m[0m
[lifecycle_manager-6] [INFO] [1657088574.953852855] [lifecycle_manager_navigation]: [34m[1mStarting managed nodes bringup...[0m[0m
[lifecycle_manager-6] [INFO] [1657088574.953895873] [lifecycle_manager_navigation]: [34m[1mConfiguring controller_server[0m[0m
[controller_server-1] [INFO] [1657088574.954774714] [local_costmap.local_costmap]: 
[controller_server-1] 	local_costmap lifecycle node launched. 
[controller_server-1] 	Waiting on external lifecycle transitions to activate
[controller_server-1] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[controller_server-1] [INFO] [1657088574.955069254] [local_costmap.local_costmap]: Creating Costmap
[planner_server-2] [INFO] [1657088574.961793289] [global_costmap.global_costmap]: 
[planner_server-2] 	global_costmap lifecycle node launched. 
[planner_server-2] 	Waiting on external lifecycle transitions to activate
[planner_server-2] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[planner_server-2] [INFO] [1657088574.962117442] [global_costmap.global_costmap]: Creating Costmap
[controller_server-1] [INFO] [1657088574.962561518] [controller_server]: Configuring controller interface
[controller_server-1] [INFO] [1657088574.962617566] [controller_server]: Controller frequency set to 20.0000Hz
[controller_server-1] [INFO] [1657088574.962630393] [local_costmap.local_costmap]: Configuring
[... 45 lines cut ...]
[recoveries_server-3] [INFO] [1657088575.047252438] [recoveries_server]: Configuring wait
[lifecycle_manager-6] [INFO] [1657088575.050751073] [lifecycle_manager_navigation]: [34m[1mConfiguring bt_navigator[0m[0m
[bt_navigator-4] [INFO] [1657088575.051007894] [bt_navigator]: Configuring
[bt_navigator-4] [ERROR] [1657088575.065807326] []: Caught exception in callback for transition 10
[bt_navigator-4] [ERROR] [1657088575.065840103] []: Original error: Could not load library: libnav2_compute_path_through_poses_action_bt_node.so: cannot open shared object file: No such file or directory
[bt_navigator-4] [WARN] [1657088575.065858199] []: Error occurred while doing error handling.
[bt_navigator-4] [FATAL] [1657088575.065863808] [bt_navigator]: Lifecycle node bt_navigator does not have error state implemented
[lifecycle_manager-6] [ERROR] [1657088575.066362202] [lifecycle_manager_navigation]: Failed to change state for node: bt_navigator
[lifecycle_manager-6] [ERROR] [1657088575.066391545] [lifecycle_manager_navigation]: Failed to bring up all requested nodes. Aborting bringup.
[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[lifecycle_manager-6] [INFO] [1657088590.384492738] [rclcpp]: signal_handler(signal_value=2)
[waypoint_follower-5] [INFO] [1657088590.384506752] [rclcpp]: signal_handler(signal_value=2)
[bt_navigator-4] [INFO] [1657088590.384518158] [rclcpp]: signal_handler(signal_value=2)
[recoveries_server-3] [INFO] [1657088590.384538126] [rclcpp]: signal_handler(signal_value=2)
[INFO] [recoveries_server-3]: process has finished cleanly [pid 3680]
[planner_server-2] [INFO] [1657088590.384543748] [rclcpp]: signal_handler(signal_value=2)
[controller_server-1] [INFO] [1657088590.384575762] [rclcpp]: signal_handler(signal_value=2)
[INFO] [controller_server-1]: process has finished cleanly [pid 3676]
[lifecycle_manager-6] [INFO] [1657088590.385227177] [lifecycle_manager_navigation]: Destroying
[waypoint_follower-5] [INFO] [1657088590.385462634] [waypoint_follower]: Destroying
[bt_navigator-4] [INFO] [1657088590.385737033] [bt_navigator]: Destroying
[INFO] [planner_server-2]: process has finished cleanly [pid 3678]
[INFO] [waypoint_follower-5]: process has finished cleanly [pid 3684]
[INFO] [lifecycle_manager-6]: process has finished cleanly [pid 3686]
[INFO] [bt_navigator-4]: process has finished cleanly [pid 3682]

SteveMacenski, 2022-07-07:
Start with the yaml file in the foxy branch, not the main branch. There are dozens of new things that are not available in Foxy in Main.

closed 2022-07-07 by SteveMacenski

rafism1997, 2022-07-08:
Thanks @SteveMacenski , but I have done the binary installation for the bringup and navigation package. Also, we tried cloning the repo from the foxy-devel branch, but the error still persists.

padhupradheep, 2022-07-08:
did you follow the steps shown here https://navigation.ros.org/getting_started/index.html ? and also used TB3 simulation as explained in the same page?

rafism1997, 2022-07-08:
@padhupradheep , we have developed a custom robot package by following this official link of ros2 nav2 : 

https://navigation.ros.org/setup_guides/footprint/setup_footprint.html 

Please do reopen the issue @SteveMacenski . We still have not figured what the problem is.

rafism1997, 2022-07-08:
@dheena2k2 , please see this issue for reference

SteveMacenski, 2022-07-08:
That cannot be true, because compute path through poses does not exist in https://github.com/ros-planning/navigation2/blob/foxy-devel/nav2_bt_navigator/src/bt_navigator.cpp#L41 or https://github.com/ros-planning/navigation2/blob/foxy-devel/nav2_bringup/bringup/params/nav2_params.yaml#L60. You've taken a parameter file or building Nav2 from a non-Foxy source that contains advanced features from more recent distributions.

Its been about a year since we've done a Foxy sync so I don't see how it is possible that you build from binaries, using the Foxy provided files as a base, and got the error you report. I think you've taken main, Galactic, or Humble branch parameters, at the very least. 

This error has nothing to do with footprints, please start from the base getting started example and I'm sure you'll find it works and you can see where your configuration errors have occurred.

rafism1997, 2022-07-09:
Hi @SteveMacenski , I understand that this problem has nothing to do with footprints. Once we  completed implementing the slam-toolbox functionality and after setting up the sensors, we moved on to develop the robot footprint where we added the yaml file . 

@padhupradheep , the yaml file that we used is this : https://github.com/ros-planning/navigation2_tutorials/blob/master/sam_bot_description/config/nav2_params.yaml .

rafism1997, 2022-07-09:
We even tried with the official sam-bot description which was having the same issues (when we added a custom navigation parameter file).

dheena2k2, 2022-07-11:
We get the same error after using the parameter files from https://github.com/ros-planning/navigation2/blob/foxy-devel/nav2_bringup/bringup/params/nav2_params.yaml

```code
Original error: Could not load library: libnav2_compute_path_through_poses_action_bt_node.so: cannot open shared object file: No such file or directory
```

As the error shows, we noticed that the library ```libnav2_compute_path_through_poses_action_bt_node.so``` is the exact same mentioned in the https://github.com/ros-planning/navigation2/blob/foxy-devel/nav2_bringup/bringup/params/nav2_params.yaml#L61, ```nav2_compute_path_to_pose_action_bt_node``` just with a prefix ```lib```.

padhupradheep, 2022-07-11:
> libnav2_compute_path_through_poses_action_bt_node

This bt node is not included in the https://github.com/ros-planning/navigation2/blob/60043d4067b9dbff73be65291f5df6aaa286cc61/nav2_bringup/bringup/params/nav2_params.yaml#L60 ! 

Are you sure that you are updating the params in the right yaml file ? 

Note that, you need to rebuild your packages after changing the yaml files, if you have not used `colcon build --symlink-install` for building it. 

Also please see that, nav2_compute_path_through_poses_action_bt_node  is different than that of the nav2_compute_path_to_poses_action_bt_node