ros-navigation/navigation2 #3062: Install navigation2 for ROS2 Humble on Ubuntu 20.04

martfra, 2022-07-07:
Hi,
I want to try out navigation2 on ROS2 Humble, but I currently cannot update to Ubuntu 22.04. Instead, I have to use Ubuntu 20.04, so I installed ROS2 Humble from source and then tried to install navigation2 from source too. Unfortunately, I encountered the following issue. Can anyone help me?

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- ROS2 Version:
  - Humble, installed from source
- Commit hash of navigation2:
  - c359d38707715516faba6d2ba71f72a0d1d5792e
- DDS implementation:
  - Cyclone DDS

#### Steps to reproduce issue
I installed ROS2 Humble on Ubuntu 20.04 according to the instructions on https://docs.ros.org/en/humble/Installation/Alternatives/Ubuntu-Development-Setup.html.

Then I tried to install navigation2 for ROS2 Humble with these commands:
```
source ~/ros2_humble/install/setup.bash
cd ~/ros/nav2_ws/src
git clone https://github.com/ros-planning/navigation2.git
cd navigation2
git checkout humble
cd ~/ros/nav2_ws
rosdep install -y -r -q --from-paths src --ignore-src --rosdistro humble
```

#### Expected behavior
All dependencies are installed.

#### Actual behavior
```
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
dwb_plugins: No definition of [angles] for OS version [focal]
nav2_behavior_tree: No definition of [behaviortree_cpp_v3] for OS version [focal]
nav2_util: No definition of [bond] for OS version [focal]
nav2_lifecycle_manager: No definition of [diagnostic_updater] for OS version [focal]
nav2_controller: No definition of [angles] for OS version [focal]
nav2_costmap_2d: No definition of [angles] for OS version [focal]
nav2_bringup: No definition of [slam_toolbox] for OS version [focal]
nav2_system_tests: No definition of [gazebo_ros_pkgs] for OS version [focal]
nav2_smoother: No definition of [angles] for OS version [focal]
nav2_constrained_smoother: No definition of [angles] for OS version [focal]
nav2_rotation_shim_controller: No definition of [angles] for OS version [focal]
nav2_waypoint_follower: No definition of [cv_bridge] for OS version [focal]
dwb_critics: No definition of [angles] for OS version [focal]
nav2_smac_planner: No definition of [angles] for OS version [focal]
nav2_bt_navigator: No definition of [behaviortree_cpp_v3] for OS version [focal]
Continuing to install resolvable dependencies...
#All required rosdeps installed successfully
```

#### Additional information
When I try to build navigation2 with
```
colcon build --symlink-install
```
I get the following output:
```
Starting >>> nav2_common
Starting >>> nav_2d_msgs
Finished <<< nav2_common [4.34s]                                                          
Starting >>> nav2_msgs
Starting >>> nav2_voxel_grid
Finished <<< nav2_voxel_grid [12.5s]                                                                                              
Finished <<< nav_2d_msgs [20.8s]                                                              
Starting >>> dwb_msgs
Finished <<< dwb_msgs [28.4s]                                                                 
Finished <<< nav2_msgs [46.4s]                     
Starting >>> nav2_util
Starting >>> nav2_simple_commander
Finished <<< nav2_simple_commander [1.86s]                                                 
--- stderr: nav2_util                           
CMake Error at CMakeLists.txt:17 (find_package):
  By not providing "Findbondcpp.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "bondcpp", but
  CMake did not find one.

  Could not find a package configuration file provided by "bondcpp" with any
  of the following names:

    bondcppConfig.cmake
    bondcpp-config.cmake
```

Aposhian, 2022-07-07:
As an alternative, you could also try installing Nav2 in a container. Here is an example for how to do that: https://github.com/fireflyautomatix/nav2-compose/tree/main

SteveMacenski, 2022-07-07:
Rosdep will not work on out of REP2000 operating systems. You will need to build all of those dependencies yourself if you want to use Humble on 20.04, or use docker as Adam suggests if you don’t have to run on bare metal. 

You can find a list of the dependencies in tools/underlay.repos. That contains our non-default ROS 2 packages required to build Nav2

closed 2022-07-07 by SteveMacenski