ros-navigation/navigation2 #3063: Footprint polygon with another moving polygon togheter.

Marcelomm103, 2022-07-07:
## Feature Request

I know that you can make n polygons in a footprint but they move as they were one. I suggest make a footprint with more than one polygon moving like they where togheter with a joint. Per example a truck with its front in one footprint and its back cargo in another footprint connected by a point (joint) ‘following’ the front footprint without colliding with it.

I can try to implement it but probably gonna need some help and assistence.

SteveMacenski, 2022-07-07:
That sounds like a good idea, however it should be possible to effectively do this now if you connect the 2 polygons via a line segment (e.g. polygon 1 has a node that draws a line to polygon 2, then at the end of polygon 2, it draws an edge back to the node in polygon 1) that way it is all represented by a single polygon. Creating multiple polygons for collision detection would involve some relatively significant reworking inside costmap 2D and all of the algorithms' collision checkers.

Either way, you'll require some node with the current state estimate of the following polygon to draw it, so it should be pretty easy to make a node at the back of the cab that links to the front of the trailer for such a situation. That seems like the path of least resistance, though if you were willing to contribute support for multiple polygons, I'd be happy to chat about design requirements and review PRs if you'd be open to implementing it!

Marcelomm103, 2022-07-08:
Thank you for your answer. I think i would go with the path of least resistance and make the node for the other polygon. I actualy didnt express it well, the two polygons would overlap each other a little bit since the cab goes a little bit above the front of the trailer. But i didnt understand how can i create another node with another footprint, can you explain it to me ? For what i could understand it would involve setting another 'base_frame' in a position in polygon 2 so it could detect the obstacles, but how could i link those two? With a static tf2 transform ?

SteveMacenski, 2022-07-08:
As so

[image]

You can see the node added in the back of the cab connecting to the trailer, you just need to make sure to close the polygon by ending the footprint with going back through that line segment. 

But really, you shouldn't want things to be in between the cab and the trailer, so you could just create an L-shape where the angle changes by the state estimate.

closed 2022-07-08 by SteveMacenski

Marcelomm103, 2022-07-08:
I think you missunderstood what i'm trying to do. Maybe this image will help it clear it.

[image]

The cab part will overlap the front part since it's connected above it. And i want to make the two polygons move separately but connected trought that red dot like a joint. So the footprint of the cab also can be used by the obstacle detection. Can you explain to me how i can make this angle change by state estimate? I dont have any sensors on the cab itself to know its location.

SteveMacenski, 2022-07-08:
How are you able to know the state of the angle of the joint? You obviously need to provide that for the footprint to know what it is. 

The same logic stands for the polygon, adding the line segment to connect them, it would slow down checks very neglibably.

Marcelomm103, 2022-07-08:
Thank you kindly for your insights. I'll give it more thought about it and try to come up with a solution. The state off the angle of the joint would be estimated by the dynamics of the front as it moves. But just so i understand, it is possible to make them move differently just by know the angle of the joint? and how could i do that?

Marcelomm103, 2022-07-15:
Hi Steve, hope your doing well.

If you could explain to me how i could make two footprints, and how to make them move differently by the angle of the joint ( which i have ) i would be very grateful.

jardeldyonisio, 2025-08-31:
> That sounds like a good idea, however it should be possible to effectively do this now if you connect the 2 polygons via a line segment (e.g. polygon 1 has a node that draws a line to polygon 2, then at the end of polygon 2, it draws an edge back to the node in polygon 1) that way it is all represented by a single polygon. Creating multiple polygons for collision detection would involve some relatively significant reworking inside costmap 2D and all of the algorithms' collision checkers.
> 
> We march through the edges of the polygon with a line iterator to check for collision, so having disconnected areas would actually be perfectly fine (unlike if we did collision detection via BFS of cells inside of the polygon).
> 
> Either way, you'll require some node with the current state estimate of the following polygon to draw it, so it should be pretty easy to make a node at the back of the cab that links to the front of the trailer for such a situation. That seems like the path of least resistance, though if you were willing to contribute support for multiple polygons, I'd be happy to chat about design requirements and review PRs if you'd be open to implementing it! It would be relatively straight forward, just tedious

@SteveMacenski Nowadays is it possible to define two footprints in the Nav2 stack, or is the cargo footprint always updated based only on the truck’s base link? Can this be done using the native Nav2 implementation, or would I need to create custom extensions?