ros-navigation/navigation2 #3067: Use Parameter Descriptors

Aposhian, 2022-07-08:
## Feature request

#### Feature description
Parameters can be declared with a [rcl_interfaces::msg::ParameterDescriptor](https://github.com/ros2/rcl_interfaces/blob/rolling/rcl_interfaces/msg/ParameterDescriptor.msg) that embeds information like the description or even bounds checking on parameters.

#### Implementation considerations
This would probably involve rewriting some nav2 utils for declaring parameters.

SteveMacenski, 2022-07-09:
Realistically speaking, unless there are user contributions in that area, that is probably not something the maintainers are going to invest the time into given the time we've spent developing and maintaining the documentation on navigation.ros.org. Its definitely not a bad enhancement, but this feels like something that wouldn't actually happen unless we had an intern or student working with us for a couple of months.

SteveMacenski, 2022-08-22:
I'm going to close this as a wontfix, I don't think realistically anyone is going to be able to justify time to work on this and we have alternative documentation containing all of this information.

closed 2022-08-22 by SteveMacenski