ros-navigation/navigation2 #3074: Add service to BT Navigator to modify or introspect the blackboard

SteveMacenski, 2022-07-13:
E.g. have a service that can change values in the blackboard and/or request a current value to be returned

richardw347, 2022-07-25:
Hey @SteveMacenski just had a quick look at this today and got a quick service going to get and set a string value in the blackboard.

Just wanted to get some advice on the use case the blackboard is pretty extensible in terms of user supported types and I'm just getting started with Nav2 and Behavior Trees so I'm not sure what the use cases are.

Would just a few basic services for get/set float/int/bool/string suffice or something a bit more fancy/dynamic?

stevenbrills, 2022-07-26:
I had a similar question too.  Given that blackboard variables are of significantly varying types would we need different services to probe each one of them?  Or is there an elegant way to probe and return all types with just a single service?

richardw347, 2022-07-26:
One way to handle multiple types in the same service is cast the value from a string, the BT library does this with the [convertFromString](https://github.com/BehaviorTree/BehaviorTree.CPP/blob/725eba7e0abbc704284dec393706e5404f1472e3/include/behaviortree_cpp_v3/basic_types.h#L61) function. 

This would work with int/float/long/double/string etc but anything more complex like a list of values e.g. a pose would be too messy. I'm not sure I even like the string cast method it does have lots of potential to silently fail if the cast succeeds but doesn't yield a meaningful value. 

Would appreciate anyone else's 2 cents?

stevenbrills, 2022-07-26:
Right, the convertFromString is what I'm using right now for some of the simpler types.  But my end goal was to probe things like pose.  Since the blackboard is used to share data like the goal and even the path between the planner and controller it would be great to intercept these complex data types for debugging.

richardw347, 2022-07-29:
Ok if the main use case is diagnostic maybe the most effective solution would be to have a blackboard publisher which could just publish the state of all keys in the blackboard, that should cover the case of just needing to introspect values. For actually injecting values we can have a smaller set of targeted services for insert/overwrite of values. 

@stevenbrills how would that suit you?

SteveMacenski, 2022-08-19:
This is a very fair point, I'm not actually sure this is realistic to add considering that. A few potential solutions:
- Have the service only operate on strings and have it be required that the client to the service convert any type into a string using the convertToString functions from the nav2_behavior_tree package so that the BTs can read the inputs. A bit hacky, but I think probably the most practical, though it makes us have to have `convertToString` for all possible types, even user specific BT node types that might not exist yet in our centralized codebase
- Have a series of services for each type, after auditing all key types in all BT nodes / trees in Nav2 +  the standard types like double/int/etc that might not be in there right now. Kind of has the same issues as previously, but more immediately visible from having N interfaces rather than forcing conversion functions to exist. 
- Create a series of BT nodes that implement the services that can be added to a tree for introspection of various types, so its on the user to include the proper nodes into the tree to allow for it. Pros: users can create their own for their own types in the future so there's no required centralized index of string conversion functions or services. Cons: forces users to include a bunch of BT nodes to introspect, making their BTs a little messy + now having N services by type again 
- Basically do the same thing as the BT nodes, but rather than being nodes in the tree, they're plugins in the BT navigator with access to the blackboard

I don't really love any of these solutions.

SteveMacenski, 2022-08-19:
I'm interested if @facontidavide has any thoughts on this as the Behavior Tree guru. I'm sure this is something itching in the back of his mind.

SteveMacenski, 2022-09-16:
Not really -- I'm thinking more and more about closing this as `wontfix` given the headaches. 

I think it would be sensible to have a (set of?) BT nodes for exposing particular ports out. So instead of having a global service, we have a templated BT base node that would take in a message type template and create services to get / set. Then users can create implementations of it for each type of interest they have and insert it into their BT trees. 

Example pseudo-code:

```
template<typename MsgT> 
class BTBlackboardNode
{
public:
  ... usual stuff

  service_get_cb() {blackboard->get<MsgT>("msg_id")}
  service_set_cb(MsgT & val) {blackboard->set<MsgT>("msg_id", val)}

protected:
  service_get
  service_set
}
```

But that still requires custom service message types for each type. So that's where it gets to a point of ballooning code. But if we just provide the base class and make it easy for it to be expanded, then we can make it other people's problems if they have more complex things they'd like to add later and add their own services. It would be sensible to have a new package `behaviortree_msgs` that lives outside of this repo with implementation service messages for the basic types of interest. If someone wants to add more over time, they're welcome to, but that would let us implement the BT node implementations for the types here.

facontidavide, 2022-09-16:
spoiler alert: in BehaviorTree 4 I am probably introducing blackboard serialization for easy publishing. That can be backported to 3.X.

This is something you might be interested to. As usually, it will be orthogonal to ROS

SteveMacenski, 2022-09-16:
I'd also be happy to move to BT v4 if/when you feel its ready

Interesting, still leaves a bit of a problem of getting message types back out for practical use. Publishing a service for serialized result and leaving it to the user to "figure it out" to convert into a message or interpret that buffer seems a little hacky to me from a Nav2 perspective, but totally sensible for logging for BT.CPP. I'd still want to be able to convert back to real-types before giving it back to a user

SteveMacenski, 2022-10-13:
@facontidavide thoughts about the `behavior_tree_msgs` package in your BT.CPP organization? I think your openness to that would be the make-or-break if my suggested plan of attack is worth going down

facontidavide, 2022-10-13:
As I said, monitoring the state on the blackboard remotely is on my roadmap. 
I am happy to brainstorm about it and to add it to the official BT.ROS / BT.ROS2 repositories.

Note that BT.CPP is ROS independent and will use JSON or flatbuffer.

SteveMacenski, 2022-10-14:
OK! Closing then :-)

closed 2022-10-14 by SteveMacenski

tonynajjar, 2023-09-18:
We found ourselves trying to implement this on our own, was there any progress made in this issue, either on Nav2 of BT.CPP side?