ros-navigation/navigation2 #3075: PlannerServer::computePlanThroughPoses will eventually fail when given a goal with a non-zero timestamp.

Ronoman, 2022-07-13:
## Bug report

- Operating System: Ubuntu 20.04
- ROS2 Version: Galactic binaries
- Version or commit hash: Galactic 1.0.12-1focal.20220526.142755
- DDS implementation: Cyclone DDS (Galactic default)

#### Steps to reproduce issue
Send a goal using RViz 2D Nav Goal (or send a single goal that has a non-zero timestamp in its header) to a running Nav2 stack.

#### Expected behavior
Regardless of the amount of time the robot will spend traveling to the goal, the goal pose should remain valid.

#### Actual behavior
If the path takes longer to complete than navigation2/costmap_2d_ros's internal tf buffer, path planning will fail when the `planner_server` attempts to transform the goal pose into the correct frame when it tries to `computePlanThroughPoses` (which will call `costmap_2d_ros::transformPoseToGlobalFrame`).

#### Additional information
It is possible that this is not a bug, and instead desired behavior. If this is intentional, what is the use case for a goal with a timestamp that will eventually fall outside of the tf buffer?

SteveMacenski, 2022-07-14:
Is this all being done in map frame? If the frame of the request is the same as that of the costmap, it should simply bypass https://github.com/ros-planning/navigation2/blob/30048b46bf02c6a60473b1dae6000e6eb2597056/nav2_costmap_2d/src/costmap_2d_ros.cpp#L588.

So an easy option that if you're not using map frame, is to convert your request to map frame (or change the global frame of the costmap to a different frame, I suppose)

Ronoman, 2022-07-14:
> Is this all being done in map frame?

In my application, its easy enough for me to provide the goal in the map frame instead of a different one. However, I can imagine a case where a Nav2 user/developer would (as an example) want to provide a goal relative to the robot. They may then expect Nav2 to lock in that target goal at the beginning of path planning/following, instead of constantly transforming that goal back into the map frame (which would keep moving the goal). This could be accomplished outside of Nav2 by pre-transforming the goal before sending it to Nav2

Regardless of whether the above example is behavior that should be supported, I'd argue that passing in a pose with a fixed timestamp in a non-map frame should at least inform the developer (through a log message perhaps) that the goal will be continuously transformed. This functionality is different from how move_base in ROS1 navigation would handle it (see [move_base.cpp:658](https://github.com/ros-planning/navigation/blob/noetic-devel/move_base/src/move_base.cpp#L658), where it looks like the goal is transformed once at the beginning, and only transformed further if the frame id changes due to a new goal being received.

closed 2022-07-18 by SteveMacenski