ros-navigation/navigation2 #3078: Issue loading saved map when saved under a subdirectory

nakai-robotics, 2022-07-17:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04
- ROS2 Version:
  - Humble
- Version or commit hash:
  - ros-humble-navigation2  1.1.0-2jammy.20220607.182532
- DDS implementation:
  - Fast-RTPS

#### Steps to reproduce issue
Call the map saver server with a map_url that includes a subdirectory: 
```
ros2 service call /map_saver/save_map nav2_msgs/srv/SaveMap "{map_topic: map, map_url: maps/nakai_world, image_format: pgm, map_mode: trinary, free_thresh: 0.25, occupied_thresh: 0.65}"
```

#### Expected behavior
Output yaml file image file will have a relative path to the yaml file, and not to current working directory. Conversely, when loading the map, map_server should not append the yaml file's path to the image path, rather it should append the currect working directory to the image path. 

#### Actual behavior
Image path inside the map's yaml file, is relative to current working directory and not to the yaml file. 
```
image: maps/nakai_world.pgm 
mode: trinary
resolution: 0.05
origin: [-9.39e-05, -0.68, 0]
negate: 0
occupied_thresh: 0.65
free_thresh: 0.25
```

#### Additional information
When trying to later load the map using map_server, getting the following error: 
```
[ERROR] [map_io]: Failed to load image file ./maps/maps/nakai_world.pgm for reason: Magick: Unable to open file (./maps/maps/nakai_world.pgm) reported by magick/blob.c:3089 (OpenBlob)
```
All commands were ran under the same working directory.

SteveMacenski, 2022-07-18:
@AlexeyMerzlyakov please add this to your queue.

> Output yaml file image file will have a relative path to the yaml file, and not to current working directory.

The actual command is going to be relative to your requested location. Right, if you give a relative location, it'll be relative to where you called it. So assuming when you call `SaveMap` with a relative path for the yaml of `./my_relative/path`, you're saying that the YAML is correct, yes? Its just the image file that is wrong? 

> Conversely, when loading the map, map_server should not append the yaml file's path to the image path, rather it should append the currect working directory to the image path.

It sounds like the `image: maps/nakai_world.pgm` needs to just me `nakai_world.pgm` and the file itself is stored in the relative directory `maps`. Is that what you're looking for? I agree that that would be sensible. 

Please be very particular, when exact commands, directories executed, and output file locations / contents describe your problem so we can reproduce a minimal case and resolve it, where appropriate.

nakai-omer, 2022-07-19:
First of all, thanks for the quick reply.

Directory structure:
- src
  - maps
    - nakai_world.pgm
    - nakai_world.yaml

Current working directory is `src`.

Inside of `src` we run the following command:
```
ros2 service call /map_saver/save_map nav2_msgs/srv/SaveMap "{map_topic: map, map_url: maps/nakai_world, image_format: pgm, map_mode: trinary, free_thresh: 0.25, occupied_thresh: 0.65}"
```
Result, files are created as expected under the `maps` folder in the structure described above.

YAML contains: 
```
image: maps/nakai_world.pgm 
mode: trinary
resolution: 0.05
origin: [-9.39e-05, -0.68, 0]
negate: 0
occupied_thresh: 0.65
free_thresh: 0.25
```
Next, still under `src` directory we load a "map_server" node with the following config:
```
yaml_filename: "maps/nakai_world.yaml"
``` 
Result, `map_server` successfully loads `nakai_world.yaml`, but fails to load the `nakai_world.pgm`
```
[ERROR] [map_io]: Failed to load image file ./maps/maps/nakai_world.pgm for reason: Magick: Unable to open file (./maps/maps/nakai_world.pgm) reported by magick/blob.c:3089 (OpenBlob)
```
It could be solved in (at least) two ways:
1) Inside the YAML file, `map_saver` will make the path to the map image relative to the location of the YAML file. In our case `./nakai_world.pgm` or just `nakai_world.pgm`, instead of `maps/nakai_world.pgm`. 
2) `map_server` will look for the image without prefixing the YAML file's directory. In our case, instead of `maps/maps/nakai_world.pgm`, it should be just `maps/nakai_world.pgm`

SteveMacenski, 2022-07-19:
Makes sense. I think (1) is the easiest, basically just strip out any relative paths when saving the yaml's `image` field. If we always save them side-by-side anyway, then there's no reason for any prefix to exist there. That should be a relatively easy fix.

closed 2022-07-21 by SteveMacenski