ros-navigation/navigation2 #3081: Regulated pure pursuit oscillation at goal

automech-rb, 2022-07-20:
## Feature request

#### Feature description

Hi,
When setting low **lookahead_dist** values for robot with low de-acceleration capability which is able to move at high speed leads to oscillation when arriving at goal.
I believe this is due late checkup for goal  slowdown (at low **lookahead_dist**) and robot is not able to stop quickly leading to overshoots.

#### Implementation considerations

Having a parameter for de-acceleration of robot with appropriate kinematic modeling.

Aposhian, 2022-07-20:
Related: #3045 and #2815

SteveMacenski, 2022-07-20:
What timing, I swear, more related issues are filed in roughly the same timeframe than seems statistically possible. This must be evidence of a deterministic consciousness :laughing: 

#3045's solution, #3047 would actually fully solve that problem for you, it decouples the lookahead distances from the slow-down criteria so you can slow further away even when using a short lookahead distance. 

I think the only thing blocking #3047 is documentation & default. So its right around the corner. I'm going to go ahead and close this issue since its covered by #3045 and the tickets are now linked, but I 100% appreciate the filed ticket @automech-rb

closed 2022-07-20 by SteveMacenski