ros-navigation/navigation2 #3095: 3d navigation stack for drones

stevedanomodolor, 2022-07-26:
## Feature request
I would love to see an implementation of 3d navigation for drones in the navigation stack.  It would be nice to discuss with @SteveMacenski as mentioned in the link `https://answers.ros.org/question/341594/navigation2-support-for-3d-navigation-drones/` to see the possibilities.

jwallace42, 2022-07-31:
I would join the slack https://navigation.ros.org/about/index.html?highlight=slack.

Stevedan-robotics, 2022-08-03:
I am not able to join the slack with my personal email, because i need an invitation @SteveMacenski. @jwallace42 can you assist me with that

stevedanomodolor, 2022-08-03:
I cannot seem to enter the slack group with my personal email, i require a special invitation, @SteveMacenski. @jwallace42 can you help me with that

jwallace42, 2022-08-03:
Sure, what is your email?

Stevedan-robotics, 2022-08-03:
stevedan.robotic@gmail.com

stevedanomodolor, 2022-08-03:
stevedan.o.omodolor@gmail.com

jwallace42, 2022-08-03:
Just send you a invite. Let me know if that works.

stevedanomodolor, 2022-08-03:
Received thank you

closed 2022-08-03 by stevedanomodolor

yrik, 2023-07-09:
how it's going with drone/plane nav support?

songanz, 2023-07-13:
Also want follow-ups here

stevedanomodolor, 2023-07-13:
My response when i asked was that there wasn't a specific plan to implement this in the near future although it is possible to contribute to nav2 because the base architecture of nav2 can support this. Another alternative is px4