ros-navigation/navigation2 #3096: Shim controller at Humble (feat. turtlebot3)

WonsokYoo, 2022-07-27:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04 docker container (on Ubuntu 18.04 host PC)
- ROS2 Version:
  - Humble
- Version or commit hash:
  - from source (this repo. 'humble' branch)
- DDS implementation:
  - Cyclone DDS (Also was same phenomenon with Fast RTPS DDS)

#### Steps to reproduce issue
- I used three repos.
1. git clone -b humble https://github.com/ros-planning/navigation2.git
2. git clone -b humble-devel https://github.com/ROBOTIS-GIT/turtlebot3.git
3. git clone -b humble-devel https://github.com/ROBOTIS-GIT/turtlebot3_simulations.git

- I tried navigation with following commands:
$ ros2 launch turtlebot3_gazebo turtlebot3_world.launch.py
$ ros2 launch turtlebot3_navigation2 navigation2.launch.py use_sim_time:=True map:=$HOME/map.yaml

※ I switched the controller to 'Shim Controller' following: 
     https://navigation.ros.org/configuration/packages/configuring-rotation-shim-controller.html

#### Additional information

- Issue
  After Initializing the pose in rviz, I used 'Nav2 Goal' button on rviz and I can see following messages on terminal:

[rviz2-2] Start navigation
[rviz2-2] [ERROR] [1658900836.738638941] [rviz2]: navigate_to_pose action server is not available. Is the initial pose set?

  What am I missing? I modified to cyclone DDS from default Fast RTPS DDS whether to check it's a DDS issue or not but it wasn't :(

padhupradheep, 2022-07-27:
what do you see in RViz ? map, robot ? or just a black screen ? 

If blank, then I guess the map is not loaded at all!

WonsokYoo, 2022-07-27:
@padhupradheep Yeah, I can see the map and I initialized robot pose. The above issue is occuring after I make a goal pose.

padhupradheep, 2022-08-01:
I think, you just have to run 

`ros2 launch nav2_bringup tb3_simulation_launch.py headless:=False`

for bringing up and running both the simulation and navigation.

WonsokYoo, 2022-08-08:
@padhupradheep Sorry for late reply. I think your answer is not the point of this issue. I already read above docs that you mentioned. By any chance, I tried with your headless option and when I switch the 'controller_server' from DWB to ShimController the same issue occurs. 

Gazebo sims are working well, there's no issue when I use DWB. The above 'navigate_to_pose action server is not available. Is the initial pose set?' message is shown only when I tried to use ShimController and this is the point of this issue.

WonsokYoo, 2022-08-08:
Finally, I figured out why it wans't works. 
The docs : https://navigation.ros.org/configuration/packages/configuring-rotation-shim-controller.html
has an error.

goal_checker_plugins: "goal_checker"    →  goal_checker_plugins: ["goal_checker"]

After I modifying like above, it works now..

SteveMacenski, 2022-08-18:
https://github.com/ros-planning/navigation.ros.org/blob/master/configuration/packages/configuring-rotation-shim-controller.rst fixed, thanks for the report!

closed 2022-08-18 by SteveMacenski