ros-navigation/navigation2 #3101: Extrapolation Error: Lookup would require extrapolation into the future

miku54, 2022-08-04:
- Operating System:
 Ubuntu20.04
- ROS2 Version:
  Galactic

#### Steps to reproduce issue
When I use gmapping to build a map, use that map to navigate, everything works fine when I run nav2, In rviz, I give the robot a 2D Nav Goal. Immediately, I get the following error:
  Failed to transform initial pose in time (Lookup would require extrapolation into the future. Requested time 1659339558.513968 but the latest data is at time 1659339558.496260, when looking up transform from frame [base_footprint] to frame [odom_combined])
[recoveries_server-15] [ERROR] [1659339575.919917542] [transformPoseInTargetFrame]: No Transform available Error looking up target frame: "odom" passed to lookupTransform argument target_frame does not exist.
My navigation parameters can be seen here
https://github.com/ros-planning/navigation2/issues/3012

padhupradheep, 2022-08-10:
Are you sure, if the "odom" frame is being broadcasted?

miku54, 2022-08-15:
Hi, thanks for the reply!
I'm pretty sure I'm not broadcasting the coordinates of the odom, including the topic of not remapped odom, my odometer data is always odom_combined.

> Are you sure, if the "odom" frame is being broadcasted?

padhupradheep, 2022-08-15:
No worries, you can read the docs on how to proceed further. https://navigation.ros.org/setup_guides/index.html

SteveMacenski, 2022-08-16:
It looks like you need to configure (or modify) gmapping to forward date the timestamps a bit more. The global localization system needs to “lead” the clock slightly so that transformations for “now” always have that element of the tree represented. We do this in slam toolbox using the transform tolerance param and was done so in ROS 1’s gmapping. 

Since there is no formally supported version of gmapping in ROS 2, I can’t comment on the quality of the port to having done these more subtle timing elements correctly. That would be something to address in that project.

closed 2022-08-16 by SteveMacenski