ros-navigation/navigation2 #3102: Planner does not respect obstacles from costmap filter

automech-rb, 2022-08-04:
## Bug report

**Required Info:**

- Operating System:
  - Docker 20.04 Ubuntu
- ROS2 Version:
  - rolling
- Version or commit hash:
  - latest 104bdc51fb72e05af4ccb4fb96a2fd421239f016
- DDS implementation:
  - rmw_cyclonedds_cpp

#### Steps to reproduce issue

```
Use SMAC lattice planner and RPP.
Robot with rectangular footprint
Costmap with keepout filter
```
Same question asked on [ROS answers](https://answers.ros.org/question/404236/nav2-planner-does-not-respect-obstacles-from-costmap-filter/)
#### Expected behavior
I am using keepout costmap filter (as shown in above picture), pink color area is from keepout zone in '**trinary map mode**'.
[image]

When using SMAC Lattice planner, the path generated is so close to obstacles (keepout) that it's definitely colliding with it (pink area with robot footprint). I don't understand why it is like that? 
According to Nav2 [wiki](https://navigation.ros.org/tutorials/docs/navigation2_with_keepout_filter.html), trinary (default mode): Darkness >= occupied_thresh means that map occupied (100). 
#### Actual behavior
Isn't planner supposed to check for collision w.r.t. the footprint (Green box in this case.)? (That is to keep footprint out of the pink area)
#### Additional information

----

SteveMacenski, 2022-08-18:
which footprint is being visualized? Is that the local costmap's or the global costmaps? Obviously the planner will only see the footprint from the global costmap so if that's not set properly, that would cause an issue.

Can you verify that that area is indeed set to LETHAL or INSCRIBED cost? it could potentially be just under it and therefore admissible if you didn't create your filter mask correctly or if the settings of the keepout layer info isn't correct. 

What does the `unsmoothed_plan` look like? 

Can you also attach your configuration and mask file (and anything else needed to reproduce)?

 I agree that this shouldn't be happening if indeed things are set up properly.

SteveMacenski, 2022-08-18:
Not that I think it should make a difference, but if you try the Hybrid-A* planner, do you see the same thing?

automech-rb, 2022-08-22:
Hi @SteveMacenski 
Here are the answers to your questions:
1 - Hybrid-A* does not help. It plans very similar (Close to pink area)
2 - Confirmed Footprint is from global costmap
3-  Pink map is from Global costmap. Local costmap visual is off in Rviz2
4-  Polygon (Rectangular) footprint is same in both costmap
5- Confirmed area is lethal by using python api to get cost value at multiple points of pink area -> value is **254**
6- unsmoothed_plan looks very similar to smooth path. Almost overlapping green unsmooth and pink smooth plan in figure below
[image]
7- I am using this map [repo ](https://github.com/SteveMacenski/nav2_rosdevday_2021/tree/main/nav2_rosdevday_2021/maps) for testing. I am unable to share my robot urdf but I think it can be easily be easily reproduced by any rectangular footprint robot.

here is my config:

```yaml
local_costmap:
  local_costmap:
    ros__parameters:
      update_frequency: 5.0
      publish_frequency: 1.0
      global_frame: odom
      robot_base_frame: base_footprint
      use_sim_time: True
      rolling_window: true
      width: 4
      height: 4
      resolution: 0.05
      #robot_radius: 0.22
      footprint: "[ [0.45,-0.35],[-0.53,-0.35],[-0.53,0.35],[0.45,0.35] ]"
      footprint_padding: 0.0
      plugins: ["obstacle_layer", "inflation_layer"]
      #filters: ["keepout_filter"]
      #keepout_filter:
      #  plugin: "nav2_costmap_2d::KeepoutFilter"
      #  enabled: True
      #  filter_info_topic: "/costmap_filter_info"
      obstacle_layer:
        plugin: "nav2_costmap_2d::ObstacleLayer"
        enabled: True
        observation_sources: front_laser_scan_sensor back_laser_scan_sensor
        front_laser_scan_sensor:
          topic: /scan/front
          max_obstacle_height: 2.0
          clearing: True
          marking: True
          data_type: "LaserScan"
          raytrace_max_range: 2.0
          raytrace_min_range: 0.0
          obstacle_max_range: 2.5
          obstacle_min_range: 0.0
        back_laser_scan_sensor:
          topic: /scan/back
          max_obstacle_height: 2.0
          clearing: True
          marking: True
          data_type: "LaserScan"
          raytrace_max_range: 3.0
          raytrace_min_range: 0.0
          obstacle_max_range: 2.5
          obstacle_min_range: 0.0
      inflation_layer:
        plugin: "nav2_costmap_2d::InflationLayer"
        cost_scaling_factor: 4.0
        inflation_radius: 1.0
      always_send_full_costmap: True

global_costmap:
  global_costmap:
    ros__parameters:
      update_frequency: 1.0
      publish_frequency: 1.0
[... 48 lines cut ...]
  ros__parameters:
    #expected_planner_frequency: 20.0
    use_sim_time: True
    planner_plugins: ["GridBased"]
    GridBased:
      plugin: "nav2_smac_planner/SmacPlannerLattice"
      allow_unknown: false                 # Allow traveling in unknown space
      max_iterations: 1000000             # Maximum total iterations to search for before failing (in case unreachable), set to -1 to disable
      max_planning_time: 5.0              # Max time in s for planner to plan, smooth
      analytic_expansion_ratio: 3.5       # The ratio to attempt analytic expansions during search for final approach.
      analytic_expansion_max_length: 3.0  # For Hybrid/Lattice nodes: The maximum length of the analytic expansion to be considered valid to prevent unsafe shortcutting
      reverse_penalty: 2.0                # Penalty to apply if motion is reversing, must be => 1
      change_penalty: 0.05                # Penalty to apply if motion is changing directions (L to R), must be >= 0
      non_straight_penalty: 1.05          # Penalty to apply if motion is non-straight, must be => 1
      cost_penalty: 2.0                   # Penalty to apply to higher cost areas when adding into the obstacle map dynamic programming distance expansion heuristic. This drives the robot more towards the center of passages. A value between 1.3 - 3.5 is reasonable.
      rotation_penalty: 5.0               # Penalty to apply to in-place rotations, if minimum control set contains them
      retrospective_penalty: 0.015
      lattice_filepath: "/home/overlay_ws/src/agv-common-ros2/agv_description/configs/navigation/state_lattice/output.json"                # The filepath to the state lattice graph
      lookup_table_size: 20.0             # Size of the dubin/reeds-sheep distance window to cache, in meters.
      cache_obstacle_heuristic: false     # Cache the obstacle map dynamic programming distance expansion heuristic between subsiquent replannings of the same goal location. Dramatically speeds up replanning performance (40x) if costmap is largely static.
      allow_reverse_expansion: false      # If true, allows the robot to use the primitives to expand in the mirrored opposite direction of the current robot's orientation (to reverse).
      smooth_path: True                   # If true, does a simple and quick smoothing post-processing to the path
```

Let me know if I can provide any other information.

SteveMacenski, 2022-08-22:
I see the problem. 

https://github.com/ros-planning/navigation2/blob/main/nav2_smac_planner/src/collision_checker.cpp#L106-L108

We check if we need to evaluate the full footprint based on the center cost of the robot. With inflation, we can compute using the same inflation function the cost at which at the center means its furthest bound is in collision (so anything that large or higher than that should be considered for a full footprint check). This is an optimization to collision checking to use point-costs where we can, but evaluate the full footprint where required. Collision checking is expensive, so this is a meaningful speed up. 

However, the keepout zones don't have such an inflation, so the center costs in your image are still `0`, leading it to use the point-check since its not possibly inscribed with an inflated obstacle. 

There are a couple of Nav2 solutions:
- We could re-define the keepout zones (@AlexeyMerzlyakov) to be relative to robot centers, so if you want to "fully" keep your robot out of a zone, it should include the across section of the robot into it
- We could reverse some of the engineering Alexey did and make the costmap filters impacted by the inflation layer (or set optionally to do so?) 
- We could add a parameter to Smac to enable forced full collision checking. It'll have a performance hit, but it'll fix the issue

There are also a solution from the user-space
- In your keepout zone, add a high cost around the keepout zone that would trigger that collision checking and at the same time incentivize the robot to not go into those areas surrounding the keepout zone (either way, that seems sensible) 

What do you think?

automech-rb, 2022-08-22:
Thank you for the breakdown.
The proposed 1st and 3rd solution doesn't look promising due to the following reasons:
- 1st one: If the robot is very rectangular (not circular), it will lead to closing up of narrow path. Some of those paths are important for taking shortcuts in my case
- 3rd one: Having a performance hit doesn't sound appealing to use this feature. Instead it would be better to directly add the keepout in main static map

I think the 2nd option, that is having a parameter to enable inflation layer buildup for lethal keepout obstacles sounds great. Not sure how much runtime performance hit it will cause.

Also, I will try to add high cost around keepout. But still this looks like a half-baked solution and having a expected keepout filter response out-of-the-box would be great!

SteveMacenski, 2022-08-23:
> 1st one: If the robot is very rectangular (not circular), it will lead to closing up of narrow path (assuming overcompensating with wider side). Some of those paths are important for taking shortcuts in my case

This is probably my favorite one since that's an intuitive result, but you're right that it could cause issues in confined spaces and elongated robots. Keep in mind though that irregardless of these issues with Smac, its not unique to Smac. Most planners in ROS only do point-checks as they are purely holonomic with no understanding of orientation so it cannot do footprint checks. So its even a special case already that Smac actually knows enough to be able to do this.

> 3rd one: Having a performance hit doesn't sound appealing to use this feature. Instead it would be better to directly add the keepout in main static map

I'm not sure how bad it would be, I'm guestimating 25%ish, so its not world endingly slower. 

> I think the 2nd option, that is having a parameter to enable inflation layer buildup for lethal keepout obstacles sounds great

I agree with this in mind that this could be a good option not only for Smac but also for the holonomic and other planners mentioned above. We decided before not to inflate the keepout zones because the users define a mask and it seems weird to inflate that mask for collision (given its not a "collision" object). The other big reason is that while keepout zones have a somewhat "valid" interpretation of inflation, other costmap filters do not (e.g. speed limited zones) so those should **never** have it applied.

@AlexeyMerzlyakov what do you think? How difficult would it be to offer an option to inflate **just** keepout zones? I'm thinking, for instance, we allow / document users to be able to place keepout zones into the normal costmap layers list (e.g. check if its a `costmap filter` and if not also of type `keepout zone` we throw exception). Or perhaps if a parameter is set + an inflation layer exists, we place it just below that inflation layer.

AlexeyMerzlyakov, 2022-08-24:
Hi! I would like to agree with @SteveMacenski: costmap filters were intentionally separated from the rest of filters in order to have not being inflated or affected by the inflation layer (example with filters produced no updates on costmap from mask like `speed_filter`, or something similar). This separation was made to avoid overlaps with previous iterations during bounding window changes, that inevitably leads to artifacts at costmaps (see https://github.com/ros-planning/navigation2/issues/2051#issuecomment-812565098 and ticket for more details). So, I do not believe that reversing back the separation is a proper idea.

Since costmap filters are just a Costmap2D plugins, you could just add `inflation_layer` to the filters instead. The inflation layer should be placed _after_ `keepout_filter` and it works for me. The config-file will look something as follows:
```
global_costmap:
  global_costmap:
    ros__parameters:
      ...
      plugins: ["static_layer", "obstacle_layer", "inflation_layer"]
      filters: ["keepout_filter", "inflation_layer"]
      ...
      keepout_filter:
        plugin: "nav2_costmap_2d::KeepoutFilter"
        enabled: True
        filter_info_topic: "/costmap_filter_info"
```

I think, this should resolve the problem.

automech-rb, 2022-08-24:
@AlexeyMerzlyakov Did some quick testing. Adding `inflation_layer` to `filters` works like charm! Thanks!!
From my end, I think this issue can be closed. Maybe having a little insight of this on Nav2 wiki will be helpful for future users!

Thanks again @SteveMacenski  @SteveMacenski

SteveMacenski, 2022-08-24:
https://github.com/ros-planning/navigation.ros.org/blob/master/configuration/packages/costmap-plugins/keepout_filter.rst done

closed 2022-08-24 by SteveMacenski