ros-navigation/navigation2 #3103: Unable to load my own map with Galactic Nav2 (No map is received)

jackykwok2002, 2022-08-04:
## Bug report

**Required Info:**
- Operating System:
  - Ubuntu 20.04
- ROS2 Version:
  - ROS2 Galactic
- Version or commit hash:
  - 104bdc51fb72e05af4ccb4fb96a2fd421239f016
- DDS implementation:
  - Fast-RTPS (default)

#### Steps to reproduce issue

```
sudo apt install ros-galactic-navigation2 -y
ros2 launch nav2_bringup tb3_simulation_launch.py headless:=False
```

#### Expected behavior
Navigation stack should be launched and the turtlebot_world map should be updated and displayed in RVIZ

#### Actual behavior
- Map is not received/updated

#### Additional information
- Also tried to launch my own map with galactic **ros2 launch nav2_bringup bringup_launch.py use_sim_time:=False autostart:=True map:= Path to Map**
- Behavior: Frozen on **[local_costmap.local_costmap]: Creating local_costmap**

----

padhupradheep, 2022-08-04:
are you sure, you installed turtlebot3 packages? 

please follow the instructions given here: https://navigation.ros.org/getting_started/index.html

SteveMacenski, 2022-08-18:
And are you sure that your path to the map is absolute? What other errors do you see, if any?  Does it work fine if you don't specify the map path and uses the default? There are a number of questions here we need answers to to help :smile:

SteveMacenski, 2022-08-22:
Closing due to lack of response and cannot reproduce. If you continue to have a problem, feel free to comment back here or file on ROS Answers.

closed 2022-08-22 by SteveMacenski