ros-navigation/navigation2 #3105: ClearCostmapAroundRobot not in Groot

MrOCW, 2022-08-09:
## Bug report

**Required Info:**

- Operating System:
  - Jetpack 4.6.1 (Ubuntu 18.04) running Ubuntu 22.04 docker container
- ROS2 Version:
  - Humble binaries
- Version or commit hash:
  - latest apt
- DDS implementation:
  - CycloneDDS

#### Steps to reproduce issue
Import node palette into Groot from /opt/ros/humble/nav2_behaviour_tree/nav2_tree_nodes.xml

#### Expected behavior
ClearCostmapAroundRobot to appear in the palette
#### Actual behavior
Only ClearEntireCostmap is available
#### Additional information
During contextual recovery, clear entire costmap is used but it takes a very long time for the global costmap to appear again after being cleared, likely due to a large static map, which almost caused a collision as there is no global costmap for the global planner to identify obstacles. Will using ClearCostmapAroundRobot speed up the time needed for the global costmap to appear again?

SteveMacenski, 2022-08-18:
It can be added to https://github.com/ros-planning/navigation2/blob/main/nav2_behavior_tree/nav2_tree_nodes.xml, would you mind submitting a PR to add this missing BT node so you can use it in groot? I'd be happy to merge it.

enesbedir1, 2022-08-27:
Hİ @SteveMacenski I created a pr #3155  for the issue 
I would be appreciated if you consider reviewing it.

SteveMacenski, 2022-08-27:
Merged!

closed 2022-08-27 by SteveMacenski