ros-navigation/navigation2 #3106: Docker: Navigation2 source installation.

prachandabhanu, 2022-08-10:
Hello,

I am trying to install Navigation2 source install in my docker container. ROS2 Humble is also source installation, the steps as show below,

```
# Build ROS2 core from source source
RUN mkdir -p ${ROS_ROOT}/src && \
    cd ${ROS_ROOT} && \
    # https://answers.ros.org/question/325245/minimal-ros2-installation/?answer=325249#post-id-325249
    rosinstall_generator --deps --rosdistro ${ROS_DISTRO} ros_base \
        angles apriltag behaviortree_cpp_v3 bondcpp camera_calibration_parsers camera_info_manager \
        compressed_image_transport compressed_depth_image_transport cv_bridge demo_nodes_cpp \
        demo_nodes_py diagnostic_updater example_interfaces image_geometry image_pipeline \
        image_transport image_transport_plugins launch_xml launch_yaml launch_testing \
        launch_testing_ament_cmake nav2_msgs ompl resource_retriever rosbridge_suite rqt_image_view \
        rviz2 sensor_msgs slam_toolbox v4l2_camera vision_opencv vision_msgs \
	> ros2.${ROS_DISTRO}.ros_base.rosinstall && \
    cat ros2.${ROS_DISTRO}.ros_base.rosinstall && \
    vcs import src < ros2.${ROS_DISTRO}.ros_base.rosinstall && \
    rm ${ROS_ROOT}/*.rosinstall

RUN cd ${ROS_ROOT} \
    && apt-get update \
    && rosdep init \
    && rosdep update \
    && rosdep install -y \
      --ignore-src \
      --from-paths src \
      --rosdistro ${ROS_DISTRO} \
      --skip-keys "fastcdr rti-connext-dds-6.0.1 rti-connext-dds-5.3.1 urdfdom_headers libopencv-dev libopencv-contrib-dev libopencv-imgproc-dev python-opencv python3-opencv" \
&& rm -Rf /var/lib/apt/lists/* \
&& apt-get clean

# Build ROS2 source
RUN cd ${ROS_ROOT} \
      && colcon build --merge-install --cmake-args -DCMAKE_BUILD_TYPE=RelWithDebInfo --packages-up-to behaviortree_cpp_v3 \
      && colcon build --merge-install --cmake-args -DCMAKE_BUILD_TYPE=RelWithDebInfo \
      && rm -Rf src build log
```

Navigation2 Installation

```
# Install nav2
RUN apt-get update && apt-get upgrade -y && mkdir -p ${ROS_ROOT}/src && cd ${ROS_ROOT}/src \
    && git clone https://github.com/ros-planning/navigation2.git && cd navigation2 && git checkout humble && cd .. \    
    && source ${ROS_ROOT}/setup.bash && cd ${ROS_ROOT} \
    && rosdep install -y -r --ignore-src --from-paths src --rosdistro ${ROS_DISTRO} \
&& rm -rf /var/lib/apt/lists/* \
&& apt-get clean

RUN source ${ROS_ROOT}/setup.bash && cd ${ROS_ROOT} \
      && colcon build --merge-install --cmake-args -DCMAKE_BUILD_TYPE=RelWithDebInfo --packages-up-to-regex nav2* --packages-ignore nav2_system_tests \
      && rm -Rf src build log
```

## Bug report

During the `colcon build` of nav2 packages is failed because of `gtest_vendor`. 

**Required Info:**

- Operating System:
  - Ubuntu 22.04
- ROS2 Version:
  - Humble
- Version or commit hash:
  - git checkout humble
- DDS implementation:
  - `ENV RMW_IMPLEMENTATION=rmw_fastrtps_cpp`

#### Steps to reproduce issue
`docker-compose up --build`

#### Actual behavior
```
--- stderr: gtest_vendor
CMake Error: The source directory "/opt/ros/humble/share/gtest_vendor" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.
---
Failed   <<< gtest_vendor [0.14s, exited with code 1]
```

#### Additional information
Cmake version: 3.23.2

The installation steps are followed similar to [isaac_ros_common](https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_common). I have replaced the base image with Ubuntu 22.04, and the ROS2 installation remains the same.

I am unable to find the solution for above-mentioned error during the colcon build. Can please help me with this.

Thank you.

padhupradheep, 2022-08-10:
Give a check on this [tutorial](https://ncdejito.github.io/nav2-docker/) , I'm really not sure if this would help! Because, in the tutorial they install humble from the binaries and you want to install it from the source.

SteveMacenski, 2022-08-17:
It looks like you did not install the gtest and potentially some other required dependencies from the ROS 2 source install step, though I cannot be entirely sure. Do you have a docker file I can test locally and reproduce?

padhupradheep, 2022-08-18:
> It could also be that something failed in the ROS 2 build step that you didn't notice and therefore it didn't complete that process

Seems that this could be the case!

I just pulled the docker with the humble desktop as mentioned here in the[ ros docs](https://docs.ros.org/en/humble/How-To-Guides/Run-2-nodes-in-single-or-separate-docker-containers.html). 

I did a source build of nav2 and did not run into any issues.

SteveMacenski, 2022-08-18:
We can tentatively leave this open to see if the user gets back to us - hopefully they find the issue or can provide more information pointing to the issue they're experiencing

SteveMacenski, 2022-08-22:
No response - if you continue to have problems, please comment back with the information requested.

closed 2022-08-22 by SteveMacenski

prachandabhanu, 2022-09-22:
@SteveMacenski @padhupradheep Thank you for the answers, and sorry for late reply.

> It could also be that something failed in the ROS 2 build step that you didn't notice and therefore it didn't complete that process or even the -r in rosdep install failed to install something but also silently continued.

Yes, there were much stderr during ROS 2 build steps, but I make sure that gtest components were build correctly.  As @SteveMacenski have suggested, Nav2 build was and still unsuccessful may because of other errors during ROS 2 build.

- You can find my docker file [here](https://github.com/prachandabhanu/docker_envs/tree/main/nav2). 
- Reference from [isaac_ros_common](https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_common/tree/main/docker)

The main change from isaac_ros_common docker file to my docker file is from,

```
ARG BASE_IMAGE=nvcr.io/nvidia/tritonserver:22.03-py3
FROM ${BASE_IMAGE}
```

to,

```
ARG BASE_IMAGE=ubuntu:22.04
FROM ${BASE_IMAGE}
```
in **Dockerfile.x86_64** file. Also I have commented few installation steps that deemed unnecessary.

prachandabhanu, 2022-09-22:
During ROS 2 build I get errors such as this 

```
--- stderr: mimick_vendor
Cloning into 'mimick-de11f8377eb95f932a03707b583bf3d4ce5bd3e7'...
HEAD is now at de11f83 Support raspberry pi zero (#21)
---
```