ros-navigation/navigation2 #3109: Deadlock when a static layer gets map updated

daisukes, 2022-08-12:
I found a deadlock bug when I tested our new prototype, which frequently updates the static costmap.

## Bug report

**Required Info:**

- Operating System:
  - Ubuntu20.04
- ROS2 Version:
  - Galactic (ros:galactic docker image)
- Version or commit hash:
  - Galactic sync 2-based (a25cd438aa4cb08735ba9a6e9213c78b8ad7f528) custom build
- DDS implementation:
  - default (cyclone dds)

#### Steps to reproduce issue
- frequently update the static costmap (2hz)

#### Expected behavior
- no deadlock

#### Actual behavior
- deadlock
- gdb traceback
  - https://gist.github.com/daisukes/5e6e1d3758daa0c34996d60822b41170

#### Additional information
- main thread locks (Thread 15 in the gdb log)
  1. the mutex of the layered costmap
    https://github.com/ros-planning/navigation2/blob/6a9508b772eb6f07cbfd350b3578aa9de2567d59/nav2_costmap_2d/src/layered_costmap.cpp#L130
  2. then, the mutex of the static layer
    https://github.com/ros-planning/navigation2/blob/6a9508b772eb6f07cbfd350b3578aa9de2567d59/nav2_costmap_2d/plugins/static_layer.cpp#L336
- updating map thread locks (Thread 13 in the gdb log)
  1. the mutex of the static layer
    https://github.com/ros-planning/navigation2/blob/6a9508b772eb6f07cbfd350b3578aa9de2567d59/nav2_costmap_2d/plugins/static_layer.cpp#L272
  2. then, the mutex of the layered costmap, when the layered costmap needs to be resized
    https://github.com/ros-planning/navigation2/blob/6a9508b772eb6f07cbfd350b3578aa9de2567d59/nav2_costmap_2d/src/layered_costmap.cpp#L104

#### Implementation considerations
The mutex of the static layer should  be unlocked before calling resizeMap of the layered costmap
https://github.com/ros-planning/navigation2/blob/6a9508b772eb6f07cbfd350b3578aa9de2567d59/nav2_costmap_2d/plugins/static_layer.cpp#L194-L198

I think this lock can be removed. What do you think?
https://github.com/ros-planning/navigation2/blob/6a9508b772eb6f07cbfd350b3578aa9de2567d59/nav2_costmap_2d/plugins/static_layer.cpp#L272

SteveMacenski, 2022-08-19:
@daisukes thanks for filing, back from PTO now and took a look at this.

I think the best solution is the following:
- Place the static layer mutex lock in the `processMap()` function. I don't think that in either `updateBounds` nor `incomingMap`  there needs to be a lock for the full function. We really just need to protect the modifications within `processMap` and the `map_buffer_` object. 
- Always set the `new_map` to `map_buffer_` in the `incomingMap` function and pass it that pointer / make `processMap` use `map_buffer_` internally. Then at the end of the function, set it to `nullptr`. That will allow everything we need protected to be under the lock within `processMap`.
- Within `processMap` unlock where required with the layered costmap

I don't think just removing https://github.com/ros-planning/navigation2/blob/6a9508b772eb6f07cbfd350b3578aa9de2567d59/nav2_costmap_2d/plugins/static_layer.cpp#L272 works, because that then exposes `map_buffer_` outside of the mutex which both threads modify. There could be a situation where `update_in_progress_` is set to `true` in `updateBounds` and while `processMap(*map_buffer)` is being called, `map_buffer_ = new_map;` is being set in `incomingMap`. 

Could you prototype that and let me know if it fixes your issue?

daisukes, 2022-08-20:
Thank you for the comments.
How about this change? I made a test code that reproduces the deadlock, and this works fine with the test code.
- I think it does not need to call mapProcess twice when map_received_ is false
- `map_buffer_ = nullptr` is not needed when update_in_progress_ is false
- isolate processMap from the lock_guard
```c++
void
StaticLayer::incomingMap(const nav_msgs::msg::OccupancyGrid::SharedPtr new_map)
{
  {
    std::lock_guard<Costmap2D::mutex_t> guard(*getMutex());
    if (!map_received_) {
      map_received_ = true;
    }

    if (update_in_progress_.load()) {
      map_buffer_ = new_map;
      return;
    }
  }
  processMap(*new_map);
}
```

original code
https://github.com/ros-planning/navigation2/blob/6a9508b772eb6f07cbfd350b3578aa9de2567d59/nav2_costmap_2d/plugins/static_layer.cpp#L269-L283

SteveMacenski, 2022-08-22:
I think that would work if in `updateBounds` the check https://github.com/ros-planning/navigation2/blob/main/nav2_costmap_2d/plugins/static_layer.cpp#L340-L342 is put under the mutex in L334.

SteveMacenski, 2022-08-23:
For a new map, does it matter? If we have a map pending to be processed (which would completely overwrite the existing one) but a new one comes in to overwrite that -- does it matter if we didn't process the one in the middle if no costmap update was requested? I would think not.

closed 2022-08-31 by SteveMacenski