ros-navigation/navigation2 #3111: Map fails to load if params-file is given but without map

superseppl, 2022-08-13:
## Bug report

**Required Info:**

- Operating System:
 - Ubuntu 20.04
- ROS2 Version:
  - Foxy binaries
- Version or commit hash:
  - ros-foxy-navigation2    0.4.7-1focal.20220307.024942
- DDS implementation:
  - [rmw_cyclonedds](https://github.com/ros2/rmw_cyclonedds).

#### Steps to reproduce issue

Hi guys,
while loading a map it occurs that whenever I use 
```sh
ros2 launch nav2_bringup bringup_launch.py map:=/path/to/map.yaml
```
the map server works. But as soon as I add a param file without any map_server param inside the file but with the map param
```sh
ros2 launch nav2_bringup bringup_launch.py params_file:=/path/to/params.yaml" map:=/path/to/map.yaml 
```
it crashes. As soon as I add 
```yaml
 map_server:
   ros__parameters:
     use_sim_time: False
     yaml_filename: "xyz.yaml"
```
to the yaml file and start it with 
```sh
ros2 launch nav2_bringup bringup_launch.py params_file:=/path/to/params.yaml" map:=/path/to/map.yaml 
```
it works. 

Issues: 
 - I cannot launch the file even tough I defined a map in the yaml file as the map parameter is compulsory.
 - The yaml file parameter is not used and can be different to the map parameter used with CLI. 

## Feature request

#### Feature description
It would be nice to use only one parameter when starting. As the map parameter is always expected when launching the node it should only start with this one and should not require anything given in the param-file.

padhupradheep, 2022-08-15:
Is there any terminal output that you can share?

padhupradheep, 2022-08-15:
Strangely I cannot reproduce the issue in the main branch. 

Are you sure that, you are providing the absolute path to the `map` argument during the launch?

SteveMacenski, 2022-08-16:
Moreover, while Foxy is still an active distribution, we no longer support it in this project. We were still in heavy development at that time and the sheer volume of changes between foxy and even Galactic are so large it became virtually impossible to provide support for Foxy backwards. 

My personal recommendation is to update to any more recent distribution. There are a litany of improvements made in stability/reliability and features. 

But I would be happy to field any PRs to Foxy to resolve the issue if you find the root cause, but it unfortunately won’t be prioritized by maintainers.

SteveMacenski, 2022-08-18:
I don't understand. The map is required for the map server. There is no valid use of the map server without a map.

SteveMacenski, 2022-08-18:
Ah, I think I see. 

There is a non-trivial default in the `tb3_simulation_launch.py` file for the `map` which would be remapped over any value in the `yaml_filename` in the configuration files using `RewrittenYaml`. But in order for rewritten yaml to work, the key has to exist in the yaml file in the first place, so requiring `yaml_filename` to be in the yaml but with a bogus and entirely ignored value. 

The issue arises that folks wanted to be able to remap the map on the commandline, but if both a launch configuration for `yaml_filename` and a value for it exists in the `yaml` file, there's a collision and launch doesn't know which to use (I'm not sure which it would resolve to, or if it would straight up throw an exception if you tried). So we have to rewrite it. However, without providing more information in the launch file, if we wanted to insert `yaml_filename` into the yaml file, we wouldn't know where to do it if it wasn't already there to overwrite. 

So it leaves us in a bit of an odd situation, where it has to be in the yaml file to work, but will always be overridden if the `map` launch configuration contains a default value. If you remove that [default value](https://github.com/ros-planning/navigation2/blob/main/nav2_bringup/launch/tb3_simulation_launch.py#L88) though, it should work to use the yaml file's map contents

SteveMacenski, 2022-08-18:
https://github.com/ros-planning/navigation2/pull/3123/files I changed to the empty string and also added a comment to that effect. What do you think?

padhupradheep, 2022-08-19:
Just leaving my opinion. As far as I saw, if you have both the `yaml_filename` and the `map` argument specified during the launch, the later is considered by the map server. There is actually no effect, even if you have the `yaml_filename` specified in your configs. 

> The only other 2 options I see are to (1) remove the commandline support for changing maps or (2) require users to always specify a yaml_filename, even if just an empty string, in the yaml file.

Considering the point that I made, I would vote for the second option, as it would definitely remove the confusion regarding the conflicts. If the user does not want to set the path either through the command_line / specify it in the default launch, then they can overwrite the empty string with the absolute path that they want to use. Also, I know a lot of people who use CLI to change the maps or select the maps during launch (me included :sweat_smile:  )

SteveMacenski, 2022-08-19:
Documentation doesn't solve everything, but this is an odd quirk that should definitely be documented if its going to exist: https://github.com/ros-planning/navigation.ros.org/pull/338 + in the default navigation yaml file there's a note to the same effect.

I think that's the best solution we're going to come up with here unless/until the ROS 2 parameter API changes letting us specify multiple parameter sets with priority levels (which I don't think will ever happen, nor even being discussed to my knowledge). 

@superseppl I'm going to close this now since we've come up with the "why" and there isn't really a solution. However, you bring up a very good point that this is very subtle and not well documented, so I went ahead and documented it both in the yaml files and in the official documentation for the parameter so that anyone reading docs or starting from our yaml files will have this clearly denoted to them.

I really appreciate the discussion this generated. 

tl;dr for later readers
- If you want to specify in yaml, remove the default from the `map` launch configuration
- If you use the default / CLI set the map on launch, any value in the yaml is bogus and going to be overridden

closed 2022-08-19 by SteveMacenski