ros-navigation/navigation2 #3116: CMake Error in src/CMakeLists.txt:   Imported target "tf2_geometry_msgs::tf2_geometry_msgs" includes   non-existent path

BATAxjh, 2022-08-16:
when I build navigation2: colcon build --symlink-install

ros2-foxy

CMake Error in src/CMakeLists.txt:
  Imported target "tf2_geometry_msgs::tf2_geometry_msgs" includes
  non-existent path

    "/home/kairui/cc_ws_release/sysroot_docker/usr/include/eigen3"

  in its INTERFACE_INCLUDE_DIRECTORIES.  Possible reasons include:

  * The path was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and references files it does not
  provide.

CMake Generate step failed.  Build files cannot be regenerated correctly.
---
Failed   <<< nav2_util [13.9s, exited with code 1]

Summary: 7 packages finished [29.4s]
  1 package failed: nav2_util
  1 package had stderr output: nav2_util
  24 packages not processed

BATAxjh, 2022-08-16:
git clone https://github.com/ros-planning/navigation2.git --branch foxy-devel    I git from here

BATAxjh, 2022-08-16:
> git clone https://github.com/ros-planning/navigation2.git --branch foxy-devel

SteveMacenski, 2022-08-16:
It looks ok on my side & CI is turning over so its likely to do with your install of the dependencies or Nav2 build branch. Please investigate further or check out out getting started / install pages on navigation.ros.org.

closed 2022-08-16 by SteveMacenski

BATAxjh, 2022-08-16:
"/home/kairui/cc_ws_release/sysroot_docker/usr/include/eigen3"  the path is not mine, Why report this error

BATAxjh, 2022-08-17:
home/sunrise/robot_robot/build/nav2_util/test/CMakeFiles/test_service_client.dir/flags.make:9:CXX_INCLUDES = -I/opt/tros/src/gtest_vendor/include -I/home/sunrise/robot_robot/src/navigation2/nav2_util/include -isystem /opt/tros/include -isystem /home/sunrise/robot_robot/install/nav2_msgs/include -isystem /home/kairui/cc_ws_release/sysroot_docker/usr/include/eigen3 

because  

opt/tros/lib/pkgconfig/orocos-kdl.pc:10:Cflags: -I${includedir}  -I/home/kairui/cc_ws_release/sysroot_docker/usr/include/eigen3
opt/tros/lib/pkgconfig/orocos_kdl.pc:10:Cflags: -I${includedir}  -I/home/kairui/cc_ws_release/sysroot_docker/usr/include/eigen3
opt/tros/share/laser_geometry/cmake/laser_geometryExport.cmake:57:  INTERFACE_INCLUDE_DIRECTORIES "${_IMPORT_PREFIX}/include;/home/kairui/cc_ws_release/sysroot_docker/usr/include/eigen3;${_IMPORT_PREFIX}/include"
opt/tros/share/orocos_kdl/cmake/OrocosKDLTargets.cmake:57:  INTERFACE_INCLUDE_DIRECTORIES "${_IMPORT_PREFIX}/include;/home/kairui/cc_ws_release/sysroot_docker/usr/include/eigen3"
opt/tros/share/tf2_eigen_kdl/cmake/tf2_eigen_kdlExport.cmake:57:  INTERFACE_INCLUDE_DIRECTORIES "${_IMPORT_PREFIX}/include;/home/kairui/cc_ws_release/sysroot_docker/usr/include/eigen3"

there files path are /home/kairui/cc_ws_release/sysroot_docker/usr/include/eigen3  
I am changed /usr/include/eigen3      ok